verilator/verilator #4632: Linting WIDTHEXPAND, WIDTHTRUNC for if branches inside always on parameter as condition

Risto97, 2023-10-27:
Run into a bit of a weird code, but I still think its valid.
However verilator is throwing linting warning, so I just want to leave it here.

```
module test #(
    parameter SECDED = 2
)(
    output logic [11+SECDED:0] code_o
);

    always_comb begin
        if (SECDED == 0) begin
            code_o = 12'h1;
        end else
            code_o = {SECDED+12{1'b0}};
    end

endmodule
```

The first branch if(SECDED==0) is excluded with a constant parameter SECDED, so the code inside the block should be irrelevant.
It still reports a WIDTHEXPAND lint warning.
This code can be fixed with if outside always and assign statements, however, I think they are equivalent and both should not be linting warnings?
Probably the problem is with verilator not checking if branch condition is a parameter or a constant if its inside a always block.

Error:
```
%Warning-WIDTHEXPAND: test.v:9:20: Operator ASSIGN expects 14 bits on the Assign RHS, but Assign RHS's CONST '12'h1' generates 12 bits.
                                 : ... In instance test
    9 |             code_o = 12'h1;
      |                    ^
                      ... For warning description see https://verilator.org/warn/WIDTHEXPAND?v=5.012
                      ... Use "/* verilator lint_off WIDTHEXPAND */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s
```

verilator --version

```
Verilator 5.012 2023-06-13 rev v5.012 (mod)
```

May we assist you in trying to fix this in Verilator yourself?

Sure, not sure when I will have time though. Just reporting it for later.

jjones-github, 2023-10-27:
To me that seems like a valid warning. You should be able to remove the warning by using similar code in the `if` as in the `else`.

Like: `code_o = {{SECDED+11{1'b0}}, 1'b1};`

kbieganski, 2023-10-27:
This is probably happening because width checking is done before dead code elimination. Changing this behavior seems like quite a big effort for relatively little gain.

wsnyder, 2023-10-27:
As @kbieganski said I don't think we want to change this at this time.  I'm going to mark not fixing for now.

I'd note the code can be written without the resizing, Verilator knows not to warn on "obvious" constants.  (But maybe you simplified something more complicated).

```
       if (SECDED == 0) begin
            code_o = 1;
        end else
            code_o = '0;
```

closed 2023-10-27 by wsnyder