verilator/verilator #4639: "Icarus Verilog" link is dead

PoignardAzur, 2023-10-29:
The README has three links to `http://iverilog.icarus.com/`. But this domain is dead and redirects to a google page showing this error:

> **Configuration error**
>
> iverilog.icarus.com points to the site "iverilog" which does not exist. Please contact the domain administrator.

A quick search for icarus verilog gives me this repository, not sure if it's the right one: https://github.com/steveicarus/iverilog

wsnyder, 2023-10-29:
It's also broken under https://en.wikipedia.org/wiki/Icarus_Verilog

Can you please ask the icarus folks what link they want and report back?

wsnyder, 2023-10-29:
P.S. e.g. file an issue at https://github.com/steveicarus/iverilog

PoignardAzur, 2023-10-30:
According to [this issue](https://github.com/steveicarus/iverilog/issues/886) the new official site is https://steveicarus.github.io/iverilog/

closed 2023-10-30 by wsnyder