verilator/verilator #4644: Assessing the compatibility of Verilator's current and future versions with Linux versions 7 and earlier

youfares, 2023-10-30:
Hello,

I've noticed that the latest Verilator version (5.016) is only compatible with Linux 8 and not with earlier versions like Linux 7. I'm curious if future Verilator versions will continue to be compatible exclusively with Linux 8 and newer, or if there are plans to make them compatible with older Linux versions, specifically Linux 7 or later.

Thank you for your prompt response.

wsnyder, 2023-10-31:
Verilator requires a C++11 compiler, anything requiring something newer is a bug which we'd accept a pull request to fix (for today anyways).    As described in https://veripool.org/guide/latest/deprecations.html we're past due requiring C++14, and will in over another year will require C++20.  Redhat typically provides newer compiler tool chains for their old distributions you can use.  Or if your site is like most (I presume you are in a large organization as no one else would be on Linux 7), you have some systems running a more reasonably new version of Linux you can alternatively use.
  
BTW Linux 7 will be 10 years old and end-of-support in June 2024 so your IT is presumably already planning a move, hopefully well past Linux 8 ;)

closed 2023-10-31 by wsnyder

udif, 2023-10-31:
By Linux 7 I guess you mean ReadHat 7, CentOS 7, which are very popular in the CAE world.
My solution for compiling modern open source CAE tools in this environment is to use a different modern Linux distribution for compilation, find the shared library dependencies of the resulting binaries, collect these into a directory, and then to use `patchelf` so that the executable takes those libraries from the specific directory.
The resulting binary runs fine on CentOS 7 even though I compiled it on a different distribution such as Ubuntu 22.04 .

youfares, 2023-11-06:
Thank you for your answer.