verilator/verilator #4648: Issue with some SystemC datatypes

mbalboni07, 2023-10-30:
v5.016

Generated code is failing at compilation time due to this error

`error: cannot pass object of non-trivial type 'sc_core::sc_out<sc_dt::sc_uint<64>>' through variadic function; call will abort at runtime [-Wnon-pod-varargs]`

wsnyder, 2023-10-31:
Did this used to work for you?  I suspect not and you're trying to hook a sc_uint<64> to a verilated port - verilator uses uint64_t.

mbalboni07, 2023-10-31:
Super thanks again for the quick answer. I was verilating same code with v4.226 and it was working. Do you know if something could have changed about support of systemc in the newest releases?
Big thanks in advance

wsnyder, 2023-10-31:
That was over a year ago, a lot has changed.

wsnyder, 2023-11-11:
Closing presuming you figured this out, feel free to follow up here if otherwise.

closed 2023-11-11 by wsnyder