verilator/verilator #4651: Garbage collector

kbieganski, 2023-10-31:
Verilator currently uses reference counting for memory management of SystemVerilog class instances (objects). However, SystemVerilog allows cyclical references between objects. Due to this, reference counting is insufficient, and easily results in memory leaks. The typical solution to this is a [tracing garbage collector](https://en.wikipedia.org/wiki/Tracing_garbage_collection).

Now that the class implementation is getting quite mature in Verilator, we should start thinking about how to incorporate such a GC. I think this will be crucial in big UVM testbenches (or any other testbenches that use classes).

The way I see it, there are two ways to go about this:

1. Write our own GC. @akiryk, @kboronski-ant, and me had a discussion about what that could look like.
   - `VlClass`, which is the base class for all verilated classes, contains a flag called `reachable`, as well as a virtual method called `mark()`.
   - Verilated classes would override this method. The method marks its receiver object as reachable, and then calls `mark()` on all its member class references (if the object is already marked, it early-returns).
   - All objects reachable from static references or from call stacks are kept in a ‘root’ list. This list could simply be implemented as 'next' pointers in `VlClass`.
   - All objects not reachable from those locations are kept in a ‘potentially unreachable’ list (also implemented as ‘next’ pointers).
   - Every so often the garbage collector runs. There are two steps:
     1. Call `mark()` on each object in the ‘root’ list.
     2. Sweep the ‘potentially unreachable’ list: any objects with `reachable == false` get deleted. Otherwise the flag gets reset.
   - How often the GC runs is configurable. Could be some number of time slots, or just how much real time passed. A good thing about SystemVerilog is that the concept of time slots gives us a “stop-the-world” moment for free.
   - The trickiest part is probably keeping the 'root'/’potentially unreachable’ lists. We'd have to move objects from ‘potentially unreachable’ to ‘root’ whenever they are assigned to a static or local variable. Each ‘root’ entry would need a reference counter to keep track of how many static or local references there are. Once that counter reaches 0, the object is moved to the ‘potentially unreachable’ list. So whenever such a reference goes out of scope or gets assigned to another static/local var, the list needs to be updated. This could probably be achieved by extending the current smart pointer implementation.
   - There’s obviously lots of room for improvements/optimization after that, like generational GC, etc. But that’s far off at this point.
2. Use an existing GC library for C++. The most mature and widely-used one is [`bdwgc`](https://github.com/ivmai/bdwgc), also known as `libgc`. Seems to be used (or have been used) by [a variety of programming languages and VMs](https://github.com/ivmai/bdwgc/wiki/Known-clients), including Mono, PyPy, D, Scala Native, Unity (IL2CPP), Nix, Racket, SWI-Prolog. There's no possible way that Verilator's custom GC gets the amount of care and attention that `bdwgc` has received over the years.
   The obvious downside is that it's an additional dependency, but I suppose it could just be vendored in releases. A less obvious downside is that it possibly doesn’t play well with coroutines (see https://github.com/ivmai/bdwgc/issues/362). Also, it’s not clear how it would affect client code that wraps the compiled design. Further research is needed here.

wsnyder, 2023-10-31:
Do we have models that break now because of this?  e.g. does the UVM library itself as currently written have circular references that can cause leaks?

The new SV standard will add weak references, and I wonder if with people adopting that for circular structures we can avoid dynamic garbage collection for a bit longer.

solomatnikov, 2023-11-03:
IMHO garbage collection in any EDA/CAD tool is a really bad idea. Decades of experience showed that there is no way to make garbage collection work well in cases where performance and memory footprint matter.

Verilator already uses a lot of memory for large designs and takes a lot of time.

Wondering if it is possible to limit garbage collection to just a few places where it is truly unavoidable, like UVM objects, and reference counting everywhere else.

kbieganski, 2023-11-06:
> Do we have models that break now because of this? e.g. does the UVM library itself as currently written have circular references that can cause leaks?

Yes. UVM's `uvm_component` has a reference to a parent component and an assoc array of child components. Assuming each and every component references its children and parent, currently no `uvm_component`s will ever be freed (unless those references get nulled manually, but I doubt it).

That's the simplest example, but I expect there to be much more of this. We found UVM to be quite the spaghetti monster while developing support for it.

> The new SV standard will add weak references, and I wonder if with people adopting that for circular structures we can avoid dynamic garbage collection for a bit longer.

Unless they rewrite big parts of UVM, I'm afraid it doesn't help. But we can special-case situations where cycles cannot exist, and use refcounting for that. It's not too hard, and some dynamic languages do it, but it's also not necessarily faster than a GC.

> IMHO garbage collection in any EDA/CAD tool is a really bad idea. Decades of experience showed that there is no way to make garbage collection work well in cases where performance and memory footprint matter.
>
> Verilator already uses a lot of memory for large designs and takes a lot of time.

Note that this only affects classes. Modules are static, there's no memory management to speak of there.
I fail to see how a GC increases the memory footprint, especially compared to now when it's trivial to leak memory, and it probably happens all over the place in UVM.
Also, the problem domain doesn't matter; SystemVerilog's classes are no different from a language like Java, both in terms of design and performance considerations.

wsnyder, 2023-11-07:
My limited understanding is UVM components are not designed to ever be deregistered, they always persist until end of simulation.  Is there another UVM example?

If there's a few references in UVM that can be made weak to "fix" UVM, I suspect we can get it upstreamed; the UVM maintainers have already accepted some warning fixes for Verilator in the next UVM version.

kiryk, 2023-11-09:
Currently it is quite easy to make the CI fail by creating [memory leaks](https://github.com/antmicro/verilator-1/actions/runs/6801751265/job/18494024703) in [legal code](https://github.com/antmicro/verilator-1/commit/bdf60e37284085a7a733bdeed1bd3a8853f0ea68). The problem disallows us from using structures like graphs or doubly linked lists.

The CI would also detect leaks in UVM runs. Even if they were caused solely by UVM components, it doesn't seem right to intentionally avoid testing it.

closed 2024-04-13 by wsnyder