verilator/verilator #4657: Verilator build phase is taking too long how to profile it

gkamendje, 2023-11-01:
For a new project,  Verilator build phase is taking too long (more than 40 minutes which is 40 times longer than what I see on even driven simulators). This is quite unusual and contrasting with my previous Verilator experience on other projects. I was wondering what the recommended approach to debug this issue would be. I would like to know which particular code block is eating up most of the time.   I have used `--prof-c` and `--prof-cfuncs` options but so far, but I only get run time profiling which is not the issue. Disabling trace does not fix the issue either. I have some delay assignments `assign #del my_sig3 = my_sig1 & my_sig2;` (and therefore need `--timing`) but even commenting out those delay does not significantly reduce the build time.  I am using the following command line (same behavior with other versions of Verilator as well).
```
(/usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' /cad/verilator/v5.014-149-g57c816f90/bin/verilator --trace -cc --exe --timing --build --Mdir /var/tmp/gkamendje/verilator_work_dirs/ltag7_top/rtl/work --timing --report-unoptflat  --output-split 500000 --timescale 1ns/100ps --x-assign unique --x-initial unique    --prof-c --prof-cfuncs -f /home/gkamendje/project.repo/ltag7/ver/sim/../../rtl/man.f -f /home/gkamendje/project.repo/ltag7/ver/sim/../th/man.f --top-module ltag7_tb  /home/gkamendje/project.repo/ltag7/ver/sim/../th/verilator_wrapper.cpp) 2>&1 | tee  /var/tmp/gkamendje/verilator_work_dirs/ltag7_top/rtl/work/ltag7_top_rtl_comp.log 
make[1]: Entering directory '/var/tmp/gkamendje/verilator_work_dirs/ltag7_top/rtl/work'
ccache g++  -I.  -MMD -I/cad/verilator/v5.014-149-g57c816f90/share/verilator/include -I/cad/verilator/v5.014-149-g57c816f90/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -pg  -fcoroutines -Os -c -o verilator_wrapper.o /home/gkamendje/project.repo/ltag7/ver/sim/../th/verilator_wrapper.cpp
ccache g++ -Os  -I.  -MMD -I/cad/verilator/v5.014-149-g57c816f90/share/verilator/include -I/cad/verilator/v5.014-149-g57c816f90/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -pg  -fcoroutines -c -o verilated.o /cad/verilator/v5.014-149-g57c816f90/share/verilator/include/verilated.cpp
ccache g++ -Os  -I.  -MMD -I/cad/verilator/v5.014-149-g57c816f90/share/verilator/include -I/cad/verilator/v5.014-149-g57c816f90/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -pg  -fcoroutines -c -o verilated_vcd_c.o /cad/verilator/v5.014-149-g57c816f90/share/verilator/include/verilated_vcd_c.cpp
ccache g++ -Os  -I.  -MMD -I/cad/verilator/v5.014-149-g57c816f90/share/verilator/include -I/cad/verilator/v5.014-149-g57c816f90/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -pg  -fcoroutines -c -o verilated_timing.o /cad/verilator/v5.014-149-g57c816f90/share/verilator/include/verilated_timing.cpp
ccache g++ -Os  -I.  -MMD -I/cad/verilator/v5.014-149-g57c816f90/share/verilator/include -I/cad/verilator/v5.014-149-g57c816f90/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -pg  -fcoroutines -c -o verilated_threads.o /cad/verilator/v5.014-149-g57c816f90/share/verilator/include/verilated_threads.cpp
/usr/bin/python3 /cad/verilator/v5.014-149-g57c816f90/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vltag7_tb.cpp Vltag7_tb___024root__DepSet_h6d652bbb__0.cpp Vltag7_tb___024root__DepSet_hfbf74690__0.cpp Vltag7_tb__Trace__0.cpp Vltag7_tb__ConstPool_0.cpp Vltag7_tb___024root__Slow.cpp Vltag7_tb___024root__DepSet_h6d652bbb__0__Slow.cpp Vltag7_tb___024root__DepSet_hfbf74690__0__Slow.cpp Vltag7_tb__Syms.cpp Vltag7_tb__Trace__0__Slow.cpp > Vltag7_tb__ALL.cpp
ccache g++ -Os  -I.  -MMD -I/cad/verilator/v5.014-149-g57c816f90/share/verilator/include -I/cad/verilator/v5.014-149-g57c816f90/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -pg  -fcoroutines -c -o Vltag7_tb__ALL.o Vltag7_tb__ALL.cpp
echo "" > Vltag7_tb__ALL.verilator_deplist.tmp
Archive ar -rcs Vltag7_tb__ALL.a Vltag7_tb__ALL.o
g++     -pg verilator_wrapper.o verilated.o verilated_vcd_c.o verilated_timing.o verilated_threads.o Vltag7_tb__ALL.a    -pthread -lpthread -latomic   -o Vltag7_tb
rm Vltag7_tb__ALL.verilator_deplist.tmp
make[1]: Leaving directory '/var/tmp/gkamendje/verilator_work_dirs/ltag7_top/rtl/work'
Elapsed time: 41:52.07[h:]min:sec. CPU time: user 2505.15 sys 6.59 (99%). Peak memory: 3240144KB.
```

wsnyder, 2023-11-01:
How big is the design - can you attach the --stats output file after removing any sensitive information?

For a large design 40m of CPU time is not unusual.  You seem to be using only a single core - with a decent system of say 16 cores in parallel it should be only 3-4 minutes.  Alternatively, try -O0 or look at what compiler optimization flags are responsible for most of the delay.

Note the most recent 4.018 release may also help by using precompiled headers.

Closing as answered but feel free to follow up here.

gkamendje, 2023-11-02:
The design is pretty small. Only 2k instances. Here are the stats obtained with v 5.018 on a machine with 4 cores (still 34mins execution time). I can't identify anything suspicious that could justify such a long runtime (for the sake of comparison, it only takes less than 1 min on an even driven simulator). 

```
Verilator Statistics Report

Information:
  Verilator 5.019 devel rev v5.018-2-g2a57ead7e
  Arguments: --trace --stats -cc --exe --timing --build -j 4 --Mdir /var/path_word_dir/work --timing --report-unoptflat --output-split 500000 --timescale 1ns/100ps --x-assign unique --x-initial unique  -O3 -f /path_to_man/rtl/man.f -f /path_to_man/../th/man.f --top-module my_tb /path_to_wapper/th/verilator_wrapper.cpp defs.sv top.sv my_tb.sv ${VER_PATH}/th/manchester_decoder.sv ${VER_PATH}/th/biphase_decoder.sv ${VER_PATH}/th/fsk_decoder.sv ${NVM_MODEL}
  Build jobs: 4
  Verilate jobs: 4

Global Statistics:

  Assertions, assert immediate statements                                                             0
  Assertions, assert non-immediate statements                                                         0
  Assertions, cover statements                                                                        0
  Assertions, full/parallel case                                                                      0
  ConstPool, Constants emitted                                                                        0
  ConstPool, Tables emitted                                                                           9
  Optimizations, Cases complex                                                                       19
  Optimizations, Cases parallelized                                                                  16
  Optimizations, Clocker decomposed vectors                                                           0
  Optimizations, Clocker seen vectors                                                                 0
  Optimizations, Combined CFuncs                                                                      2
  Optimizations, Const bit op reduction                                                             394
  Optimizations, Const bit op reduction                                                               0
  Optimizations, DFG  pre inline 1st CSE, expressions eliminated                                     90
  Optimizations, DFG  pre inline 2nd CSE, expressions eliminated                                      0
  Optimizations, DFG  pre inline Ast2Dfg, coalesced assignments                                      46
  Optimizations, DFG  pre inline Ast2Dfg, input equations                                           106
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (dtype)                                   0
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (impure)                                  0
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (lhs)                                     0
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (node)                                    0
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (timing)                                  1
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (unknown)                                 2
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (var ref)                                 0
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (width)                                   0
  Optimizations, DFG  pre inline Ast2Dfg, representable                                             103
  Optimizations, DFG  pre inline Dfg2Ast, intermediate variables                                     83
  Optimizations, DFG  pre inline Dfg2Ast, replaced variables                                          1
  Optimizations, DFG  pre inline Dfg2Ast, result equations                                          165
  Optimizations, DFG  pre inline General, modules                                                    17
  Optimizations, DFG  pre inline Peephole, fold assoc binary                                          0
  Optimizations, DFG  pre inline Peephole, fold assoc binary lhs of rhs                               0
  Optimizations, DFG  pre inline Peephole, fold assoc binary rhs of lhs                               0
  Optimizations, DFG  pre inline Peephole, fold binary                                                0
  Optimizations, DFG  pre inline Peephole, fold sel                                                   0
  Optimizations, DFG  pre inline Peephole, fold unary                                                 0
  Optimizations, DFG  pre inline Peephole, inline arraysel                                            0
  Optimizations, DFG  pre inline Peephole, pull nots through cond                                     0
  Optimizations, DFG  pre inline Peephole, push bitwise op through concat                             0
  Optimizations, DFG  pre inline Peephole, push bitwise through reduction                             0
  Optimizations, DFG  pre inline Peephole, push compare op through concat                             0
  Optimizations, DFG  pre inline Peephole, push concat through nots                                   0
  Optimizations, DFG  pre inline Peephole, push not through cond                                      0
  Optimizations, DFG  pre inline Peephole, push reduction through concat                              0
[... cut ...]
  Vars, width    64                                                  4          4          4          4
  Vars, width    96                                                                        1          1
  Vars, width   128                                                                        1          1
  Vars, width   160                                                                        4          4
  Vars, width   192                                                                        2          2

```

wsnyder, 2023-11-02:
It Verilates in under 1 second?   It seems something may be broken with gcc. What gcc version?  How long is -Os, how long is -O0?  How long are both with clang++?

gkamendje, 2023-11-02:
I am using GCC 12 
```
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/opt/gcc-12.2.0/libexec/gcc/x86_64-pc-linux-gnu/12.2.0/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: ../configure --prefix=/opt/gcc-12.2.0 --disable-multilib
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 12.2.0 (GCC) 
```

It was still running after more than 5 hours when using `-O0`. I get some ` %Warning-UNOPTFLAT:` messages when I use `-0s`. I need to fix those first.

wsnyder, 2023-11-02:
UNOPTFLAT might matter to runtime but shouldn't matter to compile time.

If -O0 is slower than -Os then I'd say 90% it's a gcc bug.  What is clang time?

gezalore, 2023-11-02:
Also see GCC [-ftime-report](https://gcc.gnu.org/onlinedocs/gcc/Developer-Options.html#index-ftime-report) which might help identify the problem pass

gezalore, 2023-11-02:
Also, you seem to be using Verilator `--output-split 500000`, which probably does not help. Lowering it might speed things up (default value is 20000), but it would still be very valuable to understand why GCC is slow, so maybe we an avoid generating problematic code.

gkamendje, 2023-11-02:
UNOPTFLAT does not show up when I use --lint_only nor when I use `-O3` or `-O0` is this expected? 
> UNOPTFLAT might matter to runtime but shouldn't matter to compile time.
> 
> If -O0 is slower than -Os then I'd say 90% it's a gcc bug. What is clang time?

gkamendje, 2023-11-02:
Any hint on how to directly pass GCC options to Verilator? 
I have never used clang with Verilator before. Any hint on how to do it?  
> Also see GCC [-ftime-report](https://gcc.gnu.org/onlinedocs/gcc/Developer-Options.html#index-ftime-report) which might help identify the problem pass

wsnyder, 2023-11-02:
UNOPTFLAT might go away with some earlier optimizations.

The best route to use clang is to rebuild Verilator with "./configure CXX=clang++"

To pass other compile flags through Verilator use "-CFLAGS -ftime-report"

gkamendje, 2023-11-02:
Here is the output of `-ftime-report`. Keeping `--output-split 500000` for now just for apple to apple comparison.
```
(/usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' /cad/verilator/v5.018-2-g2a57ead7e/bin/verilator --trace -CFLAGS "-ftime-report" --stats -cc --exe --timing --build -j 4 --Mdir /var/tmp/gkamendje/verilator_work_dirs/ltag7_top/rtl/work --timing --report-unoptflat  --output-split 500000 --timescale 1ns/100ps --x-assign unique --x-initial unique  +define+INIT_EEPROM_MEM_IN_TB=1  -O3 -f /home/gkamendje/project.repo/ltag7/ver/sim/../../rtl/man.f -f /home/gkamendje/project.repo/ltag7/ver/sim/../th/man.f --top-module ltag7_tb  /home/gkamendje/project.repo/ltag7/ver/sim/../th/verilator_wrapper.cpp) 2>&1 | tee  /var/tmp/gkamendje/verilator_work_dirs/ltag7_top/rtl/work/ltag7_top_rtl_comp.log 
make[1]: Entering directory `/var/tmp/gkamendje/verilator_work_dirs/ltag7_top/rtl/work'
ccache g++  -I.  -MMD -I/cad/verilator/v5.018-2-g2a57ead7e/share/verilator/include -I/cad/verilator/v5.018-2-g2a57ead7e/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -ftime-report   -fcoroutines -Os -c -o verilator_wrapper.o /home/gkamendje/project.repo/ltag7/ver/sim/../th/verilator_wrapper.cpp
ccache g++ -Os  -I.  -MMD -I/cad/verilator/v5.018-2-g2a57ead7e/share/verilator/include -I/cad/verilator/v5.018-2-g2a57ead7e/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -ftime-report   -fcoroutines -c -o verilated.o /cad/verilator/v5.018-2-g2a57ead7e/share/verilator/include/verilated.cpp
ccache g++ -Os  -I.  -MMD -I/cad/verilator/v5.018-2-g2a57ead7e/share/verilator/include -I/cad/verilator/v5.018-2-g2a57ead7e/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -ftime-report   -fcoroutines -c -o verilated_vcd_c.o /cad/verilator/v5.018-2-g2a57ead7e/share/verilator/include/verilated_vcd_c.cpp
ccache g++ -Os  -I.  -MMD -I/cad/verilator/v5.018-2-g2a57ead7e/share/verilator/include -I/cad/verilator/v5.018-2-g2a57ead7e/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -ftime-report   -fcoroutines -c -o verilated_timing.o /cad/verilator/v5.018-2-g2a57ead7e/share/verilator/include/verilated_timing.cpp

Time variable                                   usr           sys          wall           GGC
 phase setup                        :   0.00 (  0%)   0.00 (  0%)   0.00 (  0%)  1581k (  1%)
 phase parsing                      :   1.18 ( 87%)   0.55 ( 90%)   1.74 ( 88%)    98M ( 86%)
 phase lang. deferred               :   0.14 ( 10%)   0.04 (  7%)   0.17 (  9%)    12M ( 11%)
 phase opt and generate             :   0.03 (  2%)   0.02 (  3%)   0.06 (  3%)  1682k (  1%)
 |name lookup                       :   0.23 ( 17%)   0.07 ( 11%)   0.45 ( 23%)  4078k (  3%)
 |overload resolution               :   0.18 ( 13%)   0.06 ( 10%)   0.10 (  5%)    11M ( 10%)
 dump files                         :   0.01 (  1%)   0.01 (  2%)   0.01 (  1%)     0  (  0%)
 callgraph construction             :   0.00 (  0%)   0.00 (  0%)   0.01 (  1%)   546k (  0%)
 callgraph optimization             :   0.00 (  0%)   0.00 (  0%)   0.01 (  1%)  7128  (  0%)
 callgraph functions expansion      :   0.01 (  1%)   0.01 (  2%)   0.02 (  1%)   281k (  0%)
 callgraph ipa passes               :   0.02 (  1%)   0.00 (  0%)   0.03 (  2%)   614k (  1%)
 preprocessing                      :   0.18 ( 13%)   0.16 ( 26%)   0.30 ( 15%)  2843k (  2%)
 parser (global)                    :   0.23 ( 17%)   0.15 ( 25%)   0.45 ( 23%)    28M ( 25%)
 parser struct body                 :   0.15 ( 11%)   0.04 (  7%)   0.24 ( 12%)    17M ( 15%)
 parser function body               :   0.08 (  6%)   0.03 (  5%)   0.09 (  5%)  3817k (  3%)
 parser inl. func. body             :   0.14 ( 10%)   0.05 (  8%)   0.09 (  5%)  6754k (  6%)
 parser inl. meth. body             :   0.13 ( 10%)   0.04 (  7%)   0.18 (  9%) 10130k (  9%)
 template instantiation             :   0.39 ( 29%)   0.12 ( 20%)   0.55 ( 28%)    41M ( 36%)
 constant expression evaluation     :   0.02 (  1%)   0.00 (  0%)   0.00 (  0%)   248k (  0%)
 inline parameters                  :   0.01 (  1%)   0.00 (  0%)   0.00 (  0%)    95k (  0%)
 tree loop invariant motion         :   0.00 (  0%)   0.01 (  2%)   0.00 (  0%)     0  (  0%)
 dominance computation              :   0.00 (  0%)   0.00 (  0%)   0.01 (  1%)     0  (  0%)
[... cut ...]
 address taken                      :   0.55 (  0%)   0.00 (  0%)   0.53 (  0%)     0  (  0%)
 repair loop structures             :   0.02 (  0%)   0.00 (  0%)   0.01 (  0%)     0  (  0%)
 TOTAL                              :2082.97          6.15       2089.27         2539M
echo "" > Vltag7_tb__ALL.verilator_deplist.tmp
Archive ar -rcs Vltag7_tb__ALL.a Vltag7_tb__ALL.o
g++     verilator_wrapper.o verilated.o verilated_vcd_c.o verilated_timing.o verilated_threads.o Vltag7_tb__ALL.a    -pthread -lpthread -latomic   -o Vltag7_tb
rm Vltag7_tb__ALL.verilator_deplist.tmp

```

gkamendje, 2023-11-02:
Clang fails after roughly 13 mins (which is already quit long...) with the following error message. 
`make[1]: *** [/cad/verilator/v5.018-2-g2a57ead7e_clang/share/verilator/include/verilated.mk:269: Vltag7_tb__ALL.o] Error 255` I am running on Ubuntu 22.04.

Here is the output of `-ftime-report` it has produced so far

```
(/usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' /cad/verilator/v5.018-2-g2a57ead7e_clang/bin/verilator --trace -CFLAGS "-ftime-report" --stats -cc --exe --timing --build -j 8 --Mdir /var/tmp/gkamendje/verilator_work_dirs/ltag7_top/rtl/work --timing --report-unoptflat  --output-split 500000 --timescale 1ns/100ps --x-assign unique --x-initial unique  +define+INIT_EEPROM_MEM_IN_TB=1  -O3 -f /home/gkamendje/project.repo/ltag7/ver/sim/../../rtl/man.f -f /home/gkamendje/project.repo/ltag7/ver/sim/../th/man.f --top-module ltag7_tb  /home/gkamendje/project.repo/ltag7/ver/sim/../th/verilator_wrapper.cpp) 2>&1 | tee  /var/tmp/gkamendje/verilator_work_dirs/ltag7_top/rtl/work/ltag7_top_rtl_comp.log 
make[1]: Entering directory '/var/tmp/gkamendje/verilator_work_dirs/ltag7_top/rtl/work'
ccache clang++  -I.  -MMD -I/cad/verilator/v5.018-2-g2a57ead7e_clang/share/verilator/include -I/cad/verilator/v5.018-2-g2a57ead7e_clang/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-c++11-narrowing -Wno-constant-logical-operand -Wno-non-pod-varargs -Wno-overloaded-virtual -Wno-parentheses-equality -Wno-shadow -Wno-sign-compare -Wno-tautological-bitwise-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -ftime-report   -fcoroutines-ts -Os -c -o verilator_wrapper.o /home/gkamendje/project.repo/ltag7/ver/sim/../th/verilator_wrapper.cpp
ccache clang++ -Os  -I.  -MMD -I/cad/verilator/v5.018-2-g2a57ead7e_clang/share/verilator/include -I/cad/verilator/v5.018-2-g2a57ead7e_clang/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-c++11-narrowing -Wno-constant-logical-operand -Wno-non-pod-varargs -Wno-overloaded-virtual -Wno-parentheses-equality -Wno-shadow -Wno-sign-compare -Wno-tautological-bitwise-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -ftime-report   -fcoroutines-ts -c -o verilated.o /cad/verilator/v5.018-2-g2a57ead7e_clang/share/verilator/include/verilated.cpp
ccache clang++ -Os  -I.  -MMD -I/cad/verilator/v5.018-2-g2a57ead7e_clang/share/verilator/include -I/cad/verilator/v5.018-2-g2a57ead7e_clang/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-c++11-narrowing -Wno-constant-logical-operand -Wno-non-pod-varargs -Wno-overloaded-virtual -Wno-parentheses-equality -Wno-shadow -Wno-sign-compare -Wno-tautological-bitwise-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -ftime-report   -fcoroutines-ts -c -o verilated_vcd_c.o /cad/verilator/v5.018-2-g2a57ead7e_clang/share/verilator/include/verilated_vcd_c.cpp
ccache clang++ -Os  -I.  -MMD -I/cad/verilator/v5.018-2-g2a57ead7e_clang/share/verilator/include -I/cad/verilator/v5.018-2-g2a57ead7e_clang/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-c++11-narrowing -Wno-constant-logical-operand -Wno-non-pod-varargs -Wno-overloaded-virtual -Wno-parentheses-equality -Wno-shadow -Wno-sign-compare -Wno-tautological-bitwise-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -ftime-report   -fcoroutines-ts -c -o verilated_timing.o /cad/verilator/v5.018-2-g2a57ead7e_clang/share/verilator/include/verilated_timing.cpp
ccache clang++ -Os  -I.  -MMD -I/cad/verilator/v5.018-2-g2a57ead7e_clang/share/verilator/include -I/cad/verilator/v5.018-2-g2a57ead7e_clang/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-c++11-narrowing -Wno-constant-logical-operand -Wno-non-pod-varargs -Wno-overloaded-virtual -Wno-parentheses-equality -Wno-shadow -Wno-sign-compare -Wno-tautological-bitwise-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -ftime-report   -fcoroutines-ts -c -o verilated_threads.o /cad/verilator/v5.018-2-g2a57ead7e_clang/share/verilator/include/verilated_threads.cpp
/bin/python3 /cad/verilator/v5.018-2-g2a57ead7e_clang/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vltag7_tb.cpp Vltag7_tb___024root__DepSet_h6d652bbb__0.cpp Vltag7_tb___024root__DepSet_hfbf74690__0.cpp Vltag7_tb__Trace__0.cpp Vltag7_tb__ConstPool_0.cpp Vltag7_tb___024root__Slow.cpp Vltag7_tb___024root__DepSet_h6d652bbb__0__Slow.cpp Vltag7_tb___024root__DepSet_hfbf74690__0__Slow.c [... cut ...] k --timing --report-unoptflat --output-split 500000 --timescale 1ns/100ps --x-assign unique --x-initial unique +define+INIT_EEPROM_MEM_IN_TB=1 -O3 -f /home/gkamendje/project.repo/ltag7/ver/sim/../../rtl/man.f -f /home/gkamendje/project.repo/ltag7/ver/sim/../th/man.f --top-module ltag7_tb /home/gkamendje/project.repo/ltag7/ver/sim/../th/verilator_wrapper.cpp
Command exited with non-zero status 2
Elapsed time: 12:50.98[h:]min:sec. CPU time: user 771.24 sys 7.05 (100%). Peak memory: 12563296KB.
```

gkamendje, 2023-11-02:
I was able to isolate the code that is causing the long compiler run time: 

I have the following two tasks in my testbench that simply wait for a number of clock cycles

```
parameter shortint unsigned               W_DATA                        = 32; 
parameter int unsigned                      X_WORD_ADDR          = 5; 
parameter                  [5:0]                  C_RX_DATA_LEN       = 45; 

task sendZero;
    begin
       i2tag =0;
       repeat (`BIT_PERIOD_CC/2) @(posedge sys_clk); //
       i2tag =1;
       repeat (`BIT_PERIOD_CC/2) @(posedge sys_clk); //
       i2tag =0;
    end
endtask:sendZero 

task sendOne;
    begin
       repeat (`BIT_PERIOD_CC) @(posedge sys_clk); 
    end
endtask:sendOne 
```
I also have the following helper fonction to calculate parity. 
```
//------------------------------------------
// Calc Parity
//------------------------------------------
function logic [44:0] calc_parity ( input [W_DATA-1:0] i_data );
    logic [44:0] o_data;
    int one_cnt; 
    int offset;
        one_cnt =0;
        offset=0;
       $display("%t\t <<<<<<<----- start calculating parity on data %0h ----->>>>>",$realtime, i_data);
        for (int i =0 ; i < W_DATA ; i++) begin
            if (i_data[i] == 1'b1 ) begin
                one_cnt = one_cnt+1;
                o_data[i+offset] =1'b1;
            end else begin
                o_data[i+offset] =1'b0;
            end
            if ( ((i+1) % 8) == 0 ) begin // entire byte so far, now send the bit parity
                offset = (i+1)/8;
                if ( (one_cnt % 2) == 1) begin
                    o_data[i+offset] =1'b1;
                end else begin
                    o_data[i+offset] =1'b0;
                end
                one_cnt =0;
            end
        end
        // send 8 column parity bits
        if ((i_data[24] ^ i_data[16] ^ i_data[8]  ^ i_data[0] ) == 1'b1) begin o_data[36] =1'b1;; end else begin o_data[36] =1'b0;; end
        if ((i_data[25] ^ i_data[17] ^ i_data[9]  ^ i_data[1] ) == 1'b1) begin o_data[37] =1'b1;; end else begin o_data[37] =1'b0;; end
        if ((i_data[26] ^ i_data[18] ^ i_data[10] ^ i_data[2] ) == 1'b1) begin o_data[38] =1'b1;; end else begin o_data[38] =1'b0;; end
        if ((i_data[27] ^ i_data[19] ^ i_data[11] ^ i_data[3] ) == 1'b1) begin o_data[39] =1'b1;; end else begin o_data[39] =1'b0;; end
        if ((i_data[28] ^ i_data[20] ^ i_data[12] ^ i_data[4] ) == 1'b1) begin o_data[40] =1'b1;; end else begin o_data[40] =1'b0;; end
        if ((i_data[29] ^ i_data[21] ^ i_data[13] ^ i_data[5] ) == 1'b1) begin o_data[41] =1'b1;; end else begin o_data[41] =1'b0;; end
        if ((i_data[30] ^ i_data[22] ^ i_data[14] ^ i_data[6] ) == 1'b1) begin o_data[42] =1'b1;; end else begin o_data[42] =1'b0;; end
        if ((i_data[31] ^ i_data[23] ^ i_data[15] ^ i_data[7] ) == 1'b1) begin o_data[43] =1'b1;; end else begin o_data[43] =1'b0;; end
        //sendZero();
        o_data[44] =1'b0;
        $display("%t\t <<<<<<<----- computed parity word data %0h completed ----->>>>>",$realtime, o_data);
        return o_data;
endfunction :calc_parity 
```

Then I have two other task that loops on a constant and either calls the first or the second task depending on the value of bits in a vector. 
```
task senddata( input [W_DATA-1:0] i_data );
    logic [C_RX_DATA_LEN:0] em_data_and_parity;
    begin
       em_data_and_parity=calc_parity(i_data);
       $display("%t\t <<<<<<<----- start sending data %0h with computed parity %0h ----->>>>>",$realtime, i_data, em_data_and_parity);
       for (int i =0 ; i <= C_RX_DATA_LEN  ; i++) begin // Send MSB first
           if (em_data_and_parity[i] == 1'b1 ) begin
               sendOne();
           end else begin
               sendZero();
           end
       end
        $display("%t\t <<<<<<<----- sending data %0h completed ----->>>>>",$realtime, i_data);
    end
endtask :senddata 
```
And 
```
task sendAddr( input [X_WORD_ADDR-1:0] i_addr );
    begin
       logic  parity;
       $display("%t\t <<<<<<<----- sending addr %0h transaction ----->>>>>",$realtime, i_addr);
       parity =^i_addr; 
       for (int i =0 ; i < X_WORD_ADDR  ; i++) begin // Send MSB first
           if (i_addr[i] == 1'b1 ) begin
               sendOne();
           end else begin
               sendZero();
           end
       end
       sendZero();
       sendZero();
       if (parity == 1'b1 ) begin
           sendOne();
       end else begin
           sendZero();
       end
       $display("%t\t <<<<<<<----- sending addr %0h transaction completed ----->>>>>",$realtime, i_addr);
    end
endtask :sendAddr 
```
Build time drops to 30s (compared to ~40mins) when the two for loops are commented out.

gezalore, 2023-11-02:
So my conjecture is that the control flow complexity of the GCC input is quite extreme (i.e.: lot of `if`/`else`):
```
tree SSA incremental               :1407.79 ( 68%)   0.27 (  4%)1408.15 ( 67%)    41M (  2%)
```
This is likely due to your for loops (which contain `if` statements) being unrolled. Could you try Verilator `--unroll-count 4` (default is 64) to see if it fixes your compile times? Warning: if you have NBAs in other loops, those might be reported as unsupported if not unrolled, so you might have to manually unroll those.

gkamendje, 2023-11-02:
@wsnyder I have created a test case along with driver and golden file in `test_regress/t/t_buggy_loop.*`. 
When I try to run it with 
`./test_regress/t/t_buggy_for_loop.pl `
I get the following error message. 
```
./test_regress/t/t_buggy_for_loop.pl
======================================================================
dist/t_buggy_for_loop: ==================================================
-Skip: dist/t_buggy_for_loop: scenario 'dist' not enabled for test
dist/t_buggy_for_loop: -Skip: Skip: scenario 'dist' not enabled for test
==SUMMARY: Passed 0  Failed 0  Time 0:00
======================================================================
vlt/t_buggy_for_loop: ==================================================
        perl /home/gkamendje/tmp_compile/verilator_myfork/test_regress/../bin/verilator --prefix Vt_buggy_for_loop --make gmake --x-assign unique -cc -Mdir obj_vlt/t_buggy_for_loop --fdedup --debug-check --comp-limit-members 10 --exe --main --timing -Wno-UNOPTFLAT --clk clk  -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_buggy_for_loop +define+TEST_DUMPFILE=obj_vlt/t_buggy_for_loop/simx.vcd t/t_buggy_for_loop.v    > obj_vlt/t_buggy_for_loop/vlt_compile.log
        make -C obj_vlt/t_buggy_for_loop -f /home/gkamendje/tmp_compile/verilator_myfork/test_regress/Makefile_obj --no-print-directory VM_PREFIX=Vt_buggy_for_loop TEST_OBJ_DIR=obj_vlt/t_buggy_for_loop CPPFLAGS_DRIVER=-DT_BUGGY_FOR_LOOP  OPT_FAST=-O0 OPT_GLOBAL=-O0 Vt_buggy_for_loop    > obj_vlt/t_buggy_for_loop/vlt_gcc.log
driver: Entering directory '/home/gkamendje/tmp_compile/verilator_myfork/test_regress/obj_vlt/t_buggy_for_loop'
ccache g++ -O0  -I.  -MMD -I/home/gkamendje/tmp_compile/verilator_myfork/test_regress/../include -I/home/gkamendje/tmp_compile/verilator_myfork/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_TIME_CONTEXT  @CFG_CXXFLAGS_STD@ -fcoroutines -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_buggy_for_loop -DVM_PREFIX=Vt_buggy_for_loop -DVM_PREFIX_INCLUDE="<Vt_buggy_for_loop.h>" -DVM_PREFIX_ROOT_INCLUDE="<Vt_buggy_for_loop___024root.h>" -DT_BUGGY_FOR_LOOP   -DVL_LOCK_SPINS=10000 -c -o verilated.o /home/gkamendje/tmp_compile/verilator_myfork/test_regress/../include/verilated.cpp
g++: warning: @CFG_CXXFLAGS_STD@: linker input file unused because linking not done
g++: error: @CFG_CXXFLAGS_STD@: linker input file not found: No such file or directory
make: *** [verilated.o] Error 1

```
Here is the content of my driver file 

```
scenarios(simulator => 1);

compile(
    verilator_flags2 => ["--exe --main --timing -Wno-UNOPTFLAT"],
    make_main => 0,
    );

execute(
    check_finished => 1,
    expect_filename => $Self->{golden_filename},
    );

    #compile(
    #    verilator_flags2 => ["--exe --main --timing -Wno-UNOPTFLAT -fno-localize"],
    #    make_main => 0,
    #    );
    #
    #execute(
    #    check_finished => 1,
    #    #expect_filename => $Self->{golden_filename},
    #    );

ok(1);
1;

```
Any idea what I am missing here?

gezalore, 2023-11-02:
You seem to be passing the string `@CFG_CXXFLAGS_STD@` as argument to GCC, but that should have been replaced by `configure` when you built verilator, something seems to have gone astray when you run `configure` to build Verilator.

gkamendje, 2023-11-02:
@gezalore 
Indeed using `--unroll-count 4` brings the runtime from ~40mins down to 6s which is more inline with what I would have expected. Thanks for the hint.
> So my conjecture is that the control flow complexity of the GCC input is quite extreme (i.e.: lot of `if`/`else`):
>
> ```
> tree SSA incremental               :1407.79 ( 68%)   0.27 (  4%)1408.15 ( [...]

gezalore, 2023-11-02:
@wsnyder This makes me wonder if we should only unroll loops that contain NBAs. NBAs to variable LHS are "unsuppored" (V3Delayed), which I think it the primary reason we tend to unroll loops. Generally it bloats code. (Unrolling also might enable more block splitting later, but that is an optimization, not a functional issue)

gkamendje, 2023-11-02:
Created PR #4662 to add test case for this issue.

wsnyder, 2023-11-02:
>ipa cp                             : 443.09 ( 21%)   0.14 (  2%) 443.30 ( 21%)    26M (  1%)
>tree SSA incremental               :1407.79 ( 68%)   0.27 (  4%)1408.15 ( 67%)    41M (  2%)

Well, here we obviously have a GCC performance bug.  Please make a new bug (so you can see the resolution eventually).  Then, would you like to submit this to them, or shall we?

>Clang fails after roughly 13 mins (which is already quit long...) with the following error message.
>make[1]: *** [/cad/verilator/v5.018-2-g2a57ead7e_clang/share/verilator/include/verilated.mk:269: Vltag7_tb__ALL.o] Error 255

And as I've usually seen, where GCC fails Clang likewise fails.  Likewise need to submit to clang.

As @gezalore mentioned, the unroll count can help this, but needs to be applied correctly.

>@wsnyder This makes me wonder if we should only unroll loops that contain NBAs. NBAs to variable LHS are "unsuppored" (V3Delayed), which I think is the primary reason we unroll loops. Generally it bloats code. (U$

We also aggressively unroll for another reason that is shown in this file - people write e.g. parity trees using for loops rather than logical-XOR reduction.  Some tools also generate such loops, and without unrolling these can greatly harm performance, so I don't think we want to generally disable unrolling.

Ideally we'd allow per-loop tuning with a pragma, but I think here we just want to not unroll when there's a clock control construct.  I suggest we use this bug to indicate that fix.

For this specific case, I think we could also unroll, but be smarter about optimizing the resulting code, but I'm not suggesting that for today.

gkamendje, 2023-11-02:
@wsnyder please go ahead and submit the bug to GCC & clang

wsnyder, 2023-11-03:
The example in #4662 takes only 30 seconds with my system, is this the same case that for you takes many minutes?

gkamendje, 2023-11-10:
Unfortunately, I was not able to fully reproduce the issue (30mins or more runtime) in the example (although, one could say that 30 seconds for such a tiny example is quite long). In my testbench, with `--unroll-count 64 `, commenting out the following section
[image] 
lines 81-87 in the example Verilog, brings the compilation time from ~34 mins down to ~1 min. It could be that there are multiple things going on here.

wsnyder, 2024-02-14:
I don't think we got a reproducing example to debug, if you have one please post & reopen.

edaincloud, 2024-10-31:
Here is an example where a mix of blocking and non-blocking assignments resulted in a long build phase (14 minutes). However, after refactoring the code to use only non-blocking assignments, the build phase was reduced to 30 seconds.

Code snippet with mix of blocking and non-blocking assignment : - 

  always_ff @(posedge clk) begin
     if (rstn) begin  // Active high reset
    // Reset condition
    q16_size <= 0;
    data_out <= 0; 
  end else begin
    if (data_in_vld && data_in_rdy) begin
      q16.push_back(data_in[31:16]);
      q16.push_back(data_in[15:0]);
      q16_size += 2;
    end
    if (data_out_vld && data_out_rdy) begin
      void'(q16.pop_front());
      q16_size -= 1;
    end
    data_out = (q16_size > 0) ? q16[0]:'0;
  end
 end

Code snippet with just non-blocking assignment : - 

  always_ff @(posedge clk) begin
     if (rstn) begin  // Active high reset
    // Reset condition
    q16_size <= 0;
    data_out <= 0; 
  end else begin
    if (data_in_vld && data_in_rdy) begin
      q16.push_back(data_in[31:16]);
      q16.push_back(data_in[15:0]);
      q16_size <= q16_size + 2;
    end
    if (data_out_vld && data_out_rdy) begin
      void'(q16.pop_front());
      q16_size <= q16_size -1;
    end
    data_out <= (q16_size > 0) ? q16[0]:'0;
  end
  end

Complete design code:- 
module fifo(
  input clk,
  input rstn,
  // data_in
  input [31:0] data_in,
  input        data_in_vld,
  output       data_in_rdy,
  // data_out
  output logic [15:0] data_out,
  output        data_out_vld,
  input         data_out_rdy,
 // New signals
  output logic  fifo_full,
  output logic  fifo_empty
);

  localparam FIFO16_SIZE = 25;
  bit [15:0] q16[$];
  int q16_size;

//   assign data_in_rdy = q16_size < (FIFO16_SIZE +1) ;
  assign data_in_rdy = q16_size <= (FIFO16_SIZE - 1);
  assign data_out_vld = q16_size > 0;
  
  // New assignments for fifo_full and fifo_empty
  //assign fifo_full = (q16_size == (FIFO16_SIZE -1));
  assign fifo_full = (q16_size >= (FIFO16_SIZE - 1));
  assign fifo_empty = (q16_size == 0);

  always_ff @(posedge clk) begin
     if (rstn) begin  // Active high reset
    // Reset condition
    q16_size <= 0;
    data_out <= 0; 
  end else begin
    if (data_in_vld && data_in_rdy) begin
      q16.push_back(data_in[31:16]);
      q16.push_back(data_in[15:0]);
      q16_size += 2;
    end
    if (data_out_vld && data_out_rdy) begin
      void'(q16.pop_front());
      q16_size -= 1;
    end
    data_out = (q16_size > 0) ? q16[0]:'0;
  end
  end
endmodule

This module is part of big design with UVM in verilator, so just sharing the module level source code. 

Verilator version: Verilator 5.029 devel

wsnyder, 2024-11-02:
The code you sent runs fast - which I assume is expected as this is a smaller example. Comparing the code examples' output code I don't see any obvious reason one would be slower than the other.  Maybe see if -fno-inline makes a difference?   We'll still need a full test case that shows the slowdown to debug further.

closed 2024-11-02 by wsnyder