verilator/verilator #4658: [Bug report] Segmentation fault with many parallel adders

flaviens, 2023-11-02:
Hi there!

I found what seems to be a bug related to having many adders in parallel, causing a segmentation fault during the first eval.

Here is the extremely basic testbench.
```
#include "Vtop.h"
#include "verilated.h"

#include <iostream>

typedef Vtop Module;

int main(int argc, char **argv, char **env) {

  Verilated::commandArgs(argc, argv);
  Verilated::traceEverOn(0);

  Module *my_module = new Module;
  my_module->eval();

  std::cout << "Testbench complete!" << std::endl;
  delete my_module;
  exit(0);
}
```

[Here](https://github.com/verilator/verilator/files/13238823/top.zip) is the source file.
If you remove any of the assignments, the segmentation fault will disappear.

The command that I used is
```
verilator --cc --exe --build ../tb.cc top.sv
```

I'm using `Verilator 5.018 2023-10-30 rev v5.004-74-g41e5f6b7d`. I also got the same with `v5.016`, and working with Ubuntu 22.04 and `g++ 9.4.0`.

May you have any difficulty to reproduce, I can produce a Docker image to help.

Thanks!
Flavien

wsnyder, 2023-11-02:
When I run this on Ubuntu using the latest version I don't get a crash.  Please compile with debug and use GDB to get a backtrace of where it is failing for you, and ideally debug it backwards from there, thanks.

flaviens, 2023-11-03:
Thank you very much @wsnyder for looking into it so quickly! The bug is indeed somehow evasive, so I created a Docker image to help you reproducing.

[The repo to build it is here](https://github.com/flaviens/verilator-b1), and [the image is here](https://hub.docker.com/r/ethcomsec/verilator-b1) if you would like to pull it.

Here are example steps to reproduce:
1. Pull the Docker image:
```
docker pull docker.io/ethcomsec/verilator-b1
```
2. Run a Docker container:
```
docker run -it docker.io/ethcomsec/verilator-b1
```
3. Run the simulation in a loop until it segfaults
```
root@5fd46be4c798:/# bash ./test_loop.sh
Testbench complete!
Testbench complete!
Testbench complete!
Testbench complete!
Testbench complete!
Testbench complete!
Testbench complete!
Testbench complete!
Testbench complete!
Testbench complete!
Testbench complete!
Testbench complete!
Testbench complete!
Testbench complete!
Testbench complete!
Testbench complete!
Testbench complete!
Testbench complete!
./test_loop.sh: line 9: 19476 Segmentation fault      (core dumped) ./obj_dir/Vtop
Segmentation fault detected.
```
Please note that the number of successful completions before a segfault is not always the same, but is always in the order of magnitude of 10.

Please let me know if you need additional help for reproducing.
Thank you for your help!
Flavien

wsnyder, 2023-12-20:
Note I unfortunately am unlikely to have the time to debug docker or other complicated tests, I would suggest tracking this down yourself, or creating a test that is in test_regress format that shows the issue using Ubuntu directly.

wsnyder, 2024-01-18:
As with other issues, we need a test_regress format test without any other complicated testbench to debug.  If you update to have this please reopen.

closed 2024-01-18 by wsnyder