verilator/verilator #4659: Question about modules level and elaboration order

donlon, 2023-11-02:
After all files are loaded, V3LinkLevel::modSortByLevel determines modules' topological level and then sort modules with it. Having searched the references of AstModule::level() I found that the information is only used to

1) Determine top level module(s) and check whether a module is the top level

2) Determine if a module is applied for --public-depth and --trace-depth.

3) Perform deparam in the level order (rather than hierarchical order). so every cell referencing to a parameterized module can be processed before the module, to avoid cloning the modified module.

For 1), to determine top modules it does requires leveling. And for 2), using module depth causes some imperfects (both A.B.X and A.X are applied if module X has matched depth). Instead I think it should use Scope depth.

For 3) I guess it was used to guarentee that cloning the original module is always before its modification (constant folding & deparam cells), so we can reuse the original module to save another copy. However after #4629 it's not possible to reuse the original module since it's not possible to evaluate the default values unless doing that in a cloned module, which doesn't make sense in avoiding another copy of the module.

What's more, it makes it impossible to use some constant expressions during elaboration. E.g.

- In #1593 it failed to use parameters or signal width as elaboration constants.
It's because the module referenced by a cell always has higher level than the module containing the cell, so the latter module is always processed after the former module. Thus V3Width will throw `Parameter-resolved constants must not use dotted references` error because the cross-referenced signal is not width'ed.
But if we perform V3Param and V3Width in hierarchical order, the xref expressions can be width'ed before constant folding, thus make it possible to referene it during elaboration.

- `localparam V = ClsParam#(...)::A`
It's similar to the `#1593` case. The expression is processed before ClsParam so it fails.

Besides I noticed that there're "Vrcm" cells cloned for recursive modules and it's [said to](https://github.com/verilator/verilator/blob/7ba6647c4f30a4c5198289fb26b10b4c48a4ccdf/src/V3LinkCells.cpp#L310-L312) avoid breaking parameter resolution. I haven't found why it breaks it but I think with hierarchical order it's easier to handle such cases (not necessary to clone a new module).

So based on these I propose to deprecate the level-based elaboration (but keep some level information) so we can handle more valid cases. It might be a temporarily workaround for arbitrary parameter resolution but I think it's closer to #1540.

wsnyder, 2023-11-03:
Improving how --public-depth and --trace-depth determine depth is reasonable, though I think by the time V3Scope has been done and scopes determined it may be too late for these to be applied and do what they need to.  (Not sure.)

level() is also used to sort the modules, and likely there are assumptions hidden in visitors that the modules are in levelized order.  An easy experiment is to reverse the sort and see how much breaks.

donlon, 2023-11-08:
> An easy experiment is to reverse the sort and see how much breaks.

Good idea! I tried to reverse the order and it does break (most of them are related to the link stage). But I realized that it's not neccessary to keep the original order of the modules in AST. So I then just change V3Param's elaboration order and fortunately it breaks nothing, and also fixed something unsupported before (e.g. [t_interface_param_acc_bits.out](https://github.com/verilator/verilator/blob/9f75068059c7c263d7fd1c53b8297056b7b3dc29/test_regress/t/t_interface_param_acc_bits.out)). Maybe I can start to prepare for the pull.

But unfortunately it requires #4629 so it still need some time...

wsnyder, 2023-12-24:
Think answered, but feel free to follow up.

closed 2023-12-24 by wsnyder