verilator/verilator #4660: Internal error in newer version of Verilator

davidharrishmc, 2023-11-02:
Verilator 5.019 devel rev v5.018-1-g839315b27 is crashing with an Internal Error on SystemVerilog code that works fine in Verilator 5.011 devel rev v5.010-114-g4b7b185d0 and earlier releases that I had been using until recently. I am running on Ubuntu 20.04.6 LTS. 

See below for lzc.sv that triggers the failure, and for a debug trace of the failure.

This leading zero counter is a bit unusual in coding style because it is parameterized, but simulates properly with Questa and synthesizes properly with Design Compiler.

David Harris

```
module lzc #(parameter WIDTH = 1) (
  input  logic [WIDTH-1:0]            num,    // number to count the leading zeroes of
  output logic [$clog2(WIDTH+1)-1:0]  ZeroCnt // the number of leading zeroes
);

  integer i;
  
  always_comb begin
    i = 0;
    while ((i < WIDTH) & ~num[WIDTH-1-i]) i = i+1;  // search for leading one
    ZeroCnt = i[$clog2(WIDTH+1)-1:0];
  end
endmodule

harris@vlsi:~/cvw/sim$ verilator --lint-only ../src/generic/lzc.sv --debug
Starting Verilator 5.019 devel rev v5.018-1-g839315b27
- Verilator.cpp:614:  Option --verilate: Start Verilation
- V3Os.cpp:108:       export VERILATOR_ROOT=/usr/local/share/verilator # Hardcoded at build time
- V3ParseImp.cpp:292: parseFile: verilated_std
  Preprocessing /usr/local/share/verilator/include/verilated_std.sv
- V3PreShell.cpp:143:     Reading /usr/local/share/verilator/include/verilated_std.sv
- V3ParseImp.cpp:371: Lexing /usr/local/share/verilator/include/verilated_std.sv
- V3ParseImp.cpp:292: parseFile: lzc
  Preprocessing ../src/generic/lzc.sv
- V3PreShell.cpp:143:     Reading ../src/generic/lzc.sv
- V3ParseImp.cpp:357: Writing all preprocessed output to obj_dir/Vlzc__inputs.vpp
- V3ParseImp.cpp:371: Lexing ../src/generic/lzc.sv
- V3LinkLevel.cpp:42: modSortByLevel()
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_001_cells.tree
- V3LinkDot.cpp:3784: linkDotPrimary: 
- V3LinkJump.cpp:349: linkJump: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_006_linkjump.tree
- V3LinkInc.cpp:307:  linkIncrements: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_007_linkinc.tree
- V3Param.cpp:1450:   param: 
- V3LinkDot.cpp:3790: linkDotParamed: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_009_linkdotparam.tree
- V3Dead.cpp:538:     deadifyModules: 
- V3Width.cpp:7527:   width: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_012_width.tree
- V3WidthCommit.cpp:233:widthCommit: 
- V3Const.cpp:3906:   constifyAllLive: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_014_const.tree
- V3Undriven.cpp:540: undrivenAll: 
- V3AssertPre.cpp:494:assertPreAll: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_015_assertpre.tree
- V3Assert.cpp:542:   assertAll: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_016_assert.tree
- V3LinkLevel.cpp:144:wrapTop: 
- V3Const.cpp:3873:   constifyAllLint: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_018_const.tree
- V3SplitVar.cpp:1239:splitVariable: 
- V3Inst.cpp:619:     dearrayAll: 
- V3LinkDot.cpp:3796: linkDotArrayed: 
- V3Fork.cpp:645:     makeDynamicScopes: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_023_fork_dynscope.tree
- V3Fork.cpp:651:     makeTasks: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_024_fork.tree
- V3Begin.cpp:365:    debeginAll: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_025_begin.tree
- V3Tristate.cpp:1815:tristateAll: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_026_tristate.tree
- V3Unknown.cpp:527:  unknownAll: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_027_unknown.tree
- V3DfgOptimizer.cpp:233:extract: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_028_dfg-extract.tree
- V3DfgOptimizer.cpp:240:optimize: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_029_dfg-optimize.tree
- V3Inline.cpp:622:   inlineAll: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_030_inline.tree
- V3LinkDot.cpp:3796: linkDotArrayed: 
- V3DfgOptimizer.cpp:240:optimize: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_032_dfg-optimize.tree
- V3Const.cpp:3916:   constifyAll: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_033_const.tree
- V3Dead.cpp:546:     deadifyDTypes: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_034_deadDtypes.tree
- V3Inst.cpp:613:     instAll: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_035_inst.tree
- V3Const.cpp:3916:   constifyAll: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_036_const.tree
- V3Scope.cpp:416:    scopeAll: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_037_scope.tree
- V3LinkDot.cpp:3802: linkDotScope: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_038_linkdot.tree
- V3Class.cpp:242:    classAll: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_039_class.tree
- V3Const.cpp:3916:   constifyAll: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_040_const.tree
- V3Dead.cpp:552:     deadifyDTypesScoped: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_041_deadDtypesScoped.tree
- V3Case.cpp:605:     caseAll: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_042_case.tree
- V3Task.cpp:1992:    taskAll: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_043_task.tree
- V3Name.cpp:143:     nameAll: 
- V3Unroll.cpp:511:   unrollAll: 
%Error: Internal Error: ../src/generic/lzc.sv:35:30: ../V3Simulate.h:313: No value found for node.
-node: SEL 0x555556f3de00 <e1714> {d35be} u1=0x555556f06520 @dt=0x555556f0c840@(G/w1) decl[0:0]]
   35 |     while ((i < WIDTH) & ~num[WIDTH-1-i]) i = i+1;   
      |                              ^
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
- V3Ast.cpp:1281:     Dumping obj_dir/Vlzc_990_final.tree
- V3StatsReport.cpp:222:statsReport: 
%Error: Internal Error: Aborting since under --debug
%Error: export VERILATOR_ROOT=
%Error: ulimit -s unlimited 2>/dev/null; exec setarch --addr-no-randomize /usr/local/bin/verilator_bin_dbg --lint-only ../src/generic/lzc.sv --debug
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec setarch --addr-no-randomize /usr/local/bin/verilator_bin_dbg --lint-only ../src/generic/lzc.sv --debug
```

gezalore, 2023-11-02:
git bisect says 45edcbb03ed9e0820d1b483a6c6199b98f81ae48, which is @RRozak, but I have not looked further.

closed 2023-11-11 by wsnyder

solomatnikov, 2023-12-22:
This still fails with internal error @ at 8e1c0b085 on https://github.com/openhwgroup/cvw/commit/b701cbf82364b6e56a1f28fc6484523cb56ffa25:

```
~/misc/cvw/sim> ./verilate 
rv64gc simulating...
%Error: Internal Error: ../src/generic/lzc.sv:35:30: ../V3Simulate.h:313: No value found for node.
   35 |     while ((i < WIDTH) & ~num[WIDTH-1-i]) i = i+1;   
      |                              ^
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
Exiting after rv64gc lint due to errors or warnings
```

Unfortunately, no other easy way to re-produce it.

wsnyder, 2023-12-22:
@solomatnikov please file a new bug.  You can run verilator with --debug and use the obj_dir/*__inputs.vpp to make an example that doesn't need a complicated framework.

davidharrishmc, 2023-12-24:
I've pulled and rebuilt the latest Verilator today, and the fix does not seem to have made it into the master branch yet. I'm getting the same error as when I initially filed the issue.  Perhaps that's what @solomatnikov is seeing as well.  When should we expect the fix in the master branch, or some other branch we can test?

Thank you,

David Harris

harris@chips:~/cvw/sim$ verilator --lint-only ../src/generic/lzc.sv --debug
Starting Verilator 5.019 devel rev v5.018-100-g540cd772e
- Verilator.cpp:616:  Option --verilate: Start Verilation
- V3Os.cpp:111:       export VERILATOR_ROOT=/usr/local/share/verilator # Hardcoded at build time
- V3ParseImp.cpp:292: parseFile: verilated_std
  Preprocessing /usr/local/share/verilator/include/verilated_std.sv
- V3PreShell.cpp:143:     Reading /usr/local/share/verilator/include/verilated_std.sv
- V3ParseImp.cpp:371: Lexing /usr/local/share/verilator/include/verilated_std.sv
- V3ParseImp.cpp:292: parseFile: lzc
  Preprocessing ../src/generic/lzc.sv
- V3PreShell.cpp:143:     Reading ../src/generic/lzc.sv
- V3ParseImp.cpp:357: Writing all preprocessed output to obj_dir/Vlzc__inputs.vpp
- V3ParseImp.cpp:371: Lexing ../src/generic/lzc.sv
- V3LinkLevel.cpp:42: modSortByLevel()
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_001_cells.tree
- V3LinkDot.cpp:3830: linkDotPrimary: 
- V3LinkJump.cpp:349: linkJump: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_006_linkjump.tree
- V3LinkInc.cpp:306:  linkIncrements: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_007_linkinc.tree
- V3Param.cpp:1480:   param: 
- V3LinkDot.cpp:3836: linkDotParamed: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_009_linkdotparam.tree
- V3Dead.cpp:542:     deadifyModules: 
- V3Width.cpp:7614:   width: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_012_width.tree
- V3WidthCommit.cpp:233:widthCommit: 
- V3Const.cpp:3904:   constifyAllLive: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_014_const.tree
- V3Undriven.cpp:539: undrivenAll: 
- V3AssertPre.cpp:494:assertPreAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_015_assertpre.tree
- V3Assert.cpp:541:   assertAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_016_assert.tree
- V3LinkLevel.cpp:144:wrapTop: 
- V3Const.cpp:3871:   constifyAllLint: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_018_const.tree
- V3SplitVar.cpp:1239:splitVariable: 
- V3Inst.cpp:617:     dearrayAll: 
- V3LinkDot.cpp:3842: linkDotArrayed: 
- V3Begin.cpp:363:    debeginAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_023_begin.tree
- V3Tristate.cpp:1815:tristateAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_024_tristate.tree
- V3Unknown.cpp:526:  unknownAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_025_unknown.tree
- V3DfgOptimizer.cpp:233:extract: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_026_dfg-extract.tree
- V3DfgOptimizer.cpp:240:optimize: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_027_dfg-optimize.tree
- V3Inline.cpp:622:   inlineAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_028_inline.tree
- V3LinkDot.cpp:3842: linkDotArrayed: 
- V3DfgOptimizer.cpp:240:optimize: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_030_dfg-optimize.tree
- V3Const.cpp:3914:   constifyAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_031_const.tree
- V3Dead.cpp:550:     deadifyDTypes: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_032_deadDtypes.tree
- V3Inst.cpp:611:     instAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_033_inst.tree
- V3Const.cpp:3914:   constifyAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_034_const.tree
- V3Scope.cpp:414:    scopeAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_035_scope.tree
- V3LinkDot.cpp:3848: linkDotScope: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_036_linkdot.tree
- V3Class.cpp:241:    classAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_037_class.tree
- V3Const.cpp:3914:   constifyAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_038_const.tree
- V3Dead.cpp:556:     deadifyDTypesScoped: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_039_deadDtypesScoped.tree
- V3Case.cpp:603:     caseAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_040_case.tree
- V3Task.cpp:1980:    taskAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_041_task.tree
- V3Name.cpp:142:     nameAll: 
- V3Unroll.cpp:510:   unrollAll: 
%Error: Internal Error: ../src/generic/lzc.sv:35:30: ../V3Simulate.h:313: No value found for node.
-node: SEL 0x555556f73e00 <e1711> {d35be} u1=0x555556f44520 @dt=0x555556f4a840@(G/w1) decl[0:0]]
   35 |     while ((i < WIDTH) & ~num[WIDTH-1-i]) i = i+1;   
      |                              ^
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
- V3Ast.cpp:1280:     Dumping obj_dir/Vlzc_990_final.tree
- V3StatsReport.cpp:219:statsReport: 
%Error: Internal Error: Aborting since under --debug
%Error: export VERILATOR_ROOT=
%Error: ulimit -s unlimited 2>/dev/null; exec setarch --addr-no-randomize /usr/local/bin/verilator_bin_dbg --lint-only ../src/generic/lzc.sv --debug
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec setarch --addr-no-randomize /usr/local/bin/verilator_bin_dbg --lint-only ../src/generic/lzc.sv --debug

wsnyder, 2023-12-24:
Please try latest master.

davidharrishmc, 2023-12-24:
That fixed it for me.  Thank you for the speedy fix!