verilator/verilator #4661: FST regressions

AndrewNolte, 2023-11-02:
Working on an example, but recent observations since the new release:
- Fst enums no longer show the enum name. Not sure if this is for all enums or just specific edge cases
- FST/VCD files are put under one top level now. I don't think this is a huge deal but breaks .gtkw files

gezalore, 2023-11-02:
First one is probably mine, I'm not sure what you mean by the second. Examples would be welcome.

gezalore, 2023-11-02:
FWIW the tests that contain enum names are still good, so it seems like we don't cover your case.

wsnyder, 2023-11-04:
As @gezalore noted can you please modify the test_regress suite to show the problem (e.g. pass on old rev, fail now).  Extra thanks as always for a pull request that also fixes it ;)

AndrewNolte, 2023-12-05:
Hi sorry for the delay, I was finally able to get around to making a reproducer. Of course, it has to do with interfaces :) https://github.com/verilator/verilator/pull/4740

wsnyder, 2023-12-10:
Moving comments from #4740 here:

**@wsnyder Commented 5 days ago:**

```
$var logic 2 @ state [1:0] $end
$upscope $end
$attrbegin misc 07 $unit::state_t 4 VAL_A VAL_B VAL_C VAL_D 00 01 10 11 3 $end
$attrbegin misc 07 "" 3 $end
```

Please file a bug with sending the fst to the GTKwave team, as this output suggests that Verilator is properly creating an enum, and GTKwave is confused because the hierarchy of the enum happens to be different from that of the signal using it. It's possible Verilator is still doing so somehow incorrectly, but the GTKwave team will need to tell us what.

I'll merge in this test once it passes, as no harm in recording the test. You'll need to mark it as non-draft and fix any test failures first please.

**@AndrewNolte
 Commented 4 days ago**

I don't think it's a gtkwave issue, bc it was working fine before the fst rewrite

wsnyder, 2023-12-10:
@AndrewNolte can you finish off pull request #4740 to please pass (with the wrong result) so we can merge it?

I still do not see what is wrong with the output - can you also post it to GTK to ask why the enum shows up?  I understand it broke with the refactoring but the output while different still appears legal AFAIK.

toddstrader, 2023-12-13:
The enum problem is that this is called:
```
        tracep->declDTypeEnum(1, "$unit::state_t", 4, 2, __VenumItemNames, __VenumItemValues);
```
before this:
```
    Vt_trace_enum_fst___024root__trace_init_sub__TOP__t__DOT__sink__0(vlSelf, tracep);
```
which calls `declBus()` on the interface's enum variable.  I haven't dug into the FST library or the `VerilatedFst` code enough yet to know for certain, but it appears that the order is important here.  If I move the call to `Vt_trace_enum_fst___024root__trace_init_sub__TOP__t__DOT__sink__0()` below the `declDTypeEnum()` and re-build the enum names come back.

This was introduced by: https://github.com/verilator/verilator/commit/95c4ade718a21d51c661cbe37b252924c0a822cd

So there are two issues here.  First, is how to fix the verilated trace code.  I'll start digging, but @gezalore if you have any pointers on how to get the `init` calls to happen after `declDTypeEnum` happens that would be appreciated.

Second, is how to catch this via a test.  It appears that the `fst2vcd`'ed VCD is trying to tell us that the interface variable doesn't have enum info.  Here is what was happening before:
```
$ fst2vcd -e good.enum.fst | grep -A1 attrbegin
$attrbegin misc 07 $unit::state_t 4 VAL_A VAL_B VAL_C VAL_D 00 01 10 11 1 $end
$attrbegin misc 07 "" 1 $end
$var logic 2 " v_enumed [1:0] $end
--
$attrbegin misc 07 "" 1 $end
$var logic 2 # state [1:0] $end
```
and here is what I see now:
```
$ fst2vcd -e bad.enum.fst | grep -A1 attrbegin
$attrbegin misc 07 $unit::state_t 4 VAL_A VAL_B VAL_C VAL_D 00 01 10 11 1 $end
$attrbegin misc 07 "" 1 $end
$var logic 2 # v_enumed [1:0] $end
```
This is from a simplified version of Andrew's test, I'll put up a PR soon.  But you can see that there was an extra `$attrbegin` which came before the interface enum variable in question.

However, while I can imagine what this means, I can't find `$attrbegin` in the LRM.  I'm assuming this is a gtkwave only thing:
https://github.com/veripool/vcddiff/blob/77b4aaefa2602655c85c81b2e67699aa33a1f05c/vcddiff.c#L437

So I can look at making `vcddiff` more aware of these attributes, or if preferred I can do something bespoke for a particular test like count the number of `$attrbegin`s or something.  Thoughts?

gezalore, 2023-12-13:
Thanks for the investigation.

> So there are two issues here. First, is how to fix the verilated trace code. I'll start digging, but @gezalore if you have any pointers on how to get the init calls to happen after declDTypeEnum happens that would be appreciated.

Ok, so the problem is that EmitCImpl https://github.com/verilator/verilator/blob/c5ba6e22fa63f8e65d242bd3a4547de15e7a046b/src/V3EmitCImp.cpp#L845 will lazily emit the type definition when encountering the first AstTraceDecl using it. However, the order of AstTraceDecls in the tree is not in the order of execution, so this will go bad if we have:

Emit in some AstCFunc:
  - AstCCall to scope init function that uses enum type
  - AstTraceDecl that uses enum <- This emits the declDTypeEnum call

Emit actual scope init function that uses enum type:
  - AstTraceDecl that uses enum <- Due to the earlier one, this won't emit declDTypeEnum

We want the scope init functions to be called in the order they are currently, which corresponds to soruce declaration order. (Or at least we don't want restrictions on what order they are called in the future, and in general I think it's not guaranteed a correct ordering exists which avoids this issue in all cases.) Hence I think the best way to fix this once and for all, is to emit the declDTypeEnum into a separate function (or functions, if many, and need splitting) which gets called at the beginning of the top level trace init function.

That is, these should go in a new function:
https://github.com/verilator/verilator/blob/c5ba6e22fa63f8e65d242bd3a4547de15e7a046b/src/V3EmitCImp.cpp#L708-L734

Which gets called somewhere around here:
https://github.com/verilator/verilator/blob/c5ba6e22fa63f8e65d242bd3a4547de15e7a046b/src/V3EmitCModel.cpp#L517

Would you be happy to attempt that?

Testing-wise, I'm happy with a simple attribute count, but I defer to others if they have an opinion.

gezalore, 2023-12-13:
P.S.: The reason the refactor broke this, is because prior to that, we used to create duplicate AstTraceDecls for every copy of an interface - and miss some interesting cases - , now we just invoke the one canonical initializer function for every interfeceRef var, when they appears.

toddstrader, 2023-12-13:
> Testing-wise, I'm happy with a simple attribute count
Good because that's way easier.  I've added this to #4756 .

> Would you be happy to attempt that?
Yeah, I'll take a look at the fix.

toddstrader, 2023-12-13:
@gezalore I see that I need to forward declare and call something akin to `__trace_init_top()` in `V3EmitCModel.cpp`.  And I see that `__trace_init_top()` is emitted by `TraceDeclVisitor`.  So I can visit `AstNodeDType`s in `TraceDeclVisitor` as well and drop most of the `tracep->declDTypeEnum()` code in there.

However, what I'm uncertain about is that `EmitCTrace::emitTraceDeclDType()` is currently building `m_enumNumMap`.  But in the world I'm describing that will be built in `TraceDeclVisitor`.  So I'm wondering, is everything I'm saying so far sane?  And if so, how do I get this map from `TraceDeclVisitor` to `EmitCTrace`?

toddstrader, 2023-12-13:
I guess I can just add another `op` to `AstEnumDType`?  Lemme try that.  But please yell if I'm not making sense.

gezalore, 2023-12-13:
I have imagined this change just to be in V3EmitC*, using the same lazy emits, in visit(AstTraceDecl*) in EmitCImpl, except the text goes in a new file or buffer, then dumped into a new function. I know it's not exactly stellar to process this as text, but as we don't have AstNodes to represent the declDTypeEnum etc beyond text, I don't see a problem with that for now.

You can bring it forward if you want t. I don't see much advantage to doing that. For keeping track of the enumNumMap, please add a new int field to AstEnumDType instead of a new child node if you have to.

wsnyder, 2023-12-19:
Thanks for the pull.

closed 2023-12-19 by wsnyder