verilator/verilator #4664: interface parameters not being used correctly in generate for loop

jallennk, 2023-11-02:
I'm using Verilator 5.018 2023-10-30 rev UNKNOWN.REV in Ubuntu 22.04

Something wrong is happening when I refer to a parameter in an interface in a for-loop of a generate statement:

Interface with a parameter:

intf.sv
~~~~
interface intf
#(
 parameter WIDTH = 10
) (
  input clk
);

endinterface
~~~~

Top level module:

top.sv:
~~~~
module top ( input clk );

intf #(.WIDTH(10)) g (.clk (clk));

t i_t ( .g (g) );

endmodule
~~~~

Sub module where we instantiate a new interface:

t.sv
~~~~
module t  ( intf g );

intf #(.WIDTH(20)) h (.clk(g.clk));   // Override WIDTH

u i_u  ( .g (h) );  // Pass in new interface

endmodule
~~~~

Module with generate:

u.sv:
~~~~
module u ( intf g );

logic [3:0] array[g.WIDTH:0];   <---- g.WIDTH is 20 here, correct.

genvar i;
generate
  for (i = 0; i != g.WIDTH; i = i + 1)   <--- g.WIDTH is 10 here??? This is wrong.
  //for (i = 0; i != 20; i = i + 1)   <---- It should be this!
    tt i_tt(.g (g), .o (array[i]));
endgenerate

endmodule
~~~~

tt.sv
~~~~
module tt
  (
  intf g,
  output logic [3:0] o
  );

assign o = 4'd7;

endmodule
~~~~

You can see that when I used a constant 20 I get these additional lines in 

Vtop___024root__DepSet_heccd7ead__0__Slow.cpp

~~~~
39a40,49
>     vlSelf->top__DOT__i_t__DOT__i_u__DOT__array[0xaU] = 7U;
>     vlSelf->top__DOT__i_t__DOT__i_u__DOT__array[0xbU] = 7U;
>     vlSelf->top__DOT__i_t__DOT__i_u__DOT__array[0xcU] = 7U;
>     vlSelf->top__DOT__i_t__DOT__i_u__DOT__array[0xdU] = 7U;
>     vlSelf->top__DOT__i_t__DOT__i_u__DOT__array[0xeU] = 7U;
>     vlSelf->top__DOT__i_t__DOT__i_u__DOT__array[0xfU] = 7U;
>     vlSelf->top__DOT__i_t__DOT__i_u__DOT__array[0x10U] = 7U;
>     vlSelf->top__DOT__i_t__DOT__i_u__DOT__array[0x11U] = 7U;
>     vlSelf->top__DOT__i_t__DOT__i_u__DOT__array[0x12U] = 7U;
>     vlSelf->top__DOT__i_t__DOT__i_u__DOT__array[0x13U] = 7U;
~~~~

wsnyder, 2023-11-03:
The generate loop value is determined during early elaboration before the interface parameter has been overridden, while the variable's use of the parameter is resolved later.  This is a known bug in how Verilator does parameter resolution.  (Because it assumes an interface may be declared inside the generate, rather then following a complicated tree of parameter resolutions.)  @donlon has been working on some recent improvements in elaboration, @dolon perhaps these may eventually cover this?

jallennk, 2023-11-03:
FWIW: I was hoping to use interfaces to encapsulate many parameters that otherwise would have to be tediously typed for each of many module instantiations from my fixed point math library (each instantiation is an operator).  But I need parameter overrides to work...  Can anyone think of an alternative solution?  Typical parameters include size of fixed point number, binary point location, and latency values to make matching pipelines.  So mostly people will use same sized numbers which can be specified in a common interface, but sometimes a section needs more precision, so we make an interface with overridden parameters.

donlon, 2023-11-03:
> perhaps these may eventually cover this?

It seems to be another subtle problem - cross-references to something inside parameterized interfaces (e.g. g.WIDTH) need to be relinked to the new module with a 'visit' function, before using it. But the xrefs inside GenFor is not visited before unrolling. We just need to iterate the statements.

@jallennk So I think it'll be fixed quickly. But before the fix is merged you may try other workarounds, you can either -

Put the expression in a parameter before, e.g.

``` SystemVerilog
localparam MY_WIDTH = g.WIDTH;  // MY_WIDTH is resolved to the correct value
genvar i;
generate
  for (i = 0; i != MY_WIDTH; i = i + 1)
  // ...
```

Or, pass those parameters with packed structure, which is well supported by Verilator. An example is here ([t_struct_param.v](https://github.com/verilator/verilator/blob/915ceb2d049b99c96d5241e9aabeeb8a8811967a/test_regress/t/t_struct_param.v)).

And thank you for reporting this!

jallennk, 2023-11-03:
Thanks, @donlon the localparam workaround is great, works for me.

(BTW, Xilinx "xsim" did not have this bug, but of course is much slower)

PR #4665 (2023-11-04): Fix interface parameters used in loop generate constructs (#4664)
Fixes #4664
closed 2023-11-04 by wsnyder