verilator/verilator #4679: Support unpacked structs in DPI-C calls

mpriestleyidex, 2023-11-10:
Verilator version tried: 5.016.

Let's say I have C:
```
typedef struct {
  int f1;
  int f2;
} thing_t;
```
and SV:
```
typedef struct {
  int f1;
  int f2;
} thing_t;

import "DPI-C" function int do_thing(thing_t thing);
```

Verilator doesn't swallow this, saying:
`error: cannot convert ‘blah__DOT__thing_t__struct__0’ to ‘IData’`

I can work around it by making the SV struct `packed`, but then the fields come across in reverse order, so I've effectively ended up with (minus expletives):
```
// Structs have to be 'packed' for v e rilator.
// This means the order of members is REVERSED w.r.t their C declaration.
typedef struct packed {
  int f2;
  int f1;
} thing_t;
```

wsnyder, 2023-11-10:
Verilator doesn't currently support passing unpacked structures to DPI as described in IEEE H.7.8.  We'd welcome a pull request with appropriate tests to improve this.

The passing of a packed struct I believe is per IEEE.  The problem is bit order of fields is undefined by C language; in C you need to look at the bit endian of the compiler and do an #ifdef to select between either ordering of fields.  (Google "bit order c" for lots more on this.)

mpriestleyidex, 2023-11-10:
@wsnyder I intended this as a feature request ticket to implement that support. If there's already one, please link to it.

I'm not saying the packed behaviour is wrong, just that it's an irritating workaround in the absence of unpacked struct passing.

wsnyder, 2025-10-14:
@Alex-Mann As to #6560, IEEE defines how this should be handled, you can update the existing dpi tests to make sure they pass on others, then update mostly V3Task.cpp to handle the new complex data types.

We probably also should dump the SystemVerilog data types as the C equivalents (do other simulators do that)?

Note that if you just want to hack around this you can today

    import "DPI-C" context task example(longint unsigned strp);
    longint unsigned strp;
    ...
        strp = $c("&", cfg);
        example(strp, ...)

Just be aware the passed pointer will be to a C++ struct, not necessarily C compatible, as it has methods etc.

closed 2026-10-20 by wsxarcher