verilator/verilator #4686: Optimize redundant timestamps in trace files

wsnyder, 2023-11-12:
Traces with --timing sometimes have many timing events that do not result in any signals changing, e.g. t_timing_osc. This results in a bunch of timestamps in the VSD/FST that are unnecessary and just use disk space/cpu time. We should remove them and print a timestamp only when first value change is needed.  (Be careful to also add timestamp at file close regardless!)

This is a reminder to get back to fixing this, if someone else wants to work on it let me know!

valdaarhun, 2023-11-12:
Hi. I have never contributed to verilator before. I would like to work on this.

wsnyder, 2023-11-12:
Great!  Very welcome to have a first contribution!

Some notes:
emitTimeChange is what prints the timestamp.
Both vcd and fst trace formats use the same base class, so I think you can ignore the difference for the fix.
There's offloaded tracing, and non-offload.  You might need a separate place for each (not sure), suggest look at the fix for non-offloaded first.
Look at docs/internals.rst for details on testing, e.g. you'll need to run `test_regress/t/t_timing_osc.pl --gold` when this is done to pass the test once the extra timestamps are removed.

valdaarhun, 2023-11-13:
Thank you for your reply.

> Some notes:
> emitTimeChange is what prints the timestamp.
> Both vcd and fst trace formats use the same base class, so I think you can ignore that difference for the fix.
> There's offloaded tracing, and non-offload. You might need a separate place for each (not sure), suggest look at the fix for non-offloaded first.

I'll start with this.

valdaarhun, 2023-11-17:
Hi, I was taking a look at the contents of `t_timing_osc.out` after running `test_regress/t/t_timing_osc.pl`.

Part of the output is as shown below:

```
$version Generated by VerilatedVcd $end
$timescale 1fs $end
 $scope module TOP $end
  $scope module tb_osc $end
   $var wire 1 # dco_out $end
   $scope module dco $end
    ...
    $var wire 1 $ coarse_out $end
    $var wire 1 % medium_out $end
    $var wire 1 & fine_out $end
   $upscope $end
  $upscope $end
 $upscope $end
$enddefinitions $end

#0
0$
0%
...
0&
#138200
#139000
#150000
#152800
...
```

For example, there is no signal change at `# 138200`. Is the objective to prevent such timestamps from being written to the trace file?

wsnyder, 2023-11-18:
> For example, there is no signal change at # 138200. Is the objective to prevent such timestamps from being written to the trace file?

Correct

valdaarhun, 2023-11-25:
Hi. Based on what I have understood so far, assuming no offloading for the time being, the timestamp is printed to the trace file [here](https://github.com/verilator/verilator/blob/master/include/verilated_trace_imp.h#L586), and then signal changes are identified and written over [here](https://github.com/verilator/verilator/blob/master/include/verilated_trace_imp.h#L590-L612).

The `chg*` [functions](https://github.com/verilator/verilator/blob/master/include/verilated_trace.h#L485-L522) are responsible for identifying those changes in signals before calling the `full*` functions, which in turn call the `emit*` functions.

The trace is finally written to the file in `finishLine`. Since, each `emit*` function calls the `finishLine` function, what I understand is that at time _t_, each signal is checked and then written to the file one by one, rather than saving the set of changes in a buffer and then writing the buffer in one go.

I have two ideas on how to print the timestamp only when the signal changes.

1. One would be to simply shift the `emitTimeChange` function to each of the `full*` functions or `emit*` functions. At every time step _t_, the timestamp will be printed when a signal change is found for the first time. On subsequent finds for the same time step, the timestamp won't be printed again. We could have a flag variable to keep track of this.

2. The other approach would be to have a map that maps the signal code to the new value only for those signals that have changed at time _t_. If the mapping is empty, then the timestamp won't be printed. Otherwise, for each key-value pair in the mapping, the `finishLine` function will write it to the file.

I would love to get feedback and suggestions on how to best tackle this.

wsnyder, 2023-11-25:
I think checking in emit* is preferable (note full* calls emit*).  Ideally the check would be in a function that is already called at a low level, as that will have the smallest incremental loss in performance for the new branch tests.  emitTimeChange might also work, but note we also want to fix FST traces so adding to every emit* seems cleaner.

valdaarhun, 2023-11-29:
Sorry for the late response.

> I think checking in emit* is preferable (note full* calls emit*).

I was thinking of shifting the implementation of `emitTimeChange` present in `VerilatedVCD` and `VerilatedFST` to the `emit*` functions in `VerilatedVCDBuffer` and `VerilatedFSTBuffer` respectively.

> Ideally the check would be in a function that is already called at a low level, as that will have the smallest incremental loss in performance for the new branch tests.

I didn't really understand the performance part. Are you referring to tests related to measuring/gauging the likeliness of a particular branch being executed during the program's control flow?

Also, since the signal change check is already performed in the `chg*` functions, I think it should be feasible to simply call `emitTimeChange` in the `emit*` functions.

wsnyder, 2023-11-29:
Seems reasonable.

>Are you referring to tests related to measuring/gauging the likeliness of a particular branch being executed during the program's control flow?

Yes, e.g. make sure the test is in a function that is already called, and so when dumping and timestamp was already printed for that cycle, there is nothing added to the code path except one well-predicted branch.

gezalore, 2023-11-29:
Please also keep in mind that when we do parallel tracing (currently only in with VCD with --threads > 1), multiple threads will be executing the tracing functions, so in the proposal above there is a race condition about which thread calls `emitTimeChange`. For performance we need to be able to resolve the branch that @wsnyder mentioned to see whether we need to call `emitTimeChange` without locks or atomics (locks and atomics are fine to resolve the race when we a thread thinks it might need to emit the time change).

valdaarhun, 2023-12-07:
Will `VerilatedTraceOffloadBuffer` be used when `--threads` is greater than 1?

gezalore, 2023-12-07:
When tracing VCD, `VerilatedTraceOffloadBuffer` is never used. When using FST, `VerilatedTraceOffloadBuffer` is used only when `--trace-threads` >= 2 (you cansee `useTraceOfflaod()` in V3Options.h

valdaarhun, 2023-12-13:
Thank you for the response. I am still pondering over what approach might be best to handle this. Right now, I am thinking of setting a flag as soon as a change in signal is identified. This identification is equivalent to an `emit*` function being called. The timestamp will be printed and the flag will be set. Once the flag is set, that will be used to prevent the timestamp from being reemitted.

No locks will be required to identify the change in signal. However, I am unable to think of a way in which the value of the flag can be checked without using a lock.

So, within an `emit*` function, we would have:

```
acquire a lock
    if the flag is unset:
        set the flag
        emit the timestamp
   end if
release the lock
```

Any thoughts on this?

valdaarhun, 2023-12-13:
Also, since the simulation is evaluated only at the end of each cycle, is it safe to say that each signal will be printed at most once per cycle. If that's case, how's the idea of temporarily storing the new values in a buffer?

Then all the values can be printed with the timestamp by a single thread after the evaluation ends (even when --threads > 1)

wsnyder, 2023-12-13:
It currently does write to a buffer.  For VCD we could in theory write the timestamp to the buffer then backup over it if unused, or we could see if the buffer is non-empty and if so write the timestamp before the buffer. But neither of these tricks work for FST.

As to threads, it's fastest to make a thread-local variable.  When it's zero (no emit) then check an atomic read-and-set variable to see if emitting is needed, then set the thread local variable so all later calls on that thread don't need to execute the atomic instruction.

gezalore, 2023-12-13:
> As to threads, it's fastest to make a thread-local variable.

If you use a thread-local, please make sure it has the `tls_model("local-exec")` attribute, otherwise it can be expensive if the model is built with "-fPIC" into a shared library (https://godbolt.org/z/EE9GPdeYz).

wsnyder, 2023-12-13:
>If you use a thread-local, please make sure it has the `tls_model("local-exec")` attribute, otherwise it can be expensive if the model is built with "-fPIC" into a shared library (https://godbolt.org/z/EE9GPdeYz). 

@gezalore if this is important please make a pull to somehow make that the convention throughout Verilator, unrelated to this change.  This pull would then follow that convention, depending which pull merges first.

valdaarhun, 2023-12-16:
Thank you for the responses.

I think I've understood how to tackle this issue. In the `emit*` functions, I plan on using a thread_local variable to check if the thread is yet to emit the timestamp. If it is yet to emit it, I'll then use an `atomic_flag` type and `atomic_flag_test_and_set` to test and set the actual flag before printing the timestamp.

If this approach is alright, I'll start implementing it and will open a draft PR.

wsnyder, 2023-12-16:
Sounds good

closed 2025-04-05 by wsnyder