verilator/verilator #4691: Trying to understand coverage report

gkamendje, 2023-11-12:
I have generated coverage information as described here [coverage analysis](https://verilator.org/guide/latest/simulating.html?highlight=coverage#coverage-analysis). I have used lcov to generate hmtl coverage report. I am having hard time understanding the generated report. 

1. How could it be that the hit count on line 255 is zero while there are non zero hits for statements inside the branch (lines 256 and following). 
2. How could it be that the hit count on line 270 is non zero (352) while the hit count of the only line within that particular branch is zero?
[image]

wsnyder, 2023-11-12:
lcov is a lossy format. Please try creating a text report with all points listed using verilator_coverage. If that doesn't clarify it, post the reports for the lines you showed please.

gkamendje, 2023-11-14:
I re-ran `verilator_coverage` with the following options.
`verilator_coverage --annotate-min 1 --annotate /my_path/annotated --annotate-all --annotate-points --rank --write`
Here is the text output of `verilator_coverage` 
[image]

and here is once again the resulting lcov html.

[image]

wsnyder, 2023-11-15:
You can see that Verilator is inserting "else" points because it incorrectly thinks those if statements don't have a else clause.  So, this looks like a bug, can you please make an example that shows this issue that is standlone runable?  Perhaps you could also look at V3Coverage and see if you can identify the fix and make a pull?

gkamendje, 2023-11-15:
Ah ok, I see now what is going on. I will try to create a test case. and perhaps submit a PR with a fix. 
What about lines 278 and 285? Any idea why the count is zero?

gkamendje, 2023-11-15:
On line 222 in a similar situation the count is non zero see below.
[image]

wsnyder, 2023-11-15:
Don't look at lcov, look at the text report. You see a similar issue to the other bug.

wsnyder, 2023-11-26:
Any luck creating a test and fix?

gkamendje, 2023-11-27:
I am currently busy with a Tape-Out coming up at the end of November. I hope to be able to look at it early December.

gkamendje, 2023-11-27:
In the mean time could you please point me to a `verilator_coverage` test case, I could start from?

wsnyder, 2023-11-27:
For example test_regress/t/t_cover_line_cc_vlt.pl. You can probably change the verilog it uses t/t_cover_line.v to have whatever code causes the problem.  Then you'd run from the shell "test_regress/t/t_cover_line_cc_vlt.pl --gold" to update the reference files for expected output (though that expected output should show this bug until its fixed.)

gkamendje, 2023-11-29:
Thanks I will have a look at it

gkamendje, 2024-01-18:
I have created the file `test_regress/t/t_cover_else_points.pl` with the following content along with the Verilog test case. 
```
scenarios(simulator => 1);

top_filename("t/t_cover_else_points.sv");
golden_filename("t/t_cover_else_points.out");
compile(
    verilator_flags2 => ['-cc', '--coverage', '--exe', '--main', '--timing' ],
    make_main => 0,
    );

execute(
    check_finished => 1,
    );

# Read the input .v file and do any CHECK_COVER requests
inline_checks();

run(cmd => ["../bin/verilator_coverage",
            "--annotate-points",
            "--annotate", "$Self->{obj_dir}/annotated",
            "$Self->{obj_dir}/coverage.dat",
            ],
    verilator_run => 1,
    );

files_identical("$Self->{obj_dir}/annotated/t_cover_else_points.sv", $Self->{golden_filename});

ok(1);
1;
```
I get the following error when I run `./test_regress/t/t_cover_else_points.pl --golden`
```
     obj_vlt/t_cover_else_points_vlt/Vt_cover_else_points_vlt    > obj_vlt/t_cover_else_points_vlt/vlt_sim.log
*-* All Finished *-*
- t/t_cover_else_points.sv:46: Verilog $finish
%Warning: vlt/t_cover_else_points_vlt: File_contents file not found: obj_vlt/t_cover_else_points_vlt/coverage.dat

        ../bin/verilator_coverage --annotate-points --annotate obj_vlt/t_cover_else_points_vlt/annotated obj_vlt/t_cover_else_points_vlt/coverage.dat
%Error: Can't read obj_vlt/t_cover_else_points_vlt/coverage.dat

```
Any hint on what I doing wrong?

wsnyder, 2024-01-18:
You attached t_cover_else_points.pl but the error was on t_cover_else_points_vlt.pl so perhaps you're running the wrong test?

gkamendje, 2024-01-18:
That was a typo the file I attached is actually `test_regress/t/t_cover_else_points_vlt.pl`.

gkamendje, 2024-01-18:
@wsnyder 
The file `coverage.dat` is not generated. I usually have the line ` contextp->coveragep()->write("logs/coverage.dat");` in the `.cpp` wrapper that handles the creation of this file. Any hint on how I should achieve this in this context given that I am passing the option `make_main => 0` to the compile script? I could not figure out how the example file `test_regress/t/t_cover_line_cc_vlt.pl` controls the generation of `coverage.dat`

gkamendje, 2024-01-19:
I modified the compile directive of the test script to look like this:
```
compile(
    verilator_flags2 => ['-cc', '--coverage',   '--timing' ],
    );
```
However now I am getting the following error message. 
```
driver: Leaving directory '/home/gkamendje/tmp_compile/verilator_myfork/test_regress/obj_vlt/t_cover_else_points'
        obj_vlt/t_cover_else_points/Vt_cover_else_points    > obj_vlt/t_cover_else_points/vlt_sim.log
%Error: ../../obj_vlt/t_cover_else_points/Vt_cover_else_points__main.cpp:31: %Error: Timeout; never got a $finish
Aborting...
%Warning: vlt/t_cover_else_points: Exec of obj_vlt/t_cover_else_points/Vt_cover_else_points failed: %Error: ../../obj_vlt/t_cover_else_points/Vt_cover_else_points__main.cpp:31: %Error: Timeout; never got a $finish

```

My testbench looks like this:

```
module t  ;
    logic        is_su_mode;
    logic        is_em_emul;
    logic        is_ata_emul;
    logic [3:0]  user_word_cnt;
    logic        page;
    logic [6:0]  cfg;
    initial begin
       is_su_mode    =1'b1; 
       is_em_emul    =1'b0; 
       is_ata_emul   =1'b0; 
       page          =1'b0; 
       cfg           =7'h05; 
       #100;
       is_su_mode    =1'b0; 
       is_em_emul    =1'b1; 
       is_ata_emul   =1'b0; 
       page          =1'b0; 
       cfg           =7'h06; 
       #100;
       is_su_mode    =1'b0; 
       is_em_emul    =1'b0; 
       is_ata_emul   =1'b1; 
       page          =1'b0; 
       cfg           =7'h10; 
       #100;
       is_su_mode    =1'b0; 
       is_em_emul    =1'b0; 
       is_ata_emul   =1'b1; 
       page          =1'b1; 
       cfg           =7'h60; 
       #100;
       is_su_mode    =1'b0; 
       is_em_emul    =1'b0; 
       is_ata_emul   =1'b1; 
       page          =1'b1; 
       cfg           =7'h60; 
       #100;
       $write("*-* All Finished *-*\n");
       $finish;
    end
    a a_inst(
         .i_is_su_mode     ( is_su_mode   ), 
         .i_page           ( page         ),
         .i_cfg            ( cfg          ),
         .i_is_em_emul     ( is_em_emul   ),
         .i_is_ata_emul    ( is_ata_emul  ),
         .o_user_word_count( user_word_cnt)
    );
endmodule

```

wsnyder, 2024-01-20:
The default main made by driver.pl isn't designed to work with --timing.  Use

```
compile(
    verilator_flags2 => ["--exe --main --timing -Wall"],
    make_main => 0,
    );
```

Though... the current --main option doesn't know how to add the coverage dumping... you could for this test use your own main.cpp, but better perhaps you'd open a pull request to dump coverage when using --main?   The filename to dump to should come from a VerilatorContext variable that we have a new +verilator+coverage+file option to override.

gkamendje, 2024-01-20:
@wsnyder 
Thanks for the hint. I was able to get the test case to run with a separate main.cpp. 
The test is passing locally. However some CI checks are failing with messages similar to this. 
```
======================================================================
vlt/t_cover_else_points: ==================================================
	../bin/verilator_coverage --annotate-points --annotate obj_vlt/t_cover_else_points/annotated obj_vlt/t_cover_else_points/coverage.dat
%Error: Can't read obj_vlt/t_cover_else_points/coverage.dat
```

Regarding the PR with the new feature, That is certainly a very nice feature to have. Unfortunately, I do not have the bandwidth to look at it right now.

wsnyder, 2024-01-20:
Does test_regress/obj_vlt/t_cover_else_points/coverage.dat exist?  Note C++ will need to write to a special output filename like as 
 `VerilatedCov::write(VL_STRINGIFY(TEST_OBJ_DIR) "/coverage.dat");`

because the default is just obj_dir which the test harness overrides.

gkamendje, 2024-01-20:
@wsnyder 
I have the following in the cpp file as suggested:
```
#if VM_COVERAGE    
contextp->coveragep()->write(VL_STRINGIFY(TEST_OBJ_DIR) "/coverage.dat");
#endif
```
and the following in the `.pl` file 
```
run(cmd => ["../bin/verilator_coverage",
            "--annotate-points",
            "--annotate", "$Self->{obj_dir}/annotated",
            "$Self->{obj_dir}/coverage.dat",
            ],
    verilator_run => 1,
    );

```
Locally on my machine, I can successfully run `make -j && test_regress/t/t_cover_else_points.pl --hashset=0/3 --vlt`. However, when I do push to github, the check `[Test | ubuntu-20.04 | gcc | dist-vlt-0]`(https://github.com/verilator/verilator/actions/runs/7597284945/job/20691993074?pr=4844#logs) still fails with the message 
```
#vlt/t_cover_else_points: %Error: Exec of ../bin/verilator_coverage failed: 
		make -j && test_regress/t/t_cover_else_points.pl --hashset=0/3 --vlt

vlt/t_cover_else_points: ==================================================
	../bin/verilator_coverage --annotate-points --annotate obj_vlt/t_cover_else_points/annotated obj_vlt/t_cover_else_points/coverage.dat
%Error: Can't read obj_vlt/t_cover_else_points/coverage.dat

```
Not sure what I am doing wrong here but it still looks like the path to the coverage file is not found during CI checks. I will pull PR #4844 hoping it has a workaround the issue.

gkamendje, 2024-01-21:
I have tried with PR #4844 but unfortunately the checks are still failing. I noticed that oddly enough the check is failing only in Ubuntu 20.04 with gcc. Ubuntu 22.04 is ok for both gcc and clang. and Ubuntu 20.04 clang is passing as well.