verilator/verilator #4693: Start register output value inconsistent with the other simulators

flaviens, 2023-11-13:
Hi there!

## Description

I found a behavior of registers when the simulation starts that is inconsistent with other simulators, so I investigated it a bit.

Consider the following snippet, essentially a very basic register.

```
module top(in_data, out_data);
  input [2:0] in_data; // 0: clock, 1: reset, 2: data
  wire [2:0] in_data;
  output out_data;
  wire out_data;
  reg _0_;
  always_ff @(CLOCK_POLARITY in_data[0], RESET_POLARITY in_data[1])
    if (!in_data[1]) _0_ <= 1'b1;
    else _0_ <= in_data[2];
  assign out_data = _0_;
endmodule
```

What I observe is that after evaluating with constant inputs, the output is always `0`, but with the other simulators that I tested (Icarus and Modelsim), the output depends on the inputs.

Please see the table below for details (please note that Icarus matches with Modelsim in all cases).

Row ID | Clock polarity | Reset polarity | Data value | Clock value | Reset value | Output bit (Verilator) | Output bit (Icarus) | Output bit (Commercial)
-------|----------------|----------------|------------|-------------|-------------|------------------------|---------------------|----------------------
0      | 0              | 0              | 0          | 0           | 0           | 0                      | 1                   | 1                       
1      | 0              | 0              | 0          | 0           | 1           | 0                      | 0                   | 0                       
2      | 0              | 0              | 0          | 1           | 0           | 0                      | 1                   | 1                       
3      | 0              | 0              | 0          | 1           | 1           | 0                      | x                   | x                       
4      | 0              | 0              | 1          | 0           | 0           | 0                      | 1                   | 1                       
5      | 0              | 0              | 1          | 0           | 1           | 0                      | 1                   | 1                       
6      | 0              | 0              | 1          | 1           | 0           | 0                      | 1                   | 1                       
7      | 0              | 0              | 1          | 1           | 1           | 0                      | x                   | x                       
8      | 0              | 1              | 0          | 0           | 0           | 0                      | 1                   | 1                       
9      | 0              | 1              | 0          | 0           | 1           | 0                      | 0                   | 0                       
10     | 0              | 1              | 0          | 1           | 0           | 0                      | x                   | x                       
11     | 0              | 1              | 0          | 1           | 1           | 0                      | 0                   | 0                       
12     | 0              | 1              | 1          | 0           | 0           | 0                      | 1                   | 1                       
13     | 0              | 1              | 1          | 0           | 1           | 0                      | 1                   | 1                       
14     | 0              | 1              | 1          | 1           | 0           | 0                      | x                   | x                       
15     | 0              | 1              | 1          | 1           | 1           | 0                      | 1                   | 1                       
16     | 1              | 0              | 0          | 0           | 0           | 0                      | 1                   | 1                       
17     | 1              | 0              | 0          | 0           | 1           | 0                      | x                   | x                       
18     | 1              | 0              | 0          | 1           | 0           | 0                      | 1                   | 1                       
19     | 1              | 0              | 0          | 1           | 1           | 0                      | 0                   | 0                       
20     | 1              | 0              | 1          | 0           | 0           | 0                      | 1                   | 1                       
21     | 1              | 0              | 1          | 0           | 1           | 0                      | x                   | x                       
22     | 1              | 0              | 1          | 1           | 0           | 0                      | 1                   | 1                       
23     | 1              | 0              | 1          | 1           | 1           | 0                      | 1                   | 1                       
24     | 1              | 1              | 0          | 0           | 0           | 0                      | x                   | x                       
25     | 1              | 1              | 0          | 0           | 1           | 0                      | 0                   | 0                       
26     | 1              | 1              | 0          | 1           | 0           | 0                      | 1                   | 1                       
27     | 1              | 1              | 0          | 1           | 1           | 0                      | 0                   | 0                       
28     | 1              | 1              | 1          | 0           | 0           | 0                      | x                   | x                       
29     | 1              | 1              | 1          | 0           | 1           | 0                      | 1                   | 1                       
30     | 1              | 1              | 1          | 1           | 0           | 0                      | 1                   | 1                       
31     | 1              | 1              | 1          | 1           | 1           | 0                      | 1                   | 1                       

## Reproducing

I'm using `Verilator 5.018 2023-10-30 rev v5.004-74-g41e5f6b7d` on Ubuntu 22.04.
I also created [a repository for you to try](https://github.com/flaviens/verilator-b3).

Is this a bug or is it an undefined behavior from the Verilog standard?

Thanks!
Flavien

wsnyder, 2023-11-14:
It looks like your stimulus is racing (changing data along with clock). Please try redoing it to make sure the clock rises and falls without the data changing at the same time.

flaviens, 2023-11-14:
Hi @wsnyder, thank you for your answer! Yes I know. Does you answer confirm that both Verilator and the 2 other simulators are behaving correctly according to the SystemVerilog standard? I read in particular the parts about scheduling semantics but I may have missed mentions about what happens at the very beginning of the simulation.

wsnyder, 2023-11-14:
The standard doesn't cover the C wrapping into Verilog, so I can't answer that.  Another possibility BTW is use --timing with verilator and use all Verilog; it should then match the others (ignoring X).

flaviens, 2023-11-18:
Thank you for your responses!

closed 2023-11-18 by flaviens