verilator/verilator #4695: Using a localparam array results in Internal Error - InitArray type not assigned by AstPattern/Var visitor

jlwehle, 2023-11-15:
Running the attached test using verilator-5.018 on FreeBSD 13.2 x64 results in:

```
%Error: Internal Error: t/t_param_array9.v:12:31: ../V3Width.cpp:2483: InitArray type not assigned by AstPattern/Var visitor
                                                : ... note: In instance 't.i_a'
   12 |   localparam integer SF [4] = '{ 27, 27, 26, 25 };
      |                               ^~
```

I have not tried a recent build from git, though I did eyeball the change log.

[t_param_array9.pl.txt](https://github.com/verilator/verilator/files/13360665/t_param_array9.pl.txt)
[t_param_array9.v.txt](https://github.com/verilator/verilator/files/13360666/t_param_array9.v.txt)

donlon, 2023-11-15:
Hi,

This error occurs because an unpacked array is assigned to a parameter not declared as array, which is not supported yet by Verilator. I'll look into the issue.

Before it's fixed you have to rewrite it as

``` SystemVerilog
module b #( parameter minSF[4] ) ();
// ...
```

jlwehle, 2023-11-15:
Thanks ... this is distilled from a larger design where .minSF(SF) was used instead of .minSF(minSF).  Unfortunately I got caught up in the "Interal Error" and failed to see that this was a simple "typo" in the design.  Referencing the correct localparam does take care of the issue.

closed 2023-11-15 by jlwehle