verilator/verilator #4698: Reference module ports don't work when modules aren't inlined

RRozak, 2023-11-16:
Reference module ports don't work correctly when modules aren't inlined. It causes a failure of the following test:
```systemverilog
module my_mod (
    ref int x
);
    task static check();
       if (x != 15) $stop;
    endtask
endmodule

module t;
   int a = 15;
    my_mod mod (
        .x (a)
    );

    initial begin
       mod.check();
       $write("*-* All Finished *-*\n");
       $finish();
    end
endmodule
```
The problem can be reproduced by compiling the test with a command `verilator --binary path/to/file -fno-inline`.

wsnyder, 2023-11-22:
Probably related to #3385

gezalore, 2026-09-24:
The test case in this issue is fixed by #8427. Ref module ports that connect to a non VarRef (e.g. an ArraySel) still don't work though.

wsnyder, 2026-10-11:
Closing, the other half we'll wait for a new issue to report/fix.

closed 2026-10-11 by wsnyder