verilator/verilator #4699: disable isn't underneath a begin with name

thesourcerer8, 2023-11-16:
verilator -sc testbenchs34.sv  --coverage
%Error: k9f1208.sv:614:13: disable isn't underneath a begin with name: 'OutputZGen1'
  614 |             disable OutputZGen1;
      |             ^~~~~~~
        testbenchs34.sv:94:1: ... note: In file included from 'testbenchs34.sv'

From my point of view it should be legal to have one begin to disable another begin, so this should be a warning instead of an error?

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
[k9f1208.sv.txt](https://github.com/verilator/verilator/files/13377683/k9f1208.sv.txt)
Taken from:
https://github.com/thesourcerer8/nand_controller/

What 'verilator --version' are you using?  Did you try it with the git master version?
Verilator 5.019 devel rev v5.018-44-gb8417abee

What OS and distribution are you using?
Ubuntu Linux

May we assist you in trying to fix this in Verilator yourself?
Yes, please

wsnyder, 2023-11-16:
Agreed.  Might you consider making a pull request to add an appropriate unsupported message?  The easiest is to make the message indicate that it might be due to something unsupported.

thesourcerer8, 2023-11-16:
It seems to me that the current algorithm there just searches the tree upwards for the block with the given task name. One obvious solution would be to search the whole AST for the task with the given name, but this could be quite slow, depending on the size of the AST. So perhaps the preferred way would be to create a global list of all the tasks, so that the task that should be disabled can be found faster?
I was trying to change it to an unsupported warning message with v3warn(E_UNSUPPORTED,... but it's still treated as an error.

wsnyder, 2023-11-16:
Yes, use v3warn UNSUPPORTED.  It's an error as the design will not work correctly.  It can still be suppressed if using XML or lint and people don't care if simulation is broken.

The proper fix is to implement the feature.  For that the V3LinkDot should be used to resolve a full hierarchical name anywhere in the design, and pass to V3Fork which implements the disable.  So, some rework will be needed to do that.

Gchauvon, 2024-04-18:
While integrating some founders memory RAM controller in CVA6, I ran into this error as well. I'm quite interested in a fix.
I'm not familiar with the verilator code, so I don't think I could fix it by myself, especially if it requires big rework.

wsnyder, 2024-10-03:
The warning was fixed in April.  As to general support for this it's covered by still-open #5432

closed 2024-10-03 by wsnyder

hhhsiang, 2024-10-17:
hi @Gchauvon, I also meet this error when I integrate sram into the cache_subsystem of CVA6. Can you run simulation successfully now?

Gchauvon, 2024-10-21:
@hhhsiang I haven't try but since it's still not supported I don't think it will work.