verilator/verilator #4700: Support 2D dynamic array initialization

stevobailey, 2023-11-17:
Example code (`temp.sv`):
```
function automatic void array_2d();
  bit myarray [][];
  myarray = new[3];
  foreach (myarray[i])
    myarray[i] = new[4];
endfunction
```
Command and error message:
```
$ verilator --lint-only temp.sv
%Error: temp.sv:6:18: dynamic new() not expected in this context (data type must be dynamic array)
    6 |     myarray[i] = new[4];
      |                  ^~~
%Error: Internal Error: temp.sv:6:16: ../V3Width.cpp:6251: Node has no type
    6 |     myarray[i] = new[4];
      |                ^
```
Verilator details (built from main):
```
commit b8417abee5f10c782315059da83caec582b17785
Thu Nov 16 2023

$ $VERILATOR_ROOT/bin/verilator --version           
Verilator 5.019 devel rev v5.018-44-gb8417abee

```

wsnyder, 2023-11-17:
This should have worked, but likely hadn't been tried before. V3Width elaboration thinks that it's not a dynamic array for some reason.  Perhaps you'd be willing to look at a pull request to fix this?

stevobailey, 2023-11-18:
Okay, I'm not familiar with the internals of Verilator, but here's what I've found so far.

I print the data type of the input argument node here: 
https://github.com/verilator/verilator/blob/b8417abee5f10c782315059da83caec582b17785/src/V3Width.cpp#L3871

- If the test file uses a 1D dynamic array of type bit, it prints the data type as `bit[]`.
- If the test file uses a 2D dynamic array of type bit, the first new[] prints the data type as `UNSIZEDARRAYDTYPE[]` and the second new[] (within the foreach loop) prints the data type as `UNSIZEDARRAYDTYPE`.

So somehow the second dimension isn't seen as a dynamic array (as you said), but also the data type of both dimensions is unsized array instead of bit.

wsnyder, 2023-11-18:
AstUnsizedArrayDType should have become an AstDynArrayDType.  Somehow the function that makes it (around here https://github.com/verilator/verilator/blob/b8417abee5f10c782315059da83caec582b17785/src/V3Width.cpp#L2135) didn't do the lower data type subDType first.  Probably some recursion is off, though I do see it does call \iterateEditMoveDTypep directly above which should have handled it.

stevobailey, 2023-11-28:
I think the recursion is setting the data types correctly, but maybe the issue is that converting the unsized array into a dynamic one only occurs in the `visit(AstVar* nodep)` function. I'm guessing this visit function is called when the node's type is unknown, and types get resolved here. So this is correct the first time we visit the array (the first `new`). By the second time we visit the array and try to declare `new`, it knows the node's type is an unsized array, but not a dynamic one.

Maybe we need to move the code that converts an unsized array into a dynamic one into the `iterateEditMoveDTypep()` function, or somehow here loop through child nodes and convert each one until we reach a basic datatype.

stevobailey, 2023-12-02:
Not sure how to handle your note on dpiImports. It's probably causing the remaining issue.

I updated the visit(AstUnsizedArrayDType) without removing the code from visit(AstVar). Lint for this test passed:
```
    void visit(AstUnsizedArrayDType* nodep) override {
        if (nodep->didWidthAndSet()) return;  // This node is a dtype & not both PRELIMed+FINALed
        // Iterate into subDTypep() to resolve that type and update pointer.
        nodep->refDTypep(iterateEditMoveDTypep(nodep, nodep->subDTypep()));
        // Cleanup array size
        nodep->dtypep(nodep);  // The array itself, not subDtype
        if (!(m_ftaskp && m_ftaskp->dpiImport())) {
            UINFO(9, "Unsized becomes dynamic array " << nodep << endl);
            AstDynArrayDType* const newp 
                = new AstDynArrayDType{nodep->fileline(), nodep->subDTypep()};
            nodep->dtypep(newp);
            v3Global.rootp()->typeTablep()->addTypesp(newp);
        }    
        UINFO(4, "dtWidthed " << nodep << endl);
    }    
```
But when I run the `t_dynarray` test, compile fails with this:
```
In file included from Vt_dynarray__ALL.cpp:5:
Vt_dynarray___024root__DepSet_h32326748__0.cpp: In function 'void Vt_dynarray___024root___nba_sequent__TOP__0(Vt_dynarray___024root*)':
Vt_dynarray___024root__DepSet_h32326748__0.cpp:499:23: error: request for member 'renew' in 't__DOT__a2.VlQueue<unsigned char>::at(0)', which is of non-class type 'unsigned char'
  499 |     t__DOT__a2.at(0U).renew(1U);
```

wsnyder, 2023-12-02:
I think you're missing an userIterateChildren at the bottom.

If that doesn't help, run the test without your changes in another directory, and use verilator_difftree to compare the output .tree files before your change and after. It should suggest what V3Width is doing differently to the output.

stevobailey, 2023-12-03:
userIterateChildren didn't make a difference.

Before the change, the AST gets to `009_linkdotparam.tree`, then the original error message appears.

After the change, the AST gets to `087_cuse.tree`, then this new error message appears.

There are no differences in the tree up to 009, and the original tree doesn't get past that. So I don't think I can compare trees to debug this problem.

wsnyder, 2024-06-09:
Thanks for contributing this.

closed 2024-06-09 by wsnyder