verilator/verilator #4701: Multiple Definition Again

Phantom1003, 2023-11-17:
The bug I meet is similar to #4416, however, the difference point is that objects with different types are assigned with same name.

```
/usr/bin/ld: VTestbench__ALL.a(VTestbench___024root__DepSet_h17da5287__0__Slow.o): in function `VTestbench___024root___eval_initial__TOP__0(VTestbench___024root*)':
VTestbench___024root__DepSet_h17da5287__0__Slow.cpp:(.text+0x437): multiple definition of `VTestbench___024root___eval_initial__TOP__0(VTestbench___024root*)'; VTestbench__ALL.a(VTestbench___024root__DepSet_h17da5287__0.o):VTestbench___024root__DepSet_h17da5287__0.cpp:(.text+0x9a45): first defined here
```

In `VTestbench___024root__DepSet_h17da5287__0__Slow.cpp`, there is a `VL_ATTR_COLD void VTestbench___024root___eval_initial__TOP__0(VTestbench___024root* vlSelf)`, while the type of the `VTestbench___024root___eval_initial__TOP__0` in `VTestbench___024root__DepSet_h17da5287__0.cpp` is `VL_INLINE_OPT VlCoroutine VTestbench___024root___eval_initial__TOP__0(VTestbench___024root* vlSelf)`.

I use the latest v5.018.

I'm sorry, this is a private project, so I can't give you an example right now. Maybe there will be a stripped version if necessary.

wsnyder, 2023-11-17:
Please run with --debug, look in the generated obj_dir/*990_final.tree file for two lines with `CFUNC ... VTestbench___024root___eval_initial__TOP__0` and see what the "<letterNUMBERS>" is on those two lines, this corresponds to the filename and line number of the offending code (see internals.rst for more info).  That should help you make a test case.

wsnyder, 2023-11-17:
Good find, yes Slow and non-Slow use the same function naming space in C, so they should share the same counter.

Phantom1003, 2023-11-18:
I discovered that there's a namespace conflict in the `orderSequentially` function after some research. @wsnyder, you can find it at: https://github.com/verilator/verilator/blob/b8417abee5f10c782315059da83caec582b17785/src/V3Sched.cpp#L425-L443

To elaborate, in the `splitCheck` function (line 443), we append `__NUM` after the original function name to generate a new function name. However, we also append `__NUM` to the coroutine function's name at line 432. 

When a function has both a normal procedure and a delayed procedure (that generates a coroutine), and the normal procedure exceeds the node limit, the issue arises. In such cases, the normal procedure is split into multiple parts such as TOP_0 and TOP_1, while the delayed procedure doesn't exceed the limit and generates the TOP_0.

Hi, I'm planning to modify the naming rule for the coroutine located at line 432. My idea is to generate a name with a specific prefix, such as TOP_C0. However, I couldn't find any similar names in the Verilated file, so I'd like to ask your advice first, thanks.

Phantom1003, 2023-11-18:
By the way, I'm not quite sure if it is safe to remove the `scopep->user2Inc()` call (at line 432) and directly use `splitCheckCreateNewSubFunc` to generate the new name for coroutines.

wsnyder, 2023-11-18:
That analysis makes sense.  It should be fine to rename arbitrarily to disambiguate.

It should be ok to remove user2Inc if you do it differently, but you will somehow need to track an incrementing identifier and using a user is the most efficient way.  (We don't want a global number because that causes massive recompiles if one low-numbered function suddenly grows/shrinks to change the splitting.)

wsnyder, 2023-11-21:
Thanks for the fix!

closed 2023-11-21 by wsnyder