verilator/verilator #4705: typedef within a class not found

muzafferkal, 2023-11-19:
Hi,
I'm trying to adapt some of the https://github.com/pulp-platform/riscv-dbg/ files for my purposes.
In there there is a declaration of a class with a typedef inside:
  class riscv_dbg #(
    parameter int IrLength = 5,
    parameter IDCODE    = 'h1,
    parameter DTMCSR    = 'h10,
    parameter DMIACCESS = 'h11,
    parameter time TA = 0ns, // stimuli application time
    parameter time TT = 0ns  // stimuli test time
  );

    typedef jtag_test::jtag_driver#(.IrLength(IrLength), .IDCODE(IDCODE), .TA(TA), .TT(TT)) jtag_driver_t;
    jtag_driver_t jtag;

When I use this code as follows, I get an error which says jtag_driver_t is unexpected identifier. Is this because it's a typedef inside the class? Any idea what I'm missing?

typedef jtag_test::riscv_dbg #(
    .IrLength (5),
    .TA (JTAGPeriod*0.1),
    .TT (JTAGPeriod*0.9)
  ) riscv_dbg_t;
  `endif

  riscv_dbg_t::jtag_driver_t jtag_in = new (jtag_mst);

wsnyder, 2023-11-19:
That syntax has probably never been tested, so needs debugging. Could you consider modifying one of the test_regress issues to show the problem, check it passes on another simulator, then debugging?

muzafferkal, 2023-11-20:
I have partial evidence that at least Modelsim vsim and Vivado simulators support this syntax as the testbench comes with scripts for those: https://github.com/pulp-platform/riscv-dbg/blob/master/tb/jtag_dmi/run_vsim.sh. I'll try to make a test case when I have the time but my c/c++ and especially verilator code base skills are probably not good enough to debug it.

joecrop, 2024-10-07:
Here's a really simple example that raises this issue. I tested this on EDAPlayground and it compiles in VCS/Questa/Xcelium.

```systemverilog
class FixedNumber #(int p_BITS = 8, int p_FRAC = 8);
  typedef bit [p_FRAC-1:0] FracNum;
  bit [p_BITS - 1:0] valInt;
  FracNum valFrac;
endclass

class FixedNumberFrac #(int p_BITS = 8, int p_FRAC = 8);
  typedef FixedNumber #(p_BITS) FN;

  bit [p_BITS-1:0] valInt;
  FN::FracNum valFrac;
endclass
```

PR #5826 (2025-03-06): Two-pass package scope resolution
Makes `V3LinkDot` revisit unresolved `::`-type `AstDot`s for a second pass. This helps resolve forward-declared or otherwise initially unknown scopes.

This patch fixes the UVM `component::type_id::create` pattern.

Does **not** fully fix #4705 – in that case, there's an additional problem of the identifier not being tagged as a type identifier during parsing.
closed 2025-03-06 by wsnyder