verilator/verilator #4708: [Bug report] Generated C++ source code compilation failure in very specific path-dependent conditions

flaviens, 2023-11-20:
Hi there!

## Problem

I noticed that in some very specific conditions, Verilator generates C++ code that fails with the following error message:
```
tb.o: in function `main':
undefined reference to `Vtop::Vtop(char const*)'
undefined reference to `Vtop::~Vtop()'
collect2: error: ld returned 1 exit status
```

## Source code

The sources are extremely simple:
tb.cc:
```
#include "Vtop.h"
#include "verilated.h"

typedef Vtop Module;

int main(int argc, char **argv, char **env) {

  Verilated::commandArgs(argc, argv);
  Verilated::traceEverOn(VM_TRACE);

  Module *my_module = new Module;
  delete my_module;
  exit(0);
}
```

top.sv:
```
module top(in_data, out_data);
  input in_data;
  wire in_data;
  output [1503:0] out_data;
  wire [1503:0] out_data;
  assign out_data[0] = in_data;
endmodule
```

## Command

The following command, under specific path conditions that are hard to identify, gives the error mentioned above.
```
verilator --cc --exe --build /some/path/tb.cc /some/path/top.sv -CFLAGS '-I/som/other/path/include -I/some/path -g' --Mdir /some/path/obj_dir
```

## Observation about sensitivity

Surprisingly, if I modify any path, for example an include path or Mdir path, then the error disappears.

## obj_dir in both cases

I provide the obj_dir generated by the command above, and the one generated by the same command where I replaced obj_dir with obj_dir_nobug (in that case, the compilation succeeds).

[obj_dir.zip](https://github.com/verilator/verilator/files/13415283/obj_dir.zip)
[obj_dir_nobug.zip](https://github.com/verilator/verilator/files/13415296/obj_dir_nobug.zip)

The most obvious difference visible in `obj_dir.zip` is that `Vtop__ALL.d` is only containing `Vtop__ALL.o: Vtop__ALL.cpp`.

## Debug logs

Here are also the `--debug` logs from Verilator, although they do not provide much more information.

[cmd.log](https://github.com/verilator/verilator/files/13415365/cmd.log)
[cmd_nobug.log](https://github.com/verilator/verilator/files/13415366/cmd_nobug.log)

## Reproducibility

I've been trying to reproduce this in a Docker container but it is still hard to reproduce as I do not yet clearly understand the causes. On the machine, with the same file and paths, the error always occurs. I hope you will have some ideas.

Thanks!
Flavien

wsnyder, 2023-11-20:
Probably an OS race between the files being written and compiler reading them, a classic problem on NFS-drive mounted filesystems.  Try making a gcc.my that is a shell script with "sleep 5 ; gcc $*" or similar, then have Verilator use that.  If this fixes it we can add some hack to force a NFS flush.

wsnyder, 2023-11-20:
Create ~/bin/gcc_sleep

```
#!/bin/bash
sleep 5
gcc $*
```
Then

```
chmod a+x ~/bin/gcc_sleep
./configure CXX=~/bin/gcc_sleep
[build verilator, then run as usual]

```

flaviens, 2023-11-20:
Yes this fixes as far as I can see :+1:

wsnyder, 2023-11-22:
Please see if the git master fixes it.

wsnyder, 2023-11-22:
Actually, let me make another change too before testing.

wsnyder, 2023-11-22:
Please try latest.  I added a dir flush which has helped in the past, but isn't bulletproof.

flaviens, 2023-11-22:
Hi @wsnyder, thank you for the update! I tried the latest version [fae7f11](https://github.com/verilator/verilator/commit/fae7f11222a5288bd1974a51b58616141c946e23) and I currently cannot reproduce the issue anymore. Thank you again for your reactivity!

flaviens, 2023-11-26:
Hi there! Unfortunately I've still seen another occurrence in spite of the fixes:

```
Starting Verilator 5.019 devel rev v5.018-56-ga7a48221d
- Verilator.cpp:614:  Option --verilate: Start Verilation
- Verilator.cpp:622:  --skip-identical: No change to any source files, exiting
- Verilator.cpp:752:  Start Build
- V3Os.cpp:325:       -Info: File not utimed: /scratch/path/to/obj_dir
- V3Os.cpp:449:       Running system: make -C /scratch/path/to/obj_dir -f Vtop_replaced.mk -j 1
make: Entering directory '/scratch/path/to/obj_dir'
g++     tb_base.o verilated.o verilated_vcd_c.o verilated_threads.o Vtop_replaced__ALL.a    -pthread -lpthread -latomic   -o Vtop_replaced
/usr/bin/ld: tb_base.o: in function `run_test(Vtop*, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
/path/to/tb_base.cc:85: undefined reference to `Vtop::trace(VerilatedVcdC*, int, int)'
/usr/bin/ld: tb_base.o: in function `Vtop::eval()':
/scratch/path/to/obj_dir/./Vtop.h:54: undefined reference to `Vtop::eval_step()'
/usr/bin/ld: tb_base.o: in function `main':
/path/to/tb_base.cc:130: undefined reference to `Vtop::Vtop(char const*)'
collect2: error: ld returned 1 exit status
make: *** [Vtop_replaced.mk:66: Vtop_replaced] Error 1
make: Leaving directory '/scratch/path/to/obj_dir'
- V3Os.cpp:458:       make -C /scratch/path/to/obj_dir -f Vtop_replaced.mk -j 1 returned exit code of 2
%Error: make -C /scratch/path/to/obj_dir -f Vtop_replaced.mk -j 1 exited with 2
%Error: export VERILATOR_ROOT=/data/flsolt/gits/cellift-meta/tools/verilator
%Error: ulimit -s unlimited 2>/dev/null; exec setarch --addr-no-randomize /data/flsolt/gits/cellift-meta/tools/verilator/bin/verilator_bin_dbg --cc --debug --trace --exe --Wno-UNOPTFLAT --Wno-WIDTHTRUNC --Wno-CMPCONST --build /path/to/tb_base.cc /scratch/path/to/top_replaced.sv -CFLAGS -I/path/to/include\ -I/scratch/path/to\ -g --Mdir /scratch/path/to/obj_dir
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec setarch --addr-no-randomize /data/flsolt/gits/cellift-meta/tools/verilator/bin/verilator_bin_dbg --cc --debug --trace --exe --Wno-UNOPTFLAT --Wno-WIDTHTRUNC --Wno-CMPCONST --build /path/to/tb_base.cc /scratch/path/to/top_replaced.sv -CFLAGS -I/path/to/include\ -I/scratch/path/to\ -g --Mdir /scratch/path/to/obj_dir
```

flaviens, 2023-11-26:
- Verilator and the C++ testbench are on a NFS.
- obj_dir and the sources are on a local ext4 disk.
The whole on an Ubuntu 20.04 machine.

wsnyder, 2023-11-26:
I did what I can, I don't know of other tricks besides polling for results which I can't do without changing/wrapping gcc which I don't want to do.

I would recommend not using --build and adding a sleep of at least 5 seconds between Verilating and running make.  I don't want to do this by default as it would slow down normal usage.

Or, don't use a NFS drive.

flaviens, 2023-11-26:
Sounds good thanks!