verilator/verilator #4712: interface is giving internal error

muzafferkal, 2023-11-21:
Hi,
I have an axi interface definition (from another repo which has tests running on xilinx & modelsim tools) and I'm getting a weird internal error:

%Error: Internal Error: axi_intf.sv:4:11: ../V3LinkDot.cpp:423: Module/etc never assigned a symbol entry?
    4 | interface AXI_LITE #(
      |           ^~~~~~~~

Any idea why this is happening and how to get around it? In the meantime I remove the interface but I'd like to understand what's wrong.
Thanks

muzafferkal, 2023-11-21:
It seems this is caused by not connecting the interface at the top level It causes interface definition being rejected and also this error. Is it possible to improve error reporting?

wsnyder, 2023-11-21:
There is an error for a standalone interface, not sure why that is not firing, can you please attach a complete example?

muzafferkal, 2023-11-21:
I'll try to replicate but my issue was: interface definition exists, at module which has the interface as a port exists but there is not interface instantiation. This caused an error at the interface definition and the above error where it's used. When I added the interface instantiation both went away.

wsnyder, 2023-12-20:
Closing - if you get an example feel free to reopen, thanks.

closed 2023-12-20 by wsnyder