verilator/verilator #4713: NUL characters in Vmodexp and then Segmentation Fault

jordankrim, 2023-11-21:
After putting several Verilator pragma comments in my code:
/* verilator lint_off UNOPTFLAT */
/* verilator lint_off WIDTH */
I managed to seemingly compile my testbench, however, there are 2 problems:

1.  There is warning: Vmodexp.mk:13: warning: NUL character seen; rest of line ignored
     And indeed, for some reason the line in question is:

VERILATOR_ROOT = /mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/share/verilator<nul><nul> ...

3. There is a segmentation fault:

...
make[1]: Leaving directory '/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/generators/blocks/src/main/resources/vsrc/rsa/obj_dir'
./obj_dir/Vmodexp
make: *** [Makefile:55: ver_modexp] Segmentation fault (core dumped)

How can I find out the source of this Segmentation fault or is it the nul chars in the Vmodexp?

wsnyder, 2023-11-21:
Verilator doesn't have such a message, it looks like it is coming from make. If you are using --build, try running make outside of verilator and if it still fails look at the file indicated to see if it does have NULs.

jordankrim, 2023-11-21:
This is the line from the Makefile and the result:
make -j -C obj_dir -f Vmodexp.mk Vmodexp
make: Entering directory '/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/generators/blocks/src/main/resources/vsrc/rsa/obj_dir'
Vmodexp.mk:13: warning: NUL character seen; rest of line ignored
make: 'Vmodexp' is up to date.
make: Leaving directory '/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/generators/blocks/src/main/resources/vsrc/rsa/obj_dir'

The file Vmodexp does have NUL chars.  It is unclear to me how to fix it.

wsnyder, 2023-11-21:
I'd first check you aren't passing any NUL's into verilator, e.g. in the command line.  You can run with --stats and it will make a file __stats.txt that will have the command line as Verilator saw it.

Assuming not, Verilator shouldn't have written nulls into Vmodexp.mk, please debug further into Verilator to see where they are coming from.

jordankrim, 2023-11-22:
No file was produced.  Log is:
verilator --cc --clk clk --default-language 1364-2001 -I. ./rtl/modexp.v ./rtl/modexp_core.v ./rtl/montprod.v ./rtl/residue.v ./rtl/adder.v ./rtl/shl.v ./rtl/shr.v ./rtl/blockmem1r1w.v ./rtl/blockmem2r1w.v ./rtl/blockmem2r1wptr.v  ./rtl/blockmem2rptr1w.v ./rtl/divfunc.v --top-module modexp --stats --exe tb_modexp.cpp
make -j -C obj_dir -f Vmodexp.mk Vmodexp
make[1]: Entering directory '/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/generators/blocks/src/main/resources/vsrc/rsa/obj_dir'
**Vmodexp.mk:13: warning: NUL character seen; rest of line ignored**
/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/x86_64-conda-linux-gnu-c++  -I.  -MMD -I/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/share/verilator/include -I/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=c++17 -Os -c -o tb_modexp.o ../tb_modexp.cpp
/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/x86_64-conda-linux-gnu-c++  -I.  -MMD -I/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/share/verilator/include -I/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=c++17 -Os -c -o verilated.o /mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/share/verilator/include/verilated.cpp
/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/perl /mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vmodexp.cpp Vmodexp___024root__DepSet_h2222839b__0.cpp Vmodexp___024root__Slow.cpp Vmodexp___024root__DepSet_h2222839b__0__Slow.cpp Vmodexp__Syms.cpp > Vmodexp__ALL.cpp
/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/x86_64-conda-linux-gnu-c++  -I.  -MMD -I/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/share/verilator/include -I/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=c++17 -Os -c -o Vmodexp__ALL.o Vmodexp__ALL.cpp
echo "" > Vmodexp__ALL.verilator_deplist.tmp
Archive /mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/x86_64-conda-linux-gnu-ar -rcs Vmodexp__ALL.a Vmodexp__ALL.o
/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/x86_64-conda-linux-gnu-c++    tb_modexp.o verilated.o Vmodexp__ALL.a      -o Vmodexp
rm Vmodexp__ALL.verilator_deplist.tmp
make[1]: Leaving directory '/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/generators/blocks/src/main/resources/vsrc/rsa/obj_dir'
./obj_dir/Vmodexp
Initializing interface and resetting core
Reset complete
Input:Hello World!!
Got Here1
-----BEGIN RSA PRIVATE KEY-----
 shift = 6
shift = 6
got here3 and i = 0, out = '0??MIIEogIB'
3082
Header: ASN.1 Sequence
shift = 4
got here3 and i = 1, out = '?'
A202
make: *** [Makefile:55: ver_modexp] Segmentation fault (core dumped)

wsnyder, 2023-11-22:
Please check again.

    ls -l /mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/generators/blocks/src/main/resources/vsrc/rsa/obj_dir/*_stats*

jordankrim, 2023-11-22:
Found the file and at the top is:

Verilator Statistics Report

Information:
  Verilator 4.226 2022-08-31 rev UNKNOWN.REV (mod)
  Arguments: --cc --clk clk --default-language 1364-2001 -I. ./rtl/modexp.v ./rtl/modexp_core.v ./rtl/montprod.v ./rtl/residue.v ./rtl/adder.v ./rtl/shl.v ./rtl/shr.v ./rtl/blockmem1r1w.v ./rtl/blockmem2r1w.v ./rtl/blockmem2r1wptr.v ./rtl/blockmem2rptr1w.v ./rtl/divfunc.v --top-module modexp --stats --exe tb_modexp.cpp

wsnyder, 2023-11-22:
Ok, so you'll need to figure out where the NUL is coming from then inside Verilator.

jordankrim, 2023-11-22:
I am not sure where verilator is inserting the nulls in and I don't really know how to find out, so instead I just am going to fix the gen'ed Makefile (Vmodexp.mk).  Now my Makefile target is:

	verilator --cc --clk clk --default-language 1364-2001 -I. ./rtl/modexp.v ./rtl/modexp_core.v ./rtl/montprod.v ./rtl/residue.v ./rtl/adder.v ./rtl/shl.v ./rtl/shr.v ./rtl/blockmem1r1w.v ./rtl/blockmem2r1w.v ./rtl/blockmem2r1wptr.v  ./rtl/blockmem2rptr1w.v ./rtl/divfunc.v --top-module modexp --stats --exe tb_modexp.cpp
	**tr < ./obj_dir/Vmodexp.mk -d '\000' > ./obj_dir/Vmodexp.mk1**
        **cp ./obj_dir/Vmodexp.mk1 ./obj_dir/Vmodexp.mk**
	make -j -C obj_dir -f Vmodexp.mk Vmodexp
	./obj_dir/Vmodexp

Unfortunately, I still have a segmentation fault and it is reflected in my debug output:

-----BEGIN RSA PRIVATE KEY-----
 shift = 6
shift = 6
got here3 and i = 0, out = '0??MIIEogIB'
3082
Header: ASN.1 Sequence
shift = 4
got here3 and i = 1, out = '?'
A202
 shift = -179293808

This shift value shouldn't be -179293808.  C++ is great until things go out of bounds then it gets ugly to debug. :(

wsnyder, 2023-11-22:
It's certainly true that the NULs will not affect make, the concern is something is strange in Verilator so it's hard to know what else it might be breaking.  You should add printf's or however you prefer to debug to trace back where they are coming from .

But most likely the core dump is another completely unrelated issue.   Perhaps you are just passing illegal inputs?  Compile with VL_DEBUG  (-CFLAGS -DVL_DEBUG) to see if you get warnings.  If nothing obvious then go into gdb and backtrace what is happening.

jordankrim, 2023-11-22:
I got something completely different this time by using your suggestion.  When the sim ran it aborted with an error (shown) related to the clock (clk):

(/mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env) [jkrim@doritos rsa]$ make ver_modexp 2>&1 | tee ver_modexp.log
verilator -CFLAGS -DVL_DEBUG --cc --clk clk --default-language 1364-2001 -I. ./rtl/modexp.v ./rtl/modexp_core.v ./rtl/montprod.v ./rtl/residue.v ./rtl/adder.v ./rtl/shl.v ./rtl/shr.v ./rtl/blockmem1r1w.v ./rtl/blockmem2r1w.v ./rtl/blockmem2r1wptr.v  ./rtl/blockmem2rptr1w.v ./rtl/divfunc.v --top-module modexp --stats --exe tb_modexp.cpp
tr < ./obj_dir/Vmodexp.mk -d '\000' > ./obj_dir/Vmodexp.mk1
cp ./obj_dir/Vmodexp.mk1 ./obj_dir/Vmodexp.mk
make -j -C obj_dir -f Vmodexp.mk Vmodexp
make[1]: Entering directory '/mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/generators/mitll-blocks/src/main/resources/vsrc/rsa/obj_dir'
/mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/x86_64-conda-linux-gnu-c++  -I.  -MMD -I/mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/share/verilator/include -I/mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_DEBUG  -std=c++17 -Os -c -o tb_modexp.o ../tb_modexp.cpp
/mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/x86_64-conda-linux-gnu-c++  -I.  -MMD -I/mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/share/verilator/include -I/mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_DEBUG  -std=c++17 -Os -c -o verilated.o /mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/share/verilator/include/verilated.cpp
/mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/perl /mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vmodexp.cpp Vmodexp___024root__DepSet_h2222839b__0.cpp Vmodexp___024root__Slow.cpp Vmodexp___024root__DepSet_h2222839b__0__Slow.cpp Vmodexp__Syms.cpp > Vmodexp__ALL.cpp
/mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/x86_64-conda-linux-gnu-c++  -I.  -MMD -I/mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/share/verilator/include -I/mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_DEBUG  -std=c++17 -Os -c -o Vmodexp__ALL.o Vmodexp__ALL.cpp
echo "" > Vmodexp__ALL.verilator_deplist.tmp
Archive /mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/x86_64-conda-linux-gnu-ar -rcs Vmodexp__ALL.a Vmodexp__ALL.o
/mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/x86_64-conda-linux-gnu-c++    tb_modexp.o verilated.o Vmodexp__ALL.a      -o Vmodexp
rm Vmodexp__ALL.verilator_deplist.tmp
make[1]: Leaving directory '/mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/generators/mitll-blocks/src/main/resources/vsrc/rsa/obj_dir'
./obj_dir/Vmodexp
Initializing interface and resetting core
%Error: unknown:0: Testbench C set input 'clk' to value that overflows what the signal's width can fit
Aborting...
make: *** [Makefile:57: ver_modexp] Aborted (core dumped)

wsnyder, 2023-11-22:
It is as the message says, an illegal value is driven into the clk from C++, so it is not a Verilator problem, fix that.

Closing, we'll deal with NUL if/when debug further, but feel free to comment more.

closed 2023-11-22 by wsnyder

jordankrim, 2023-12-04:
Finally got around to this again.  I got around the problem with:
%Error: unknown:0: Testbench C set input 'clk' to value that overflows what the signal's width can fit
The code in tb_modexp.cpp was:
top->clk = ~top->clk;
I changed it to:
top->clk = !top->clk);
This fixed it (but I have no idea why) and now I am right back to the seg fault.  So where do I go from here?

wsnyder, 2023-12-05:
>This fixed it (but I have no idea why)

Because you had a e.g. 1 bit signal which Verilator takes as only using bit 0 of an e.g. 32-bit integer, when you say "~" in C you are setting buts 31 downto 1 to ones, which is non-sensical to Verilator as that's outside the legal values of the signal.

If you still get a segfault, then run the compiler with debug symbols and no optimization, and see where "gdb" says the fault is coming from.  If the fault goes away, try keeping the default optimization.

jordankrim, 2023-12-05:
I was able to debug the seg fault problem using printf statements, however, for some strange reason, I can comment out all my "debug" printf statements except one.  When I comment that one out there is a failure.  It is not a seg fault but the testbench just misbehaves thus going to a state that it should not:

Header: ASN.1 Sequence
Seperator: Integer -> 1 bytes long
version: 
7c 
Seperator: Invalid
make: *** [Makefile:64: vcd] Error 1

Line 64 in Makefile is just the execution line:
./obj_dir/Vmodexp

The way it should run is this (Notice the debug printf which is required to not error out):

Header: ASN.1 Sequence
In read: shift = 4
Seperator: Integer -> 1 bytes long
In read: shift = 3
version: 
In read: shift = 1
Seperator: Integer -> 257 bytes long
In read: shift = 5
priModulus: 
...

wsnyder, 2023-12-05:
Try compiling with the address sanitizer `-CFLAGS -fsanitize=address,undefined -LDFLAGS -fsanitize=address,undefined`

jordankrim, 2023-12-05:
That resulted in no compile:
...
collect2: error: ld returned 1 exit status
make[1]: *** [Vmodexp.mk:66: Vmodexp] Error 1
rm Vmodexp__ALL.verilator_deplist.tmp
make[1]: Leaving directory '/mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/generators/mitll-blocks/src/main/resources/vsrc/rsa/obj_dir'
make: *** [Makefile:63: vcd] Error 2

wsnyder, 2023-12-05:
Presumably there was an error line before the "exist status"?  that's the important part.

jordankrim, 2023-12-06:
I don't see it.  There are 65K lines of output, so didn't want to paste them all here so did grep for error (-i) in my log output and got this:

/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/../lib/gcc/x86_64-conda-linux-gnu/12.1.0/../../../../x86_64-conda-linux-gnu/bin/ld: verilated.o: in function `VerilatedContext::errorCount(int)':
/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/../lib/gcc/x86_64-conda-linux-gnu/12.1.0/../../../../x86_64-conda-linux-gnu/bin/ld: verilated.o: in function `VerilatedContext::errorCountInc()':
/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/../lib/gcc/x86_64-conda-linux-gnu/12.1.0/../../../../x86_64-conda-linux-gnu/bin/ld: verilated.o: in function `VerilatedContext::errorLimit(int)':
/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/../lib/gcc/x86_64-conda-linux-gnu/12.1.0/../../../../x86_64-conda-linux-gnu/bin/ld: verilated.o: in function `VerilatedContext::fatalOnError(bool)':
/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/../lib/gcc/x86_64-conda-linux-gnu/12.1.0/../../../../x86_64-conda-linux-gnu/bin/ld: verilated.o: in function `VerilatedContext::fatalOnVpiError(bool)':
/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/../lib/gcc/x86_64-conda-linux-gnu/12.1.0/../../../../x86_64-conda-linux-gnu/bin/ld: verilated.o: in function `VerilatedContext::gotError(bool)':
/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/../lib/gcc/x86_64-conda-linux-gnu/12.1.0/../../../../x86_64-conda-linux-gnu/bin/ld: verilated.o: in function `VL_FERROR_IN(unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)':
/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/../lib/gcc/x86_64-conda-linux-gnu/12.1.0/../../../../x86_64-conda-linux-gnu/bin/ld: verilated.o: in function `VerilatedScope::exportFindError(int) const':
/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/../lib/gcc/x86_64-conda-linux-gnu/12.1.0/../../../../x86_64-conda-linux-gnu/bin/ld: verilated.o: in function `VerilatedScope::exportFindNullError(int)':
/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/../lib/gcc/x86_64-conda-linux-gnu/12.1.0/../../../../x86_64-conda-linux-gnu/bin/ld: verilated.o: in function `Verilated::overWidthError(char const*)':
/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/../lib/gcc/x86_64-conda-linux-gnu/12.1.0/../../../../x86_64-conda-linux-gnu/bin/ld: verilated.o: in function `Verilated::nullPointerError(char const*, int)':
collect2: error: ld returned 1 exit status
make[1]: *** [Vmodexp.mk:66: Vmodexp] Error 1
make: *** [Makefile:63: vcd] Error 2

wsnyder, 2023-12-06:
Those are the "end" of an earier error, what's the first error?

jordankrim, 2023-12-06:
There are thousands of lines like this:  
/mnt/data/projects/project/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/../lib/gcc/x86_64-conda-linux-gnu/12.1.0/../../../../x86_64-conda-linux-gnu/bin/ld: Vmodexp__ALL.cpp:(.text+0x132ba): undefined reference to `__ubsan_handle_type_mismatch_v1'  
Are those the errors?

jordankrim, 2023-12-06:
I did a clean as you suggested but got exactly the same result.  I just don't see any error other than the ld one at the end.  I can send you privately the entire 65K lines of output (log).  But other than that, I don't know where the error is.

jordankrim, 2023-12-07:
I attempted to use the guide you suggested, unfortunately, doing a "normal" compile results in a seg fault, but when I compile it for gdb and invoke gdb, then file ./obj_dir/Vmodexp then run, it ends like this:

...
Seperator: Invalid
[Inferior 1 (process 317050) exited with code 01]
Missing separate debuginfos, use: yum debuginfo-install glibc-2.28-236.el8.7.x86_64
(gdb) bt
No stack.

Is this because of the Missing separate debuginfos message?

wsnyder, 2023-12-08:
>Is this because of the Missing separate debuginfos message?

Likely; install what it says.

Is your OS distro something standard and stable?  If you're using something not well tested that would explain all of this.  Try it on e.g. Ubuntu 22.04 LTS alternatively if you can.

jordankrim, 2023-12-08:
I ran gdb but somehow it does not have the correct pointers to files:
Breakpoint 1, main (argc=1, argv=0x7fffffffbf38, env=0x7fffffffbf48)
    at /mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/share/verilator/include/RSA.h:63
63	/mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/share/verilator/include/RSA.h: No such file or directory.
The only files in verilator/include are named verilated*

wsnyder, 2023-12-09:
For info on setting gdb source paths see here: https://sourceware.org/gdb/current/onlinedocs/gdb.html/Source-Path.html

jordankrim, 2023-12-11:
I think I have found the problem with the seg fault.  I didn't get around to retrying gdb, but instead honed in on where it was occuring and suddenly realized that an array was set to three elements, however 0 to 3 were being accesses, obviously 3 was out of bounds.  Making the array size 4 appears to have solved the issue.