verilator/verilator #4714: unsupported stream operation

muzafferkal, 2023-11-21:
I'm using existing IP from PULP which has the following construct:

    task set_ir(input logic [IrLength-1:0] opcode);
      logic opcode_unpacked [IrLength];
      // check whether IR is already set to the right value
      if (this.ir_select == opcode) return;
      {<<{opcode_unpacked}} = opcode;

The unpacking stream operation gives the following error:

%Error-UNSUPPORTED: jtag_test.sv:56:8: Unsupported: Stream operation on a variable of a type 'logic$[0:4]'

Any idea why this is happening and how to get around it?

wsnyder, 2023-11-21:
Stream operation on unpacked arrays are not supported.  Perhaps the code can be changed not to use an unpacked array, or please create a standalone test case to show and test the proper behavior here, and ideally make pull request to fix that.

muzafferkal, 2023-11-21:
I am looking at these tests which use the stream operators pretty extensively: https://github.com/pulp-platform/riscv-dbg/blob/master/tb/jtag_dmi/jtag_test.sv and it seems it works with Modelsim & Xilinx simulators.
I'll look for a different testbench for my needs.

fuad1502, 2024-03-17:
Hi @wsnyder , I would like to handle this issue if that's alright 👍

fuad1502, 2024-03-17:
My idea is to add `AstCvtUnpackedToPacked` and `AstCvtPackedToUnpacked` and the corresponding macro. Just like how dynamic arrays are handled in `V3Const.cpp` line 2159:2232. 

Would that approach work? And if it does, do you think that's a good approach for handling this?

wsnyder, 2024-03-17:
Yes, that seems reasonable, thanks for looking into this.

closed 2024-03-21 by wsnyder