verilator/verilator #4716: Struct On Module Ports Problem

Entwinedime, 2023-11-23:
When I use a struct type as a module port, it cannot be compiled.
I used `verilator --cc --build STRUCT_TEST.sv` as parameter.
Looks like errors result in cpp files.
[image]

wsnyder, 2023-11-23:
Unpacked structure are not supported as primary i/o hooking to C++.  I will add an unsupported error for this.

If this is just a little test of some larger broken example where the struct is not primary i/o please post it as text.

wsnyder, 2023-11-23:
Note as workaround use a packed struct.

Entwinedime, 2023-11-23:
OK, thank you very much