verilator/verilator #4717: Wrong result when a signal is divided by itself

flaviens, 2023-11-24:
Hi there!

The IEEE 1800-2017 standard says:

> For the division or modulus operators, if the second operand is a zero, then the entire result value shall be x.

Given that Verilator represents x by zero, I would expect to get 0 every time a division by zero occurs.
However, I noticed that dividing a number by itself gives 1, which is incorrect if my understanding is correct.
An occurrence can be as simple as below, with the assignment `a=0`:

```
module toplevel (
  input logic a,
  output logic y
);
  assign y = a / a;
endmodule
```

By curiosity, I added the flags `-fno-const-bit-op-tree -fno-dfg-peephole -fno-dfg-pre-inline -fno-dfg-post-inline -fno-gate -fno-subst -fno-subst-const` but it did not erase the buggy behavior.

I'm using `Verilator 5.018 2023-10-30 rev v5.004-74-g41e5f6b7d`.

Thanks!
Flavien

gezalore, 2023-11-24:
I suspect 
https://github.com/verilator/verilator/blob/fae7f11222a5288bd1974a51b58616141c946e23/src/V3Const.cpp#L3580-L3581

`-fno-const` might change the result.

wsnyder, 2023-11-24:
Verilator doesn't currently make guarantees about what value is used in place of X (outside of the -x-initial and -x-assign options).  As @gezalore pointed out, it currently chooses to put in a 1, as otherwise it would need to make a runtime check for the variable rather than optimizing.  So my preference is not to change this.  If we did, perhaps we need a new --x-result option defaulting to "fast".

flaviens, 2023-11-24:
Thank you for your response! `-fno-const` does not seem to change the result.

wsnyder, 2023-11-24:
In gcc at least with -O2, "a/a" optimizes to "1" also.

gezalore, 2023-11-24:
That's because C division by zero is UB, so they are free to do anything, while in SV it's defined but we do something slightly inconsistent.

wsnyder, 2023-11-24:
That's because C division by zero is UB, so they are free to do anything, while in SV it's defined but Verilator chooses to do what's fastest because it doesn't support X's.

closed 2023-11-26 by wsnyder