verilator/verilator #4718: IEEE 1800-2023

wsnyder, 2023-11-24:
IEEE plans on approving 1800-2023 early December, with publication around Feb/Mar.

I have several related pulls ready in a private area.  Currently the draft standard is only available to certain organizations (e.g. committee contributors), so it is not appropriate to release these enhancement pulls publicly yet.

If anyone is working or expects to work on 1800-2023 features before the IEEE release, please drop a note here first in order to prevent duplication of work.  Thanks

wsnyder, 2023-11-24:
Summary of changes by Dave Rick at DVcon:

https://dvcon-proceedings.org/document/whats-next-for-systemverilog-in-the-upcoming-ieee-1800-standard/
https://dvcon-proceedings.org/document/whats-next-for-systemverilog-in-the-upcoming-ieee-1800-standard-2/

wsnyder, 2024-03-02:
I released the base changes for 1800 including preprocessing.  Remaining features will be handled as needed with separate issues.

The standard is available here [IEEE 1800-2023](https://ieeexplore.ieee.org/browse/standards/get-program/page/series?id=80)

closed 2024-03-02 by wsnyder

gezalore, 2024-03-03:
It's probably wishful thinking but if anyone is aware of a redline version of the 2023 version vs 2017 it would be very helpful.