verilator/verilator #4719: Error when Verilator tries to convert some signals to 32-bit integers

flaviens, 2023-11-25:
Hi there!

I think I found an issue in Verilator. When trying to verilate the source below against a simple C++ testbench, I get this error:

```
%Error: /path/to/top.sv:14:125: Value too wide for 32-bits expected in this context 98'h3ffffffffffffffffffffffff
   14 |   assign { _03_[97:88], out_data[63:32], _03_[55:0] } = { _01_, in_data, _05_[0] } >> { _04_[97:94], _02_, _04_[92:76], _01_, _04_[74:0] };
      |                                                                                                                             ^
%Error: Exiting due to 1 error(s)
```

top.sv:
```
module top(in_data, out_data);
  wire _00_;
  wire _01_;
  wire _02_;
  wire [97:0] _03_;
  wire [97:0] _04_;
  wire [97:0] _05_;
  wire [218:0] _06_;
  input [95:0] in_data;
  wire [95:0] in_data;
  output [95:0] out_data;
  wire [95:0] out_data;
  assign _00_ = _02_ | ~(_01_);
  assign { _03_[97:88], out_data[63:32], _03_[55:0] } = { _01_, in_data, _05_[0] } >> { _04_[97:94], _02_, _04_[92:76], _01_, _04_[74:0] };
  assign { _06_[218:136], _04_[97:94], _02_, _04_[92:76], _01_, _04_[74:0], _06_[37:0] } = { in_data[46:20], in_data, in_data } ~^ { in_data[90:64], in_data, in_data };
  assign out_data[0] = ~((in_data[52] & _05_[0]) | _00_);
  assign _05_[0] = ~((in_data[24] & in_data[16]) | (in_data[16] & in_data[42]));
  assign _03_[87:56] = out_data[63:32];
  assign { _04_[93], _04_[75] } = { _02_, _01_ };
  assign _05_[97:1] = { _01_, in_data };
  assign _06_[135:38] = { _04_[97:94], _02_, _04_[92:76], _01_, _04_[74:0] };
endmodule
```

Interestingly,
- Changing `>> { _04_[97:94], _02_, _04_[92:76], _01_, _04_[74:0] }` with `_04_` moves the error to the next line.
- Changing it with `_05_` removes the error.

Thanks!

wsnyder, 2023-11-25:
I wonder if is related to #4709

Perhaps you can use --debug, look at the tree file to see if the widths of the expressions are correct around the relevant lines, and then attempt a pull request to fix this?

flaviens, 2023-11-26:
Thank you @wsnyder for your response! I digged a bit in Verilato's sources but I'm somehow unfamiliar with it at the moment ^^'. Looking at the tree files, I could not find a striking mismatch, yet it may also be due to a lack of expertise there.

Yet I found a super simple design that causes the issue:
```
module top(in_data, out_data);
  input [64:0] in_data;
  wire [64:0] in_data;
  output out_data;
  wire out_data;
  assign out_data = in_data <<< ~(in_data ^ in_data);
endmodule
```

Additionally, I could find out that the error occurs at `constifyAllLint`, in `V3Ast.cpp`
```
        // Calling on top level; we know the netlist won't get replaced
        nodep->accept(v);
```
Then, the `accept`/`visit` calls are somehow hard for me to track a bit blindly. Yet I hope this information helps!

flaviens, 2023-11-26:
Thank you @wsnyder for your response! In this case it does not seem to matter. The error happens for any size of out_data, including 1 and 32 bits.

flaviens, 2023-11-27:
Thank you! That's it indeed. However, the standard does not seem to limit this size. In the case of that huge constant, couldn't one hardwire the output to zero (except for arithmetic right shift where we would replicate the MSB)?
Thanks!

wsnyder, 2023-11-28:
Fair point, fixed.