verilator/verilator #4722: More functionality in `-f` argument files

sifferman, 2023-11-28:
Hello!

Verilator has `-f` and `-F` to read in an arguments file. however, it does not support ``` `` ```. I ran into this limitation while trying to do

```plain
`yosys-config --datdir`/simcells.v
`yosys-config --datdir`/simlib.v
```

VCS's `-file filename` option works with the above example, and can handle all "escape characters and metacharacters". (I'm not sure if I am allowed to include longer snippets from `vcs --help`. I can if it would be helpful). I assume VCS offloads this functionality to a shell to parse the file.

I wonder if instead of using a custom parser (as what is being done in [`V3Os::filenameSubstitute()`](https://github.com/verilator/verilator/blob/7167d4bdc0054943e140cc427cb2f54434cd9bd7/src/V3Os.cpp#L205-L247) to elaborate `$`), Verilator should read from `popen(("sh -c 'echo \"" + file_contents + "\"'").c_str(), "r")`.

If you feel this would be a valuable addition to Verilator, I am happy to work on this! Though I may want some more help brainstorming solutions to make this as cross-platform as possible.

<details open>
<summary>Proof of concept:</summary>
<br>

```c++
#include <string>
#include <fstream>
#include <iostream>

std::string parse(std::string filename) {
    // read in file
    std::ifstream ifs{filename};
    if (!ifs)
        std::runtime_error("Could not file.");
    std::string file_contents((std::istreambuf_iterator<char>(ifs)), std::istreambuf_iterator<char>());
    ifs.close();

    // sanitize input by prefacing all quotes with backslashes
    std::string sanitized_contents = file_contents;
    for (int i = sanitized_contents.length()-1; i >= 0; i--) {
        if (sanitized_contents[i] != '"') continue;
        if (i != 0 && sanitized_contents[i-1] == '\\') continue;
        sanitized_contents.insert(i, "\\");
    }

    // open pipe
    std::string parsed_contents;
    // Use sh to parse environment variables, escape characters, and metacharacters
    FILE* pipe = popen(("sh -c 'echo \"" + sanitized_contents + "\"'").c_str(), "r");
    if (!pipe)
        std::runtime_error("Could not open pipe.");

    // read from pipe
    char buffer[128];
    while (fgets(buffer, sizeof(buffer), pipe) != nullptr)
        parsed_contents += buffer;
    if(!feof(pipe))
        std::runtime_error("Error reading from pipe.");
    pclose(pipe);

    // return
    return parsed_contents;
}

int main() {
	std::cout << parse("test.txt") << std::endl;
}
```

</details>

wsnyder, 2023-11-29:
Can you please try the example also on at least 2 other simulators e.g. NC?

sifferman, 2023-11-29:
I tested

* Xcelium `xrun(64): 19.09-s010`
* ModelSim SE-64 `vlog 10.7d Compiler 2019.02`

and neither seem to support this.

wsnyder, 2023-11-29:
Ok, in that case I'd prefer not to add the risk/complexity at this time, is that ok?

I'd suggest using an environment variable to pass in the relative location, which I think all simulators support.

sifferman, 2023-11-29:
Yeah, I think that makes better sense. All resolved!

closed 2023-11-29 by sifferman