verilator/verilator #4724: Double bug report: VCD corruption and optimization error for 5 optimization types

flaviens, 2023-11-29:
Hi there!

I found a double bug for Verilator on the same design.
- The VCD trace is corrupted
- The value is wrong, even if traces are enabled. Note that the bug is erased if I manually connect all signals to the output. Icarus matches with the no-optimization result.

[I created a repository for you to reproduce the issue](https://github.com/flaviens/verilator-b6).

The design is relatively complex which may make debugging somehow complex, yet there is hope that either the optimization bug is tightly coupled to the VCD trace corruption, or that a non-corrupted trace could further help.

Regarding disabling the optimizations, here is a summary:

| Option                     | Value   |
|----------------------------|---------|
| -fno-acyc-simp             | 244     |
| -fno-assemble              | 244     |
| -fno-case                  | 244     |
| -fno-combine               | 244     |
| -fno-const                 | 244     |
| -fno-const-bit-op-tree     | 1403003 |
| -fno-dedup                 | 244     |
| -fno-dfg                   | 244     |
| -fno-dfg-peephole          | 244     |
| -fno-dfg-peephole          | 244     |
| -fno-dfg-pre-inline        | 244     |
| -fno-dfg-post-inline       | 244     |
| -fno-expand                | 1403003 |
| -fno-gate                  | 1403003 |
| -fno-inline                | 244     |
| -fno-life                  | 244     |
| -fno-life-post             | 244     |
| -fno-localize              | 244     |
| -fno-merge-cond            | 244     |
| -fno-merge-cond-motion     | 244     |
| -fno-merge-const-pool      | 244     |
| -fno-reloop                | 244     |
| -fno-reorder               | 244     |
| -fno-split                 | 244     |
| -fno-subst                 | 1403003 |
| -fno-subst-const           | 1403003 |
| -fno-table                 | 244     |
| -fno-all                   | 1403003 |

I'm using `Verilator 5.019 devel rev v5.018-58-g7167d4bdc` on Ubuntu 20.04.

Thanks!
Flavien

wsnyder, 2023-11-29:
Thanks for the report and example.

As with your other issues it would help greatly to make a small test_regress test, and look yourself into a pull request to fix the issue(s). It's very difficult otherwise to find time to debug large test cases.

wsnyder, 2024-01-18:
Please convert to test_regress format for us to further analyze this bug, there's too much overhead in understanding your harness otherwise, thanks.

wsnyder, 2024-04-13:
If get small reproducer please reopen or refile.

closed 2024-04-13 by wsnyder