verilator/verilator #4730: Missing precompiled hearder file for gcc

Phantom1003, 2023-12-01:
After upgrading gcc to version 12.3.0, I discovered that the compile commands generated by verilator contain a missing file.

```
$ ccache g++ ... -fcoroutines -include VTestbench__pch.h.slow -c -o VTestbench__Slow.o VTestbench___Slow.cpp
<command-line>: fatal error: VTestbench__pch.h.slow: No such file or directory

$ ls VTestbench__pch.h.slow*
VTestbench__pch.h.slow.d  VTestbench__pch.h.slow.gch
```
The file `VTestbench__pch.h.slow` does not exist. The only file present is `VTestbench__pch.h.slow.gch`. However, according to the configuration, the file with a `gch` suffix should only be generated when the user uses clang.

https://github.com/verilator/verilator/blob/feae9ca4aac4e0ede4af718e80aaed460b998936/configure.ac#L572-L579

https://github.com/verilator/verilator/blob/feae9ca4aac4e0ede4af718e80aaed460b998936/include/verilated.mk.in#L184-L185

It appears that the older version of GCC (11.4.0) ignores these missing headers. I am planning to resolve this issue and would like to confirm if it is acceptable for both GCC and Clang to use the same precompiled headers. Thanks.

wsnyder, 2023-12-01:
The files themselves can't be shared, but the assumption is a "make clean" will be done when switching compilers.

GCC and CLANG can use the same CFG_GCH_IF_CLANG setting.  However, please rename this variable, and make sure it works correctly on older versions.  A better way to have configure determine this is for it to make an experiment to see what filename is created and set the variable appropriately, then it will be totally version-agnostic.

Phantom1003, 2023-12-02:
I'm not reproducing the error on ubuntu23, it looks like there was a problem with my previous environment, I'll close this first.

closed 2023-12-02 by Phantom1003