verilator/verilator #4732: Compilation Error when Trying to Print an Unpacked Array with ‘%p’

kdstrike, 2023-12-01:
Verilator produces a compilation error when trying to print an unpacked array with `%p`. This issue is not seen when simulating with Questa.

### Example file `my_file.sv`
```
module my_file;

localparam integer unpacked_array [1:0] = '{0, 1};

function integer foo();
    $fatal(1, "%p", unpacked_array);
    return 0;
endfunction

localparam integer return_val = foo();

initial begin
    $finish;
end

endmodule // my_file
```

The command to execute is `verilator --top-module my_file --binary --assert -Wall -Wno-fatal --timescale 1ns/1ps --trace --trace-structs my_file.sv`. The error is shown below.
```
%Error: my_file.sv:10:33: Expecting expression to be constant, but can't determine constant for FUNCREF 'foo'
                                                                                               : ... In instance my_file
        my_file.sv:6:5: ... Location of non-constant CEXPR: Unknown node type, perhaps missing visitor in SimulateVisitor
        my_file.sv:10:33: ... Called from foo() with parameters:
   10 | localparam integer return_val = foo();
      |                                 ^~~
```

The expected behavior is that there would be no compilation error and the value of the array would be printed out. As an example, this is the output when the same code is simulated with Questa.
```
# ** Fatal: '{0, 1}
#    Time: 0 ns  Scope: my_file.foo File: my_file.sv Line: 6
```

wsnyder, 2023-12-01:
Currently Verilator doesn't implement unpacked structure %p dumping at compilation/elaboration time, which is the reason for this error.

We'd welcome a pull request to improve this, I can go into more details if you'd like to work on it.

kdstrike, 2023-12-05:
Yes could you please provide more details?