verilator/verilator #4738: Compile a c++ testbench into a shared library

viniul, 2023-12-05:
Let's assume I have a c++ testbench which uses internal Verilator functions, I would like to compile this into a shared library rather an executable. 

Assume the following minimal example:
```
mkdir test_our
cd test_our

cat >our.v <<'EOF'
  module our;
     initial begin $display("Hello World"); $finish; end
  endmodule
EOF

cat >sim_main.cpp <<'EOF'
  #include "Vour.h"
  #include "verilated.h"
  int main(int argc, char** argv) {
      VerilatedContext* contextp = new VerilatedContext;
      contextp->commandArgs(argc, argv);
      Vour* top = new Vour{contextp};
      while (!contextp->gotFinish()) { top->eval(); }
      delete top;
      delete contextp;
      return 0;
  }
EOF
```
Compiled with:
```
verilator --cc --exe --build -j 0 -Wall sim_main.cpp our.v
```

Instead of compiling this function as an executable, I would like for it be compiled into a static library that I can then link into another codebase. Something like this:
```
 #include "Vour.h"
  #include "verilated.h"
  int simulate(int argc, char** argv) {
      VerilatedContext* contextp = new VerilatedContext;
      contextp->commandArgs(argc, argv);
      Vour* top = new Vour{contextp};
      while (!contextp->gotFinish()) { top->eval(); }
      delete top;
      delete contextp;
      return 0;
  }
```
And then the library should expose the `simulate` symbols. Is this possible? If so, is there a specific verilator flag I can use to do so?

wsnyder, 2023-12-05:
There is no flag to do this, but you can treat the model like any other C++ program and call the linker appropriately to create a shared library - search for other examples on the web.

Note if you might be loading two or more models that are both Verilated you need to make sure they were compiled using identical Verilator "include/" C++ files, or bad things may happen.

Closing as answered but feel free to follow up.

closed 2023-12-05 by wsnyder

viniul, 2023-12-05:
Hi @wsnyder ,

Thank you so much! I tried this path, but I believe I am missing a library that I do not know where to find. My command is:
```
g++ -I/usr/share/verilator/include/ -Iobj_dir test.cpp obj_dir/Vour__ALL.o -o test
```
which results in (among others) the following error:
``
ndefined reference to `VerilatedSyms::VerilatedSyms(VerilatedContext*)'
```

Could you point me to the library I need to include? I could not find it through Google..

Thank you so much again!

wsnyder, 2023-12-05:
You need to link against all of the .cpp files listed in the generated .mk file, e.g. verilated.cpp and others.