verilator/verilator #4749: Addtional c/c++ is silently ignored without `--exe`

FanShupei, 2023-12-10:
I'm verilating my model to library, since the model use DPI-C, it contains both SystemVerilog and C/C++ code. However, I found that the generated `Vtop__ALL.a` seems not contain any C/C++ code, leading to mysterious linking error...  

After some investigation, I found it seems verilator didn't process any C/C++ inputs at all.

# Reproduce

Prepare two file `top.sv` and `additional.cpp`. The `additional.cpp` contains grammar error (so it cannot compile).

Run the following command
```
verilator --cc top.sv additional.cpp
make -C obj_dir -f Vtop.mk
```

It should not compile, since `additonal.cpp` contains grammar error. However, it does generate `Vtop__ALL.a` without any error, which means `additonal.cpp` is silently ignored.

# Analysis

It seems in `Vtop.mk`, the corresponding rule for `additional.o` is generated, but it's not triggered in building. Sorry I'm not an Makefile expert, so I didn't investigate more.
```
...
...
additional.o: additional.cpp
	$(OBJCACHE) $(CXX) $(CXXFLAGS) $(CPPFLAGS) $(OPT_FAST) -c -o $@ $<
```

# Version

The problem exists in both stable (`5.018`) and master (`Verilator 5.019 devel rev v5.018-80-gb60117c71`).

I test it on Arch Linux.

wsnyder, 2023-12-10:
The library rule currently does not include user .cpp files.  They are added with "--exe" into the executable.  Similar rule applies to any .a or .o's provided on the command line.

This rule was historically how it worked, also it matches the rule for ".a" that are on the command line that can only be done in the final link.

Note this issue is mistitled, your example used -cc, it works with --exe.  If your request to add the .cpp's when making the archive?

FanShupei, 2023-12-10:
Thanks for your explanation. Now I fully understand what happens.

Yes, I'm requesting to add the .cpp's when making the archive.

In my use case, main part of the program is not written in C/C++. In the build process, verilator is invoked as an external command, by `verilator --build --cc ...`. The C/C++ contains DPI-C code and FFI glue. If all code of verialted model is in the archive, it will simplify the build process.

Note the current behavior is a bit footgun especially when using `verilator --build --cc xxx.sv yyy.cpp`. In this case, since it successfully generates a `Vxxx__ALL.a` but `yyy.o` is not generated, users may expect the archieve contains everything, but it's not. This is why I say C/C++ inputs are "silent ignored" in the original title.

wsnyder, 2023-12-10:
That makes sense. Presumably:

- .cpp/c/cc/o: Add to library
- .a/so: Add to --exe

Might you contribute a pull request to implement this?  Please update the tests and docs/exe_verilator.rst documentation on the options appropriately too,

FanShupei, 2023-12-11:
Yes, I'm willing to working on a pull request.

My rough idea is:
- In exe mode, left as is.
- In lib mode, the final achieve `prefix__ALL.a` should addtionally contains .cpp/c/cc/o

Verilator support a bunch of features like `--hierarchical`, `--lib-create`, `--protect-lib`, they all interact with the build system. I think I may spend time to have a complete understanding of the build system for verilated model, then I will do the actual work.

FanShupei, 2023-12-13:
I almost have a draft pull request, but I'd like to discuss my proposal here first. Since it technically may contain breaking change.

The verilator contains three modes:
- exe mode: with `--exe` specified, Verilator will create a executable.
- dpi library mode: with `--lib-create` specified, Verilator will create a '.o' and a '.so' with dpi library wrapper.
- default lib mode: wihout `--exe` nor `--lib-create`, it will generate `{prefix}__ALL.a` .

For first two modes, it has wel-defined use cases and Verilator controls the build process. Generally users only care the build artifact, not the build process. I have no intention to change behaviors of the two modes. 

For default lib mode, it's quite different. It seems have no intended use case. It just exposes internals of verilated Makefile. The generated `{prefix}__ALL.a` is not very useful, it lacks `VK_USER_OBJS` and `VK_GLOBAL_OBJS`. I even suspect whether it has real users...

I **propose** to change the behavior of default lib mode: it will generate `lib{prefix}.a` containing everything. It's useful for users using external build system. Roughly speaking, it behave like `--lib-create`, but generates only '.a' and without dpi wrapper. It will put one more line in `{prefix}.mk`: \
`lib{PREFIX}.a: $(VK_USER_OBJS) $(VK_GLOBAL_OBJS) $(VK_OBJS) $(VM_HIER_LIBS)`

Any comments for the proposal? Is there any backward compatibility concerns? 

I also have some additional thoughts:
1. How should we handle `VK_GLOBAL_OBJS`? It makes trouble when we link two verilated models together.
2. For really advanced users, Verilator may have an option to generate a build plan in xml/json? At least it's a better choice for my use case.

wsnyder, 2023-12-13:
The default lib mode is certainly used.  Such a created library is fed into users' own makefiles that were expected to contain their own rules that use VK_USER_OBJS and VK_GLOBAL_OBJS.  However I suspect there is little if any use of this with command-line provided .cpp/.o and VK_USER_OBJS, since when writing an upper makefile the user could have more directly referenced whatever .cpp/o they wanted directly.

Generating lib{prefix}.a seems reasonable.  Please leave the existing __ALL rule for compatibility.

>How should we handle VK_GLOBAL_OBJS? It makes trouble when we link two verilated models together.

This is why it isn't in the existing .a file, an why we want to leave the existing rules ;)

>For really advanced users, Verilator may have an option to generate a build plan in xml/json? At least it's a better choice for my use case.

I'm open to a separate pull that does that; we are intending to move to json over XML so please use json format; see #4715.  If that means you don't need this fixed than that's fine too.

FanShupei, 2023-12-13:
In the long run, I prefer Verilator could output a build plan, but it's a larger change and contains a lot of design work. I plan to work on it later.

For this issue, since currently my workflow is somewhat blocked by this and I should it should be a improvement for usability. I just send a simple pull request for it.

PR #4754 (2023-12-23): Add user C/C++ code to final achieve in default lib mode (#4749)
Close #4749.

The first commit is an independent small cleanup. We should use `$(VK_OBJS) $(VM_HIER_LIBS)` directly.

The second commit is the main work. 

Basically it changes make rules in default lib mode as follows:
```
default: lib{prefix}
lib{prefix}.a: $(VK_USER_OBJS) $(VK_OBJS) $(VM_HIER)
libverilated_rt.a: $(VK_GLOBAL_OBJS)
lib{prefix}: lib{prefix}.a libverilated_rt.a
```

For users who write their own Makefile, they shall not be affected. I don't do any backward incompatible change here.
For users who invokes Makefile d [... cut]
closed 2023-12-23 by wsnyder