verilator/verilator #4753: "Replication value of 0" error inside unreachable code

KireinaHoro, 2023-12-12:
Essentially the same issue as #2866 #2625 and potentially more, Verilator complains about replication value of 0 inside unreachable code (example from [here](
https://github.com/alexforencich/verilog-axi/blob/38915fb5330cb8270b454afc0140a94489dc56db/rtl/axi_adapter_wr.v#L356-L358)):

```verilog
parameter EXPAND = M_STRB_WIDTH > S_STRB_WIDTH;
...
    end else if (EXPAND) begin
        // master output is wider; merge writes
        m_axi_wdata_int = {(M_WORD_WIDTH/S_WORD_WIDTH){s_axi_wdata}};
```

```
%Error: /local/home/pengxu/work-local/enzian-pio-nic/hw/deps/verilog-axi/rtl/axi_adapter_wr.v:358:55: Replication value of 0 is only legal under a concatenation (IEEE 1800-2017 11.4.12.1)  
                                                                                                    : ... note: In instance 'PioNicEngine.axiAdapter.axi_adapter_wr_inst'
  358 |         m_axi_wdata_int = {(M_WORD_WIDTH/S_WORD_WIDTH){s_axi_wdata}};                 
      |                                                       ^
```

I would propose we take the same route as in #2754 and downgrade this error into a warning; I can work on a PR if that's desirable.

wsnyder, 2023-12-14:
>I would propose we take the same route as in https://github.com/verilator/verilator/pull/2754 and downgrade this error into a warning, maybe something called ZEROREPL

Sounds good, a pull would be appreciated.

closed 2024-01-03 by wsnyder

davidmetz, 2025-02-13:
Hi, I just wanted to chime in to say that it might be good to actually print this as a warning and not an error. 
I was quite confused when I found this issue, since with version 5.029 it still prints as an error.

wsnyder, 2025-02-13:
>Hi, I just wanted to chime in to say that it might be good to actually print this as a warning and not an error.
I was quite confused when I found this issue, since with version 5.029 it still prints as an error.

IEEE requires this to be an error, as noted on the description web page https://verilator.org/warn/ZEROREPL.  You can degrade it to a warning if you prefer with -Wwarn-ZEROREPL.

PR #4762 (2024-01-03): Downgrade zero replication width error to warning (#4753)
Resolves #4753.