verilator/verilator #4757: Calling $finish twice exits immediately.

Timmmm, 2023-12-13:
This (intentional!) bug exists since 2005. If you call `$finish` twice, Verilator will exit immediately using `std::exit(0)`. This is pretty disastrous if you are using Verilator for verification! You may be using the exit code to report test failures, and calling `std::exit(0)` will cause spurious passes.

A library should *never* call `std::exit()`.

The changelog message that added this says:

> $finish will no longer exit, but set Verilated::gotFinish(). This enables support for final statements, and for other cleanup code. If this is undesired, redefine the vl_user_finish routine. Top level loops should use Verilated::gotFinish() as an exit condition for their loop, and then call top->final(). **To prevent an infinite loop, a double $finish will still exit; this may be removed in future releases.**

Unfortunately the commit that added this was before [the very first commit](https://github.com/verilator/verilator/commit/ce10dbd11c9e43c6a041d26520f02ccd69218520) in Git, so we can't really know any more about what this potential infinite loop would be. But given it is 18 year old code I think there's a good chance that concern no longer applies.

Even if it does, an infinite loop is better than a spurious pass. I would suggest just removing that code. If infinite loops happen later it can be solved in a better way (e.g. throwing an exception).

Also the default `vl_finish()` has this code in it which seems useless (this code is also from before 2005).

```
    if (false && hier) {}
```

wsnyder, 2023-12-13:
Fair points.  The double finish was also long before we supported --main.  Could you submit a pull request to fix this?

>    if (false && hier) {}

That's intentional, it avoids an unused argument warning on hier.

wsnyder, 2023-12-24:
Fixed, thanks for the report.

closed 2023-12-24 by wsnyder

Timmmm, 2024-01-13:
Hi, sorry I actually wrote a patch for this too but totally forgot to send it! I fixed the unused variable code too though so I just sent that: #4826