verilator/verilator #4759: Segmentation fault during evaluation

flaviens, 2023-12-14:
Hi there!

I stumbled into a segmentation fault during evaluation.

```
Thread 1 "Vtop" received signal SIGSEGV, Segmentation fault.
0x000055555555c274 in VlDeleter::deleteAll (this=0x555555593698) at /usr/include/c++/10/bits/stl_iterator.h:995                              
995           __normal_iterator(const _Iterator& __i) _GLIBCXX_NOEXCEPT                                                                      
(gdb) bt                                                              
#0  0x000055555555c274 in VlDeleter::deleteAll (this=0x555555593698) at /usr/include/c++/10/bits/stl_iterator.h:995                          
#1  0x0000555555566dad in Vtop::eval_step (this=0x5555555935c0) at Vtop.cpp:50                                                               
#2  0x0000555555558d00 in Vtop::eval (this=0x5555555935c0) at ./Vtop.h:54                                                                    
#3  run_test (my_module=0x5555555935c0, simlen=1000, trace_filename=...) at /path/to/tb_base.cc:92             
#4  0x00005555555589cf in main (argc=<optimized out>, argv=<optimized out>, env=<optimized out>)                                             
    at /path/to/tb_base.cc:135
```

[Here is a report from Valgrind.](https://github.com/verilator/verilator/files/13673510/valgrind.log)

The Valgrind report suggests that the first memory leak happens before the first evaluation, hence, this issue may be linked with previous issues.

[Here is a zip with the sources.](https://github.com/verilator/verilator/files/13673512/eval_segfault.zip)

Thanks!
Flavien

wsnyder, 2023-12-15:
As with your other issues, we really would like your further debug of these and a pull request to fix.

wsnyder, 2024-01-25:
@flaviens I didn't see the shell script to make the executable work/fail so I don't know how to reproduce it.  Could you convert this to a simple as possible case that shows the failure, ideally similar to the case I posted in #4857, thanks.

wsnyder, 2024-02-14:
Please reopen if post a small example (or just make another bug).

closed 2024-02-14 by wsnyder