verilator/verilator #4760: Evasive compilation-sensitive wrong simulation result

flaviens, 2023-12-14:
Hi there!

I found an evasive bug that consistently produces wrong runtime values, but that disappears after recompilation.
Interestingly, according to Valgrind, this is due to some memory corruption:

```
==1480240== Use of uninitialised value of size 8
==1480240==    at 0x49C797B: ??? (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.32)
==1480240==    by 0x49C80E9: std::ostreambuf_iterator<char, std::char_traits<char> > std::num_put<char, std::ostreambuf_iterator<char, std::
char_traits<char> > >::_M_insert_int<unsigned long>(std::ostreambuf_iterator<char, std::char_traits<char> >, std::ios_base&, char, unsigned 
long) const (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.32)
==1480240==    by 0x49D6339: std::ostream& std::ostream::_M_insert<unsigned long>(unsigned long) (in /usr/lib/x86_64-linux-gnu/libstdc++.so.
6.0.32)
==1480240==    by 0x10CA27: operator<< (ostream:171)
==1480240==    by 0x10CA27: main (tb_base.cc:140)
==1480240== 
==1480240== Conditional jump or move depends on uninitialised value(s) 
==1480240==    at 0x49C798D: ??? (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.32)
==1480240==    by 0x49C80E9: std::ostreambuf_iterator<char, std::char_traits<char> > std::num_put<char, std::ostreambuf_iterator<char, std::
char_traits<char> > >::_M_insert_int<unsigned long>(std::ostreambuf_iterator<char, std::char_traits<char> >, std::ios_base&, char, unsigned 
long) const (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.32)
==1480240==    by 0x49D6339: std::ostream& std::ostream::_M_insert<unsigned long>(unsigned long) (in /usr/lib/x86_64-linux-gnu/libstdc++.so.
6.0.32)
==1480240==    by 0x10CA27: operator<< (ostream:171)
==1480240==    by 0x10CA27: main (tb_base.cc:140)
```

[Here is a zip](https://polybox.ethz.ch/index.php/s/zGwwGKxdH5oUuhI) with two versions of the same design+testbench, one that fails (`notok`) and one that works (`ok`).
I also added tb_base.cc into the zip, which will be useful if you would like to look at it or recompile the binaries.

Please note that if you recompile the `notok` version, then it will work as expected and produce a non-zero "output signature".

You will also have to copy `random_inputs.txt` to the path `/scratch/simufuzz-workdir/tmp/obj_dir_example_sim_1225036/random_inputs.txt` for the testbench to work.
My apologies for the inconvenience, I am not yet certain how to create the conditions for reliably reproducing such a buggy binary.

I am running `Verilator 5.019 devel rev v5.018-58-g7167d4bdc` on Ubuntu 20.04.2 LTS on an x86-64 machine.

Thanks!

wsnyder, 2023-12-15:
The backtrace says the issue is in your own tb_base.cc unrelated to Verilator.

As with your other issues, we really would like your further debug of these and if it is in Verilator, a pull request to fix.

flaviens, 2023-12-15:
Hi @wsnyder, thank you for your response! Yes indeed here is where it crashes. Yet, it crashes for a simple print, which seems to indicate that there is a corruption earlier during the program execution, which causes this line to crash subsequently.

Regarding debugging, I'd love to help, yet at the moment I have no clue about Verilator internals.

Thanks!

wsnyder, 2023-12-15:
Given this is a crash, you need to find the source of corruption using some C++ tool of choice, knowing the Verilator internals shouldn't matter until you isolate where that is.  If it turns out to be some C++ code that Verilator generates let us know what looks wrong and will provide additional guidance.

closed 2024-01-18 by wsnyder