verilator/verilator #4763: Bug report: under some conditions, 0**0 still gives 0

flaviens, 2023-12-15:
Hi there!

Using the last version of Verilator (commit fced4d6), I notice that despite the fix of #4721, under some conditions, `0 ** 0` still gives `0`.

In particular, I observed this for the line:
```
assign celloutsig_127z = 12'h000 ** { celloutsig_58z, celloutsig_67z, celloutsig_19z, celloutsig_69z, celloutsig_39z[65] };
```

in the sources that I'm attaching below. My apologies again for the size of the design.
I tested disabling the optimizations one by one and enabling tracing. None of these changes make the problem disappear, so it is not an Verilator optimization problem.
I also added the expected and actual VCD dumps, where you can focus on `celloutsig_127z`.

[Here are the files.](https://github.com/verilator/verilator/files/13686674/0_power_0_again.zip)

Thanks!
Flavien

wsnyder, 2023-12-15:
Likely something is wrong in V3Const, but not obvious to me on quick source inspection.  Please make this into a verilog-only test, in test_regress format, that is also self checking without needing waves.  Then once you see it failing (due to this bug), keep removing signals while keeping it failing the same way.  Then post that test.  Thanks.

flaviens, 2023-12-15:
Thank you for your feedback! I will reduce the test case and come back to you :+1: .

flaviens, 2023-12-19:
Hi @wsnyder ,

I could reduce the bug. In fact, its effect is very simple. The bug appears if the exponent is not constant.

For example, the design below is buggy (returns :

```
module top(out_data);
    output [11:0] out_data;
    wire [11:0] out_data;
    wire [11:0] a;
    wire [2:0] b;
    assign a = 12'h000 ** { b };
    assign b = 3'b0;
    assign out_data = a;
endmodule
```

But if I replace the power line with the line below, then it works.
```
    assign a = 12'h000 ** { 3'b0 };
```

Thanks!

closed 2023-12-20 by wsnyder