verilator/verilator #4767: Internal error with nested virtual interfaces/bind statements

eliasphanna, 2023-12-15:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```
module my_sub_module (
    input myinput
);
    wire x, y, z;
endmodule

interface Sub_if(
    input a,
    input b,
    output c
);
endinterface

interface Top_if(
    input topifinput
);

    bind my_sub_module Sub_if sub_if(
        .a(x),
        .b(y),
        .c(z)
    );
    virtual Sub_if sub_if = sub_mod1.sub_if;

endinterface

module my_top_module(
    input mywire
);

    my_sub_module sub_mod1(
        .myinput(mywire)
    );

endmodule

class My_testcase;

    virtual Sub_if sub_if;

    task initialize(virtual Top_if vif);
        this.sub_if = vif.sub_if;
        $display("Test is over");
    endtask

endclass

module test();

    My_testcase testcase = new();

    bind my_top_module Top_if top_if(
        .topifinput(mywire)
    );

    my_top_module dut(
        .mywire(foo)
    );

    initial begin
        testcase.initialize(dut.top_if);
    end

endmodule
```

I am compiling this code with `verilator -sv  -cc --top-module test test.sv` and am getting the following:
`
%Error: Internal Error: test.sv:17:38: ../V3Scope.cpp:74: Can't locate varref scope
                                     : ... note: In instance 'test.dut.top_if'
   17 |     virtual Sub_if sub_if = sub_mod1.sub_i
`
I believe that it is likely either an issue with the nested interfaces, or that it is an issue with the scoping when using the bind statement. This code does compile and run when using VCS.

What 'verilator --version' are you using?  Did you try it with the git master version?

I am using the current stable branch of verilator, v5.018.

What OS and distribution are you using?

Linux Ubuntu 20.04

May we assist you in trying to fix this in Verilator yourself?

If the change is simple I could work on it with guidance, although I do not have a lot of time or compilers experience. I am mostly looking for a potential workaround in the meantime as this pattern appears many times in the testbench I am currently adapting to work with Verilator. Any help is appreciated, thank you in advance.

wsnyder, 2023-12-20:
Here's a slightly reformatted example that can be dropped into test_regress:

module SubMod
  (
   input myinput
   );
   wire  x, y, z;
endmodule

interface Sub_Ifc
  (
   input  a,
   input  b,
   output c
   );
endinterface

interface Top_Ifc
  (
   input topifinput
   );

   bind SubMod Sub_Ifc sub_Ifc
     (
      .a(x),
      .b(y),
      .c(z)
      );
   virtual Sub_Ifc sub_Ifc = sub_mod1.sub_Ifc;

endinterface

module Dut
  (
   input mywire
   );

   SubMod sub_mod1
     (
      .myinput(mywire)
      );

endmodule

class My_testcase;
   virtual Sub_Ifc sub_Ifc;

   task initialize(virtual Top_Ifc vif);
      this.sub_Ifc = vif.sub_Ifc;
      $write("*-* All Finished *-*\n");
      $finish;
   endtask
endclass

wsnyder, 2023-12-20:
The issue seems to be V3Scope.cpp doesn't properly reconcile hierarchical references into locally bound objects.  The VARREF that throws the error does appear properly linked though by earlier stages.

ganghuang, 2024-01-11:
I run into problem the similar (or same). 
Is there a solution available?

Here is my simplified test showing the problem using cocotb with verilator as backend using the latest verilator:
```
verilator --version
Verilator 5.021 devel rev v5.020-23-g5f52c066e
```

s.sv:
```
interface if0 ();
if1 iif1();
endinterface

interface if1();
endinterface

module m2 (if1 iif1);
endmodule

module m1 (if0 iif0);
m2 m2(.iif1(iif0.iif1));
endmodule

module m0(if0 iif0);
m1 m1(.iif0(iif0));
endmodule

module test_tb(input clk100);
if0 iif0();
m0 m0(.iif0(iif0));
endmodule
```

test_tb.py
```
import cocotb
from cocotb.triggers import Timer, RisingEdge
from cocotb.clock import Clock
import sys

@cocotb.test()
async def clk(dut):
    cocotb.start_soon(Clock(period=10,signal=dut.clk100,units="ns").start())
    await Timer(10)
```

Makefile:
```
TGT=test
SIM ?= verilator
TOPLEVEL_LANG ?= verilog
VERILOG_SOURCES= s.sv
EXTRA_ARGS += --trace --trace-structs --trace-fst --debug
COMPILE_ARGS += -Wno-fatal -Wno-WIDTHTRUNC -Wno-WIDTHEXPAND -Wno-SELRANGE -Wno-IMPLICIT -Wno-PINMISSING

TOPLEVEL = $(TGT)_tb
MODULE = $(TGT)_tb

include $(shell cocotb-config --makefiles)/Makefile.sim
```

Error:
```
m -f results.xml
make -f Makefile results.xml
make[1]: Entering directory '/tmp/2/lcls2_llrf/firmware/llrf/cocotb/test'
verilator -cc --exe -Mdir sim_build -DCOCOTB_SIM=1 --top-module test_tb -Wno-fatal -Wno-WIDTHTRUNC -Wno-WIDTHEXPAND -Wno-SELRANGE -Wno-IMPLICIT -Wno-PINMISSING --vpi --public-flat-rw --prefix Vtop -o Vtop -LDFLAGS "-Wl,-rpath,/usr/local/lib/python3.10/dist-packages/cocotb/libs -L/usr/local/lib/python3.10/dist-packages/cocotb/libs -lcocotbvpi_verilator" --trace --trace-structs --trace-fst --debug --timescale 1ns/1ps s.sv /usr/local/lib/python3.10/dist-packages/cocotb/share/lib/verilator/verilator.cpp
Starting Verilator 5.021 devel rev v5.020-23-g5f52c066e
[... 56 lines cut ...]
- V3Ast.cpp:1280:     Dumping sim_build/Vtop_031_dfg-optimize.tree
- V3Const.cpp:3922:   constifyAll: 
- V3Ast.cpp:1280:     Dumping sim_build/Vtop_032_const.tree
- V3Dead.cpp:550:     deadifyDTypes: 
- V3Ast.cpp:1280:     Dumping sim_build/Vtop_033_deadDtypes.tree
- V3Inst.cpp:611:     instAll: 
- V3Ast.cpp:1280:     Dumping sim_build/Vtop_034_inst.tree
- V3Const.cpp:3922:   constifyAll: 
- V3Ast.cpp:1280:     Dumping sim_build/Vtop_035_const.tree
- V3Scope.cpp:414:    scopeAll: 
%Error: Internal Error: s.sv:12:18: ../V3Scope.cpp:73: Can't locate varref scope
-node: VARREF 0x555556f091e0 <e987> {d12as} @dt=0x555556eb1cc0@(w1)iface  iif1__Viftop [RV] <- VAR 0x555556f06480 <e943> {d2af} @dt=0x555556eb1cc0@(w1)iface  iif1__Viftop [VSTATIC]  IFACEREF
   12 | m2 m2(.iif1(iif0.iif1));
      |                  ^~~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
- V3Ast.cpp:1280:     Dumping sim_build/Vtop_990_final.tree
- V3StatsReport.cpp:219:statsReport: 
%Error: Internal Error: Aborting since under --debug
%Error: export VERILATOR_ROOT=
%Error: ulimit -s unlimited 2>/dev/null; exec setarch --addr-no-randomize /usr/local/bin/verilator_bin_dbg -cc --exe -Mdir sim_build -DCOCOTB_SIM=1 --top-module test_tb -Wno-fatal -Wno-WIDTHTRUNC -Wno-WIDTHEXPAND -Wno-SELRANGE -Wno-IMPLICIT -Wno-PINMISSING --vpi --public-flat-rw --prefix Vtop -o Vtop -LDFLAGS -Wl\,-rpath\,/usr/local/lib/python3.10/dist-packages/cocotb/libs\ -L/usr/local/lib/python3.10/dist-packages/cocotb/libs\ -lcocotbvpi_verilator --trace --trace-structs --trace-fst --debug --timescale 1ns/1ps s.sv /usr/local/lib/python3.10/dist-packages/cocotb/share/lib/verilator/verilator.cpp
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec setarch --addr-no-randomize /usr/local/bin/verilator_bin_dbg -cc --exe -Mdir sim_build -DCOCOTB_SIM=1 --top-module test_tb -Wno-fatal -Wno-WIDTHTRUNC -Wno-WIDTHEXPAND -Wno-SELRANGE -Wno-IMPLICIT -Wno-PINMISSING --vpi --public-flat-rw --prefix Vtop -o Vtop -LDFLAGS -Wl\,-rpath\,/usr/local/lib/python3.10/dist-packages/cocotb/libs\ -L/usr/local/lib/python3.10/dist-packages/cocotb/libs\ -lcocotbvpi_verilator --trace --trace-structs --trace-fst --debug --timescale 1ns/1ps s.sv /usr/local/lib/python3.10/dist-packages/cocotb/share/lib/verilator/verilator.cpp
make[1]: *** [/usr/local/lib/python3.10/dist-packages/cocotb/share/makefiles/simulators/Makefile.verilator:57: sim_build/Vtop.mk] Error 255
make[1]: Leaving directory '/tmp/2/lcls2_llrf/firmware/llrf/cocotb/test'
make: *** [/usr/local/lib/python3.10/dist-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2
```

ganghuang, 2024-01-11:
Using verilator directly also show the same error (of course)
```
verilator -sv -cc --debug  --top-module test_tb s.sv
Starting Verilator 5.021 devel rev v5.020-23-g5f52c066e
- Verilator.cpp:620:  Option --verilate: Start Verilation
- V3File.cpp:218:        --check-times failed: no input obj_dir/Vtest_tb__verFiles.dat
- V3Os.cpp:111:       export VERILATOR_ROOT=/usr/local/share/verilator # Hardcoded at build time
- V3ParseImp.cpp:292: parseFile: verilated_std
  Preprocessing /usr/local/share/verilator/include/verilated_std.sv
- V3PreShell.cpp:143:     Reading /usr/local/share/verilator/include/verilated_std.sv
- V3ParseImp.cpp:371: Lexing /usr/local/share/verilator/include/verilated_std.sv
- V3ParseImp.cpp:292: parseFile: s
  Preprocessing s.sv
- V3PreShell.cpp:143:     Reading s.sv
- V3ParseImp.cpp:357: Writing all preprocessed output to obj_dir/Vtest_tb__inputs.vpp
- V3ParseImp.cpp:371: Lexing s.sv
- V3LinkCells.cpp:210:Link --top-module: MODULE 0x555556e990e0 <e632#> {d19ai} u1=0x555556eb8a80 u4=0x555556f02630  test_tb  L0 [NONE]
- V3LinkLevel.cpp:42: modSortByLevel()
- V3Ast.cpp:1280:     Dumping obj_dir/Vtest_tb_001_cells.tree
- V3LinkDot.cpp:3838: linkDotPrimary: 
- V3LinkJump.cpp:349: linkJump: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vtest_tb_006_linkjump.tree
- V3LinkInc.cpp:306:  linkIncrements: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vtest_tb_007_linkinc.tree
- V3Param.cpp:1480:   param: 
- V3LinkDot.cpp:3844: linkDotParamed: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vtest_tb_009_linkdotparam.tree
- V3Dead.cpp:542:     deadifyModules: 
- V3Width.cpp:7614:   width: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vtest_tb_012_width.tree
- V3WidthCommit.cpp:233:widthCommit: 
- V3Const.cpp:3912:   constifyAllLive: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vtest_tb_014_const.tree
- V3Undriven.cpp:539: undrivenAll: 
- V3AssertPre.cpp:496:assertPreAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vtest_tb_015_assertpre.tree
- V3Assert.cpp:541:   assertAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vtest_tb_016_assert.tree
- V3LinkLevel.cpp:144:wrapTop: 
- V3Const.cpp:3879:   constifyAllLint: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vtest_tb_018_const.tree
- V3SplitVar.cpp:1239:splitVariable: 
- V3Inst.cpp:617:     dearrayAll: 
- V3LinkDot.cpp:3850: linkDotArrayed: 
- V3Begin.cpp:363:    debeginAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vtest_tb_023_begin.tree
- V3Tristate.cpp:1815:tristateAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vtest_tb_024_tristate.tree
- V3Unknown.cpp:526:  unknownAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vtest_tb_025_unknown.tree
- V3DfgOptimizer.cpp:233:extract: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vtest_tb_026_dfg-extract.tree
- V3DfgOptimizer.cpp:240:optimize: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vtest_tb_027_dfg-optimize.tree
- V3Inline.cpp:622:   inlineAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vtest_tb_028_inline.tree
- V3LinkDot.cpp:3850: linkDotArrayed: 
- V3DfgOptimizer.cpp:240:optimize: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vtest_tb_030_dfg-optimize.tree
- V3Const.cpp:3922:   constifyAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vtest_tb_031_const.tree
- V3Dead.cpp:550:     deadifyDTypes: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vtest_tb_032_deadDtypes.tree
- V3Inst.cpp:611:     instAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vtest_tb_033_inst.tree
- V3Const.cpp:3922:   constifyAll: 
- V3Ast.cpp:1280:     Dumping obj_dir/Vtest_tb_034_const.tree
- V3Scope.cpp:414:    scopeAll: 
%Error: Internal Error: s.sv:12:18: ../V3Scope.cpp:73: Can't locate varref scope
-node: VARREF 0x555556f091e0 <e987> {d12as} @dt=0x555556ea7cc0@(w1)iface  iif1__Viftop [RV] <- VAR 0x555556f06480 <e943> {d2af} @dt=0x555556ea7cc0@(w1)iface  iif1__Viftop [VSTATIC]  IFACEREF
   12 | m2 m2(.iif1(iif0.iif1));
      |                  ^~~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
- V3Ast.cpp:1280:     Dumping obj_dir/Vtest_tb_990_final.tree
- V3StatsReport.cpp:219:statsReport: 
%Error: Internal Error: Aborting since under --debug
%Error: export VERILATOR_ROOT=
%Error: ulimit -s unlimited 2>/dev/null; exec setarch --addr-no-randomize /usr/local/bin/verilator_bin_dbg -sv -cc --debug --top-module test_tb s.sv
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec setarch --addr-no-randomize /usr/local/bin/verilator_bin_dbg -sv -cc --debug --top-module test_tb s.sv

```

ldoolitt, 2024-01-11:
+1

Thanks @ganghuang for the simplified test case!
I confirmed the crash is still there on the next Verilator git commit ab47fc665.

wsnyder, 2025-12-18:
Now passes on master.

closed 2025-12-18 by wsnyder