verilator/verilator #4769: Can't find definition in dotted scope/variable with conditional instantiation

solomatnikov, 2023-12-16:
Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```
module m1;
   wire val1 = '0;
endmodule

module m2;
   wire val2 = '1;
endmodule

module s;
   parameter A = 0;
//   generate
      if (A == 1) begin : gen_scope
        m1 m1;
      end
      else if (A == 2) begin : gen_scope
        m2 m2;
      end
//   endgenerate
endmodule

module t;
   s #(0) s0();
   s #(1) s1();
   s #(2) s2();
   initial begin
      if (s1.gen_scope.m1.val1 != '0) $stop;
      if (s2.gen_scope.m2.val2 != '1) $stop;
      $write("*-* All Finished *-*\n");
      $finish;
   end
endmodule
```

```
%Error: t/t_gen_ifelse.v:32:24: Can't find definition of 'm1' in dotted scope/variable: 's1.gen_scope.m1'
   32 |       if (s1.gen_scope.m1.val1 != '0) $stop;
      |                        ^~
        ... Known scopes under 's1.gen_scope': m2
```

This seems to work with Questa: https://github.com/openhwgroup/cvw/blob/0eed57a0b797273ea38c2f5014e99b60244db775/src/ifu/ifu.sv#L321

What 'verilator --version' are you using?  Did you try it with the git master version?
```

bin/verilator --version
Verilator 5.019 devel rev v5.018-85-gc5ba6e22f
```

What OS and distribution are you using?

```
uname -a
Linux loran 4.18.0-526.el8.x86_64 #1 SMP Sat Nov 18 00:54:11 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```

May we assist you in trying to fix this in Verilator yourself?

**Maybe**

solomatnikov, 2023-12-17:
However, this works:

```
module m1;
   wire val1 = '0;
endmodule

module m2;
   wire val2 = '1;
endmodule

module s;
   parameter A = 0;
//   generate
      if (A == 1) begin : gen_scope1
        m1 m1;
      end
      else if (A == 2) begin : gen_scope2
        m2 m2;
      end
//   endgenerate
endmodule

module t;
   s #(0) s0();
   s #(1) s1();
   s #(2) s2();
   initial begin
      if (s1.gen_scope1.m1.val1 != '0) $stop;
      if (s2.gen_scope2.m2.val2 != '1) $stop;
      $write("*-* All Finished *-*\n");
      $finish;
   end
endmodule
```

solomatnikov, 2023-12-21:
Surprising this didn't show up earlier - this seems very basic for any configurable design.

Fixing this issue will help open source design https://github.com/openhwgroup/cvw used by upcoming textbook, _RISC-V System-on-Chip Design_.

donlon, 2023-12-22:
The problem is that currently we have two linking passes in Verilator (V3LinkDot::linkDotPrimary and V3LinkDot::linkDotParamed), where they're before and after the parameterization pass (V3Param), respectively.

V3Param resolves generate statements and removes unused branches, so the correct branch (the correct 'gen_scope') cannot be determined during the first pass, thus it throws the definition of 'm1' not found error.

> Some complicated work is needed to make this work properly

I think a simple solution could be ignoring such errors in the first pass, then after the generate statements are resolved, the second pass will do the linking properly.

Though it's not perfect (such dotted names could  not be used inside parameter expressions), but can handle most of the common situations.

I'll have a try to fix it after my recent busy time.

> Surprising this didn't show up earlier

#4427 is very similar to this case. A dotted reference was linked to an incorrect branch in the first pass. But it has been fixed by refreshing the usedParam field of the variable after each pass.