verilator/verilator #4770: Trivial register file implementation using a task fails "verilator --cc"

aappleby, 2023-12-18:
The following minimized test case passes in most tools, but fails in Verilator built from head, master branch:

```verilog
module regfile (
  input  logic       clock,
  output logic[31:0] out,
  input  logic[7:0]  addr
);

  logic[31:0] data_[256];
  logic[31:0] out_;

  always_ff @(posedge clock) begin
    tick(addr);
  end

  task automatic tick(logic[7:0] addr);
    data_[addr] <= 1;
    out_ <= data_[addr];
  endtask

  always_comb out = out_;

endmodule
```

Test run in Ubuntu 22.04.3 on x86_64:

```console
aappleby@jankytron:~$ verilator --version
Verilator 5.019 devel rev v5.018-85-gc5ba6e22f
aappleby@jankytron:~$ verilator --cc verilator_bad_regfile.sv
%Error: Internal Error: verilator_bad_regfile.sv:15:17: ../V3EmitCFunc.h:1408: Unknown node type reached emitter: BEGIN
   15 |     data_[addr] <= 1;
      |                 ^~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

I'll attempt to fix it myself.

aappleby, 2023-12-19:
Bisected to b8417abee5f10c782315059da83caec582b17785

wsnyder, 2023-12-19:
Thanks for bisecting, @kbieganski take a look?

kbieganski, 2023-12-19:
It's because of this part: https://github.com/verilator/verilator/blob/654ab117f2bc7f10a8fc97feec711c335f70d474/src/V3Fork.cpp#L419-L423
Later, without `--timing`, we don't handle this newly created fork properly.

I don't think `V3Fork` should be executed without `--timing`, as it's pretty useless in that case. I suggest wrapping this: https://github.com/verilator/verilator/blob/654ab117f2bc7f10a8fc97feec711c335f70d474/src/Verilator.cpp#L239-L240
in a:
```cpp
if (v3Global.opt.timing().isSetTrue()) {
        ...
}
```
similar to:
https://github.com/verilator/verilator/blob/654ab117f2bc7f10a8fc97feec711c335f70d474/src/Verilator.cpp#L400-L406

kbieganski, 2023-12-20:
Actually, I think this condition:
https://github.com/verilator/verilator/blob/654ab117f2bc7f10a8fc97feec711c335f70d474/src/V3Fork.cpp#L417
is insufficient. It should also check if the `VarRef` is actually writing to the variable. I'll open a PR with a fix.

PR #4773 (2023-12-20): Refine dynamic NBA condition (#4770)
The condition in `V3Fork` that determines if an NBA should be put in a fork only checks if there's a reference to a local variable on the LHS of the assignment. It should also check if the reference is a write reference. Read references are OK as they do not cause lifetime issues.

Fixes #4770.
closed 2023-12-20 by kbieganski