verilator/verilator #4771: Compiling an empty file with multiprocessing never stops (`--binary -j 0`)

goekce, 2023-12-19:
Instructions:
```
touch f.sv
verilator --binary -j 0 f.sv
# verilator does not exit
```

Without `-j 0` everything is fine.

`--debug` output:

```
Starting Verilator 5.018 2023-10-30 rev UNKNOWN.REV
- Verilator.cpp:614:  Option --verilate: Start Verilation
- V3File.cpp:218:        --check-times failed: no input obj_dir/Vf__verFiles.dat
- V3Os.cpp:108:       export VERILATOR_ROOT=/usr/share/verilator # Hardcoded at build time
- V3ParseImp.cpp:292: parseFile: verilated_std
  Preprocessing /usr/share/verilator/include/verilated_std.sv
- V3PreShell.cpp:143:     Reading /usr/share/verilator/include/verilated_std.sv
- V3ParseImp.cpp:371: Lexing /usr/share/verilator/include/verilated_std.sv
- V3ParseImp.cpp:292: parseFile: f
  Preprocessing f.sv
- V3PreShell.cpp:143:     Reading f.sv
- V3ParseImp.cpp:357: Writing all preprocessed output to obj_dir/Vf__inputs.vpp
- V3ParseImp.cpp:371: Lexing f.sv
- V3LinkLevel.cpp:42: modSortByLevel()
- V3Ast.cpp:1281:     Dumping obj_dir/Vf_001_cells.tree
- V3LinkDot.cpp:3784: linkDotPrimary: 
%Error: No top level module found
-node: NETLIST 0x555556201af0 <e1> {a0aa}  $root [1ps/1ps]
- V3Ast.cpp:1281:     Dumping obj_dir/Vf_004_linkdot-preerr.tree
- V3LinkJump.cpp:349: linkJump: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vf_008_linkjump.tree
- V3LinkInc.cpp:307:  linkIncrements: 
- V3Ast.cpp:1281:     Dumping obj_dir/Vf_009_linkinc.tree
```

Version 5.018 on ArchLinux. Did not try git.

I don't have the capacity to deliver a fix currently.

wsnyder, 2023-12-20:
This works for me, it might have been fixed by recent #4672.  Can you try the latest master please?

goekce, 2023-12-20:
You are right, it's fixed :)

closed 2023-12-20 by goekce