verilator/verilator #4772: Is there an automatic way to read all signals in a design through VPI interface?

yulong-lan, 2023-12-19:
If we want to read values of some signals, as presented by the example in verilator documentation, firstly we need to add comment behind the signal name: 

```
module our #(
     parameter WIDTH = 32
  ) (input clk);
     reg [WIDTH-1:0] readme   /*verilator public_flat_rd*/;
     reg [WIDTH-1:0] writeme;
     initial $finish;
  endmodule

```

And then read its value in the C++ wrapper:

```
 vpiHandle vh1 = vpi_handle_by_name((PLI_BYTE8*)"TOP.our.readme", NULL);
     
 if (!vh1) vl_fatal(__FILE__, __LINE__, "sim_main", "No handle found");
     
      s_vpi_value v;
      v.format = vpiIntVal;
      vpi_get_value(vh1, &v);
      printf("Value of %s: %d\n", name, v.value.integer); 
```

This is an example of reading one signal. Is there any way to automatically add the comment `/*verilator public_flat_rd*/` behind all the signals in all the modules in a design, by signals I refer to all input and output ports and all internally defined `logic` signals. More than that, automatically create a vpi handle for all these signals and read the value,  as in  `vpi_get_value(vh1, &v)`.

If not, does it mean one currently can only do this by hand and go through the signals of a design one by one?

wsnyder, 2023-12-19:
You should be able to use a wildcard with the public_flat Verilator configuration option (https://verilator.org/guide/latest/exe_verilator.html?highlight=public_flat#cmdoption-arg-public_flat_rd)

For speed reasons it's better to make as little public as possible.

As another option note #1515 which could do it, but hasn't been implemented yet.  If you are interested in contributing that please reply there.

closed 2023-12-19 by wsnyder

yulong-lan, 2023-12-20:
Thank you, now I understand the verilator comment could be added using a wildcard but a vpi handle for each signal in the C++ wrapper still has to be created by hand, right?

wsnyder, 2023-12-20:
You can vpi_iterate to find signals under a vpiModule. Please refer to some VPI tutorial. Verilator has limited VPI support but this works.