verilator/verilator #4780: Virtual interface assign from class task generates no trace dumps and doesn't trigger assign statements

jwoutersymatra, 2023-12-24:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

interface if_test;
  bit test_signal_if;
endinterface: if_test

class test;
  string name;

  virtual interface if_test tb_if;

  function new(string name = "test");
    this.name = name;
    $display("New '%s'", this.name);
  endfunction: new

  function void run();
    tb_if.test_signal_if <= 1'b0;
    #10ns;

    for(int iter=0; iter<5; iter+=1) begin
      $display("[%t] tb_if.test_signal_if(%0d) <= (%0d)", $time, tb_if.test_signal_if, ~tb_if.test_signal_if);
      tb_if.test_signal_if <= ~tb_if.test_signal_if;
      #10ns;
    end
  endfunction: run

endclass: test

module top;

bit  test_signal_always;
bit  test_signal_assign1;
bit  test_signal_assign2;

test    i_test;
if_test i_if_test();

assign test_signal_assign1 = i_if_test.test_signal_if; // This doesn't work / generates no trace
assign test_signal_assign2 = test_signal_always;       // This works

always @(i_if_test.test_signal_if) begin
//always @(*) begin
  test_signal_always <= i_if_test.test_signal_if; // test_signal_if generates no trace activity
end

initial begin
  i_test = new("i_test");

  i_test.tb_if = i_if_test;

  i_test.run();

  $finish;
end

// Waveforms
initial begin
  $dumpfile("top.vcd");
  $dumpvars(0,top);
end

endmodule

What 'verilator --version' are you using?  Did you try it with the git master version?

5.018

What OS and distribution are you using?

Windows + WSL

May we assist you in trying to fix this in Verilator yourself?

Sure!

wsnyder, 2023-12-24:
Adding

   always @(test_signal_assign1) $display("ass1");
   always @(test_signal_assign2) $display("ass2");

shows that there are indeed the correct events, they are just not tracing.

I suspect that the code which determines what signals may change is not properly accounting for the interface. This might be straightforward to fix, look at how Vm_traceActivity works.

As to tracing into the virtual interface, currently all tracing is determined statically, there is also no tracing of class internals. This is likely complicated to fix - we'd need a runtime dumper of each interface/class and code to appropriately call into them based on the pointers to their objects.

jwoutersymatra, 2023-12-28:
[image]

jwoutersymatra, 2023-12-28:
Not sure if this is the correct thing to do, but at least it now traces the virtual interface itself (which was a bit unexpected, but better as what I was hoping for) and it also traces the assign statement to "test_signal_assign1"

ArturBieniek4, 2026-09-02:
@jwoutersymatra FYI timing controls in `function` are illegal SV. When changing the repro to use `task` there, it started working after https://github.com/verilator/verilator/pull/7465 was merged.

closed 2026-09-02 by ArturBieniek4