verilator/verilator #4781: How to debug when there is a segmentation fault but not when using gdb?

jordankrim, 2023-12-26:
When I run the (Verilator) testbench I get a segmentation fault, but when I run gdb there isn't one and the testbench runs to completion successfully, so how do I debug the seg fault under these conditions?

wsnyder, 2023-12-26:
Something in your or the Verilated code is likely corrupting memory, and hence makes the bug not fully reproducible.  You can debug as with other C++ programs, I'd suggest 

1. Try -CFLAGS -DVL_DEBUG
2. Try using the address sanitizer at runtime ("-CFLAGS -fsanitize=address,undefined -LDFLAGS -fsanitize=address,undefined")
3. Worse case, setup your shell to make a core dump file, then use gdb to load that core dump file to see where the code was.

jordankrim, 2023-12-28:
Apologies but I am a novice when it comes to gdb, so I have no idea how to do #3, which it looks like I will have to do because both #1 and #2 didn't work.  Again, #2 has thousands of lines of output, and I cannot find the source of the error using grep for error.  Is it difficult to do #3 and if not how do I do it?

jordankrim, 2023-12-28:
Or this is what valgrind gives:
==1176318== Conditional jump or move depends on uninitialised value(s)
==1176318==    at 0x111832: _vl_moddiv_w(int, unsigned int*, unsigned int const*, unsigned int const*, bool) (verilated.cpp:506)
==1176318==    by 0x18C836: Vmodexp___024root___nba_sequent__TOP__0(Vmodexp___024root*) (Vmodexp___024root__DepSet_h2222839b__0.cpp:1894)
==1176318==    by 0x1397B5: Vmodexp___024root___eval_nba(Vmodexp___024root*) (Vmodexp___024root__DepSet_h2222839b__1.cpp:118)
==1176318==    by 0x139988: Vmodexp___024root___eval_phase__nba(Vmodexp___024root*) (Vmodexp___024root__DepSet_h2222839b__1.cpp:161)
==1176318==    by 0x139B80: Vmodexp___024root___eval(Vmodexp___024root*) (Vmodexp___024root__DepSet_h2222839b__1.cpp:232)
==1176318==    by 0x1390FD: Vmodexp::eval_step() (Vmodexp.cpp:64)
==1176318==    by 0x10E474: runForClockCycles(unsigned int) (RSA.h:35)
==1176318==    by 0x10E4D6: read_word(unsigned int) (tb_modexp.cpp:30)
==1176318==    by 0x10E521: wait_ready() (RSA.h:45)
==1176318==    by 0x10E950: modexp_encrypt() (RSA.h:203)
==1176318==    by 0x10DADD: main (tb_modexp.cpp:189)
==1176318==  Uninitialised value was created by a stack allocation
==1176318==    at 0x111392: _vl_moddiv_w(int, unsigned int*, unsigned int const*, unsigned int const*, bool) (verilated.cpp:437)
==1176318== 
But, I am having trouble correlating this to the exact problem.

wsnyder, 2023-12-29:
The sanitizer dump suggests you are not initializing every input before calling eval, that could easily cause core dumps.  (It could also just be a harmless internal Verilator issue though.)  You can try just using 
"-CFLAGS -fsanitize=address -LDFLAGS -fsanitize=address" as those are more critical.

wsnyder, 2023-12-29:
Also make sure you are compiling with VL_DEBUG

jordankrim, 2023-12-29:
Is that -CFLAGS -DVL_DEBUG or just VL_DEBUG (on the verilator command line)?

wsnyder, 2023-12-29:
If using verilator to build, `-CFLAGS -DVL_DEBUG`

wsnyder, 2023-12-29:
C strings have a terminating 0.

jordankrim, 2023-12-29:
OK. So how do I fix it?   I changed it to 64 in the for loop ((but maybe this is not the correct fix?):

s = strlen((char*)b64);
for(t=0;((unsigned) t<s)&(c!=b64[t]);t++);

The problem goes away but then I got this with further (?) sanitizing:
  
/home/jkrim/verilator/include/verilated.cpp:475:55: runtime error: index 17 out of bounds for type 'unsigned int [17]'
/home/jkrim/verilator/include/verilated.cpp:475:57: runtime error: store to address 0x7ffcfbdc5a34 with insufficient space for an object of type 'uint32_t'
0x7ffcfbdc5a34: note: pointer points here
  00 00 00 00 01 00 00 00  fc 01 00 00 00 00 00 00  7f 00 00 00 81 00 00 00  7f 00 00 00 01 00 00 00

This is based this code:

uint32_t un[VL_MULS_MAX_WORDS + 1];  // Fixed size, as MSVC++ doesn't allow [words] here
and VL_MULS_MAX_WORDS = 16 so bombs on index OOB on 17

This is internal to Verilator (verilated.cpp -- right?) so is this real, or did I just create another artifact of C++ via my "fix" for the OOB b64 variable?

jordankrim, 2023-12-29:
I changed this in verilatedos.h:

#define VL_MULS_MAX_WORDS 128  ///< Max size in words of MULS operation

Progress because now I only get these?

/home/jkrim/verilator/include/verilated.cpp:482:59: runtime error: shift exponent 32 is too large for 32-bit type 'unsigned int'
/home/jkrim/verilator/include/verilated.cpp:493:59: runtime error: shift exponent 32 is too large for 32-bit type 'unsigned int'
/home/jkrim/verilator/include/verilated.cpp:540:62: runtime error: shift exponent 32 is too large for 32-bit type 'unsigned int'

And the lines in question are:

    for (int i = vw - 1; i > 0; --i) {
        vn[i] = (rwp[i] << s) | (shift_mask & (rwp[i - 1] >> (32 - s)));      <==== LINE 482
    }

    for (int i = uw - 1; i > 0; --i) {
        un[i] = (lwp[i] << s) | (shift_mask & (lwp[i - 1] >> (32 - s)));     <===== LINE 493
    }

        for (int i = 0; i < vw; ++i) {
            owp[i] = (un[i] >> s) | (shift_mask & (un[i + 1] << (32 - s)));    <=== LINE 540
        }

So, on these lines what is the error "shift exponent 32 is too large for 32-bit type" referring to?
Thanks for the help.

jordankrim, 2023-12-29:
One more thing.  With all the above, the design no longer works, so changing the line to #define VL_MULS_MAX_WORDS 128 seems to get rid of a lot of sanitize problems, but then perhaps the  runtime error: shift exponent 32 is too large for 32-bit type 'unsigned int' somehow breaks the input to the design causing it to fail?

jordankrim, 2023-12-29:
My design has 2 very large elements.  One is a 2048 bit X 2048 bit multiplier which results in an accumulator that is 4096 bits.  Then that accumulator is fed into a 4096 bit (dividend) by 2048 bit (divisor) with a quotient and a remainder output.  Am I missing some input on the command line to verilator that will allow these large values and if so what because I tried to add:
-DDETECTARRAY_MAX_INDEXES=4096 --unroll-count 4096
But, unfortunately, that did not solve the issue output by the sanitization: 
/home/jkrim/verilator/include/verilated.cpp:477:55: runtime error: index 17 out of bounds for type 'unsigned int [17]'
/home/jkrim/verilator/include/verilated.cpp:477:57: runtime error: store to address 0x7fffe4e9b714 with insufficient space for an object of type 'uint32_t'
0x7fffe4e9b714: note: pointer points here
  00 00 00 00 01 00 00 00  fc 01 00 00 00 00 00 00  7f 00 00 00 81 00 00 00  7f 00 00 00 01 00 00 00

wsnyder, 2023-12-30:
>VL_MULS_MAX_WORDS

There is a check for this value when you Verilate.  If you see sanitizer errors that go away with this defined much larger (at least 130), there must be some missing part of the check, but anyhow it's fine to increase it.  However the reason for that static might no longer apply now that MSVC supports more recent C++ standards.

The shift error is harmless, I will clean it up.

jordankrim, 2023-12-30:
I had to modify both of these in verilatedos.h:

#define VL_MULS_MAX_WORDS 128  ///< Max size in words of MULS operation

#ifndef VL_VALUE_STRING_MAX_WORDS
    #define VL_VALUE_STRING_MAX_WORDS 128  ///< Max size in words of String conversion operation
#endif

The results are as follows:

Sanitize results in these messages: 
/home/jkrim/verilator/include/verilated.cpp:483:59: runtime error: shift exponent 32 is too large for 32-bit type 'unsigned int'
/home/jkrim/verilator/include/verilated.cpp:495:59: runtime error: shift exponent 32 is too large for 32-bit type 'unsigned int'
/home/jkrim/verilator/include/verilated.cpp:543:62: runtime error: shift exponent 32 is too large for 32-bit type 'unsigned int'
Fails first part of test, so even though you say the shift is not a problem, with sanitize in play it seems to be.

Test with GDB:
Passes first part of test.

Normal (No seg. Fault.  Hurrah!)
Passes first part of test

Test with valgrind (based on the input object from the GDB):
Passes first part of test.

Because the "normal" method resulted in no seq. fault, I ran the both parts of the test and both parts passed, so seems the OOB causing the seg fault is solved and that just leaves the question of why the sanitize fails with those messages related to the shift.

wsnyder, 2023-12-30:
The shift should be fixed in master.

I added an assert you should see fire if you leave VL_MULS_MAX_WORDS at the default. Please see what Verilog code is creating that, you should have gotten a verilation-time error saying to increase it, I would like to fix that.

wsnyder, 2023-12-30:
The shift should be fixed in master.

I added an assert you should see fire if you leave VL_MULS_MAX_WORDS at the default. Please see what Verilog code is creating that, you should have gotten a verilation-time error saying to increase it, but as you did not I would like to fix that to give the error.

jordankrim, 2023-12-30:
I tested your fix and this is what I got:
Writing MOD 53: Vmodexp: /home/jkrim/verilator/include/verilated.cpp:447: WData* _vl_moddiv_w(int, WDataOutP, WDataInP, WDataInP, bool): Assertion `uw <= 16' failed.
make: *** [Makefile:109: vcd_new_div_sans_secworks] Aborted (core dumped)

jordankrim, 2023-12-31:
I tried this:
verilator -DVL_MULS_MAX_WORDS=128 ...
and 
verilator +define+VL_MULS_MAX_WORDS=128 ...
But I still get the assertion.

wsnyder, 2023-12-31:
1. What I am interested in is what Verilog source is causing the error. 
   a. Verilate with --debug
   b. Run and see the assert, use "gdb" to backtrace the assert to see what Verilated C++ line is causing the assertion.
    c. Look at the obj/*999*.tree file to see what line corresponds to that generated C++ line - you can find the function name in the .tree file then look forward from there for  some MULS or similar.
    d. Once you have the .tree file there's an e.g. {b123} which means line 123 in a file - see docs/internal.rst for more information.
    e. Send the .tree file line, and the corresponding Verilog source (presumably a multiply) along with the Verilog definitions of variables that e.g. multiply uses.

2. Change the default define for VL_MULS_MAX_WORDS to 128 here https://github.com/verilator/verilator/blob/ca3b5fbe995d9afd0a62c2b46dfbe796415c7a72/include/verilatedos.h#L453 then recompile Verilator.

jordankrim, 2023-12-31:
The bt is this:

0  0x00007ffff74b5acf in raise () from /lib64/libc.so.6
#1  0x00007ffff7488ea5 in abort () from /lib64/libc.so.6
#2  0x00007ffff7488d79 in __assert_fail_base.cold.0 () from /lib64/libc.so.6
#3  0x00007ffff74ae426 in __assert_fail () from /lib64/libc.so.6
#4  0x000055555555d450 in _vl_moddiv_w (lbits=4096, owp=0x7fffffff8380, lwp=0x7fffffff8780, rwp=0x7fffffff8580, is_modulus=true)
    at /mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/x86_64-conda-linux-gnu/include/c++/12.1.0/x86_64-conda-linux-gnu/bits/stl_list.h:447
#5  0x00005555555d872a in ::Vmodexp___024root___nba_sequent__TOP (vlSelf=0x555555646440)
    at /mnt/data/projects/ipvault/jkrim/CEP-Setup/cep_v4_4_new2_test_verilator/CEP/generators/mitll-blocks/src/main/resources/vsrc/rsa/obj_dir/verilated_funcs.h:1881
#6  0x0000555555585930 in Vmodexp___024root___eval_nba (vlSelf=0x555555646440)
    at /mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/x86_64-conda-linux-gnu/include/c++/12.1.0/bits/verilated.h:118
#7  0x0000555555585b03 in Vmodexp___024root___eval_phase__nba (vlSelf=0x555555646440)
    at /mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/x86_64-conda-linux-gnu/include/c++/12.1.0/bits/verilated.h:161
#8  0x0000555555585cfb in Vmodexp___024root___eval (vlSelf=0x555555646440)
    at /mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/x86_64-conda-linux-gnu/include/c++/12.1.0/bits/verilated.h:232
#9  0x0000555555585278 in Vmodexp::eval_step (this=0x555555646280) at Vmodexp__Syms.h:64
#10 0x000055555555a497 in runForClockCycles (pCycles=<optimized out>) at /home/jkrim/verilator-master/include/hex2ascii.h:35
#11 0x000055555555a4e2 in read_word (pAddress=<optimized out>) at /home/jkrim/verilator-master/include/RSA.h:34
#12 0x000055555555a513 in wait_ready () at /home/jkrim/verilator-master/include/hex2ascii.h:44
#13 0x000055555555a942 in modexp_encrypt () at /home/jkrim/verilator-master/include/hex2ascii.h:202
#14 0x0000555555559aee in main (argc=<optimized out>, argv=0x7fffffffbee8, env=<optimized out>) at /home/jkrim/verilator-master/include/RSA.h:193

First, there is no file by that name.  All the tree files are named Vmodexp_###.tree. 
I am not sure which of these lines is the function I need to look for.  Line 193 in RSA.h is this:
    write_word(ADDR_MODULUS_LENGTH , 0x00000041);
I tried to grep for write_word in all the tree files but it doesn't appear.  Neither does _vl_moddiv_w.  So, I am not sure which of the lines in the bt I am supposed to be looking for in the tree file.  The file internals.rst is quite large and I do not exactly know what I am looking for in it, too.

On a slight tangent, I did change the VL_MULS_MAX_WORDS to 128 in my verilatedos.h file, but I did not recompile Verilator, so why did it work i.e. my sim runs good without an assert?

wsnyder, 2023-12-31:
Ok, I changed the default width in master, we'll consider that good enough.

closed 2023-12-31 by wsnyder

jordankrim, 2023-12-31:
Thank you for your help.   I learned a lot of good debugging tools (i.e. gdb and valgrind along with Verilator sanitize) that can help me pinpoint things like segmentation faults in the future if I come across them and thanks for fixing the Verilator code to resolve my issue.