verilator/verilator #4783: Problem compiling a simple test bench

davidharrishmc, 2023-12-28:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

I'm unable to build a simple testbench that works in Questa.  I'm new to Verilator simulation and am unsure if the problem is in my installation, my Linux environment, or the Verilog code.

Here is the failing file (remove the .txt extension)

[fulladder.sv.txt](https://github.com/verilator/verilator/files/13786456/fulladder.sv.txt)

What 'verilator' command line do we use to run your example?
either of the following two
verilator --binary --top-module testbench fulladder.sv
verilator --timescale "1ns/1ns" --timing -cc --exe --build --top-module testbench fulladder.sv

I get a bunch of errors about coroutines.
```
harris@chips:~/cvw/examples/verilog/fulladder$ verilator --timescale "1ns/1ns" --timing -cc --exe --build --top-module testbench fulladder.sv
make: Entering directory '/home/harris/cvw/examples/verilog/fulladder/obj_dir'
ccache g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable        -c -o verilated_timing.o /usr/local/share/verilator/include/verilated_timing.cpp
In file included from /usr/local/share/verilator/include/verilated_timing.h:47,
                 from /usr/local/share/verilator/include/verilated_timing.cpp:23:
/usr/include/c++/11/coroutine:334:2: error: #error "the coroutine header requires -fcoroutines"
  334 | #error "the coroutine header requires -fcoroutines"
      |  ^~~~~
In file included from /usr/local/share/verilator/include/verilated_timing.cpp:23:
/usr/local/share/verilator/include/verilated_timing.h:100:10: error: ‘coroutine_handle’ in namespace ‘std’ does not name a template type
  100 |     std::coroutine_handle<> m_coro;  // The wrapped coroutine handle
...
```

What 'verilator --version' are you using?  Did you try it with the git master version?
Verilator 5.019 devel rev v5.018-101-g54ad04b6e

What OS and distribution are you using?
I am running Ubuntu 22.04LTS and updated all of of the dependent tools per the Verilator installation instructions.  I compiled Verilator from source last week.

May we assist you in trying to fix this in Verilator yourself?
I think the question is too basic.

wsnyder, 2023-12-28:
Can you attach what ./configure prints, and make sure after building with that configure'd it still doesn't work?

davidharrishmc, 2023-12-29:
Thank you for the speedy reply. That solved the issue.

I had upgraded from Ubuntu 20.4LTS to 22.4LTS yesterday in an effort to solve the coroutine problem, but had not recompiled Verilator.  Rerunning configure and fixed the issue.  It might be worth mentioning in the installation instructions that Verilator no longer supports 20.4LTS.

closed 2023-12-29 by davidharrishmc

wsnyder, 2023-12-29:
We regress 20.04 every commit, so it is supported. The problem was the switching. Glad it is fixed.

dreamerx0, 2024-06-05:
I meet almost the same issue too on ubuntu 20.04. 
for ubuntu 20.04 support, first you need to install gcc-10-multilib & g++-10-multilib and make it default gcc/g++, the default version gcc-9 didn't support coroutine
then you need to rebuild verilator, rerun configure & make & make install

version: Verilator 5.025 devel rev v5.024-62-gd4b358330