verilator/verilator #4784: Building verilator on MacOS fails

Rasfunk, 2023-12-28:
I need the HEAD version of verilator on MacOS, as I want to use it with CocoTB and the CocoTB AXI extensions (see #4618).

Simply using `brew install --head verilator` fails with the following messages:
```
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C obj_dbg       TGT=../../bin/verilator_bin_dbg VL_DEBUG=1 -f ../Makefile_obj
/opt/homebrew/bin/python3 ../astgen -I .. --astdef V3AstNodeDType.h --astdef V3AstNodeExpr.h --astdef V3AstNodeOther.h --dfgdef V3DfgVertices.h V3Const.cpp
g++ -std=gnu++11   -Og -ggdb -gz -DVL_DEBUG -D_GLIBCXX_DEBUG -DVERILATOR_INTERNAL_ -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -Qunused-arguments -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/Users/rasfunk/Documents/Projects/git/isp/verilator\" -c V3Const__gen.cpp -o V3Const__gen.o
g++ -std=gnu++11   -Og -ggdb -gz -DVL_DEBUG -D_GLIBCXX_DEBUG -DVERILATOR_INTERNAL_ -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -Qunused-arguments -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/Users/rasfunk/Documents/Projects/git/isp/verilator\" -c ../V3Error.cpp -o V3Error.o
In file included from ../V3Error.cpp:18:
In file included from ../V3Error.h:28:
../V3ThreadPool.h:65:29: error: no member named 'make_unique' in namespace 'std'
        : m_ptWrapperp{std::make_unique<PTWrapper<Signature>>(std::move(pt))} {}
                       ~~~~~^
../V3ThreadPool.h:65:61: error: expected '(' for function-style cast or type construction
        : m_ptWrapperp{std::make_unique<PTWrapper<Signature>>(std::move(pt))} {}
                                        ~~~~~~~~~~~~~~~~~~~~^
../V3ThreadPool.h:192:5: error: 'auto' return without trailing return type; deduced return types are a C++14 extension
    auto enqueue(Callable&& f) VL_MT_START;
    ^
../V3ThreadPool.h:225:12: error: 'auto' return without trailing return type; deduced return types are a C++14 extension
    static auto waitForFutures(std::list<std::future<T>>& futures) {
           ^
../V3ThreadPool.h:341:1: error: 'auto' return without trailing return type; deduced return types are a C++14 extension
auto V3ThreadPool::enqueue(Callable&& f) VL_MT_START {
^
5 errors generated.
make[2]: *** [V3Error.o] Error 1
make[1]: *** [../bin/verilator_bin_dbg] Error 2
make: *** [verilator_exe] Error 2
```

As can be seen from the log, the compiler is fixed at `g++ -std=gnu++11` while the code seems to use C++14 features.

Checking the `./configure` output, it seems like the `gnu++11` option is part of the compiler command.

```
configuring for Verilator 5.019 devel
checking whether to perform partial static linking of Verilator binary... yes
checking whether to use tcmalloc... check
checking whether to use -m32... no
checking whether to build for coverage collection... no
checking whether to use hardcoded paths... yes
checking whether to show and stop on compilation warnings... no
checking whether to run long tests... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking for g++... g++
checking whether the compiler supports GNU C++... yes
checking whether g++ accepts -g... yes
checking for g++ option to enable C++11 features... -std=gnu++11
checking for a BSD-compatible install... /usr/bin/install -c
compiler is g++ -std=gnu++11 --version = Apple clang version 15.0.0 (clang-1500.1.0.2.5)
checking that C++ compiler can compile simple program... yes
checking for ar... ar
checking for perl... /opt/homebrew/bin/perl
checking for python3... /opt/homebrew/bin/python3
checking for flex... /usr/bin/flex
/usr/bin/flex --version = flex 2.6.4 Apple(flex-34)
checking for bison... /usr/bin/bison
/usr/bin/bison --version = bison (GNU Bison) 2.3
checking for ccache... no
checking for stdio.h... yes
checking for stdlib.h... yes
[... 71 lines cut ...]
config.status: creating include/verilated_config.h
config.status: creating verilator.pc
config.status: creating verilator-config.cmake
config.status: creating verilator-config-version.cmake
config.status: creating src/config_package.h
config.status: src/config_package.h is unchanged

Now type 'make' (or sometimes 'gmake') to build Verilator.
```

I was able to work around the issue by manually removing the `-std=gnu++11` from `src/Makefile_obj` after `./configure` but prior to building. This is not a solution of course. If I have a better solution, I will create a PR.

```
# Programs
CC = gcc
CXX = g++ -std=gnu++11
LINK = g++ -std=gnu++11
LEX = /usr/bin/flex
LFLAGS = -d
PYTHON3 = /opt/homebrew/bin/python3
YACC = /usr/bin/bison
OBJCACHE ?=  
```

Might be related to #3081 and #2502.

Rasfunk, 2023-12-28:
`make CXX="g++ -std=gnu++20"` is another work around to get it to build.

wsnyder, 2023-12-28:
Where is the -std=gnu++11 coming from?  The Verilator config script should be trying more recent versions first, and the log file you made seems like that is being passed into configure directly, which would be the problem to address, as opposed to something verilator is mis-configuring.

Rasfunk, 2023-12-28:
I agree, and Verilator configure actually selects gnu++17 in `CFG_CXXFLAGS_STD_NEWEST`, as you can see in this check:
```
checking whether g++ -std=gnu++11 accepts -std=gnu++17... yes
```
But somehow, this gnu++17 parameter is not propagated to anywhere else.
No idea where autoconf gets the idea to append `-std=gnu++11` to the CXX command.

`./configure` is called without any parameters.

I checked a few places but no luck. Just using plain g++ works, as the default for my compiler would be c++17. g++ on my system is actually clang though, not sure if this plays a role here.

But even if this is is purely a problem of my system, I still see two potential issues for the Verilator autoconf:

1. Why doesn't Verilator build with clang and std=gnu++11? In theory, it should be supported and build just fine.
2. Why do the autoconf checks succeed with this compiler setting, but the build fails? Ideally, autoconf should prevent that.

I am not blocked by this issue, but I will look further into the it and share updates here if I find anything.

wsnyder, 2023-12-29:
c++14 was required as Jan 23 though we have not officially forced it yet. https://verilator.org/guide/latest/deprecations.html

Rasfunk, 2023-12-29:
Ok, that makes a lot of sense and is also a straight forward fix for my issue.
I have created a PR with some trivial changes to ensure C++14.

wsnyder, 2023-12-29:
Thanks for the pull.