verilator/verilator #4785: `verilated.mk` expects PCHs to be named without `.gch` suffix.

jerryz123, 2023-12-28:
Hello, I'd appreciate any assistance in debugging this error. This appeared when we bumped from 5.008 to 5.018.
```
Verilator 5.018 2023-10-30 rev conda-forge build 0
```

I see the compilation of the simulator referencing a file `VTestDriver__pch.h.fast`...
```
/nscratch/jerryz/chipyard-proj/chipyard-master/.conda-env/bin/x86_64-conda-linux-gnu-c++ -Os  -I.  -MMD -I/nscratch/jerryz/chipyard-proj/chipyard-master/.conda-env/share/verilator/include -I/nscratch/jerryz/chipyard-proj/chipyard-master/.conda-env/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -g  -O3 -std=c++17 -I/nscratch/jerryz/chipyard-proj/chipyard-master/.conda-env/riscv-tools/include -I/nscratch/jerryz/chipyard-proj/chipyard-master/tools/DRAMSim2 -I/nscratch/jerryz/chipyard-proj/chipyard-master/sims/verilator/generated-src/chipyard.harness.TestHarness.RocketConfig/gen-collateral   -DVERILATOR -include /nscratch/jerryz/chipyard-proj/chipyard-master/sims/verilator/generated-src/chipyard.harness.TestHarness.RocketConfig/chipyard.harness.TestHarness.RocketConfig.plusArgs -include /nscratch/jerryz/chipyard-proj/chipyard-master/sims/verilator/generated-src/chipyard.harness.TestHarness.RocketConfig/chipyard.harness.TestHarness.RocketConfig/VTestDriver.h -DVL_TIME_CONTEXT   -fcoroutines -include VTestDriver__pch.h.fast -c -o VTestDriver.o VTestDriver.cpp
```

However, no such file exists in the output directory, some similarly named files are: 
```
generated-src/chipyard.harness.TestHarness.RocketConfig/chipyard.harness.TestHarness.RocketConfig/VTestDriver__pch.h
generated-src/chipyard.harness.TestHarness.RocketConfig/chipyard.harness.TestHarness.RocketConfig/VTestDriver__pch.h.fast.d
generated-src/chipyard.harness.TestHarness.RocketConfig/chipyard.harness.TestHarness.RocketConfig/VTestDriver__pch.h.fast.gch
```

Investigating further, the installed makefile expects the pch files with a `.gch` suffix only if Clang, but this seems to contradict what the GCC docs say....

```
# Compiler's filename prefix for precompiled headers, .gch if clang, empty if GCC
CFG_GCH_IF_CLANG = 
```

https://gcc.gnu.org/onlinedocs/gcc/Precompiled-Headers.html

The `verilator` command I am using is:
```
verilator \
  --main \
  --timing \
  --cc \
  --exe \
  -CFLAGS "-g  -O3 -std=c++17 -I/nscratch/jerryz/chipyard-proj/chipyard-master/.conda-env/riscv-tools/include -I/nscratch/jerryz/chipyard-proj/chipyard-master/tools/DRAMSim2 -I/nscratch/jerryz/chipyard-proj/chipyard-master/sims/verilator/generated-src/chipyard.harness.TestHarness.RocketConfig/gen-collateral   -DVERILATOR -include /nscratch/jerryz/chipyard-proj/chipyard-master/sims/verilator/generated-src/chipyard.harness.TestHarness.RocketConfig/chipyard.harness.TestHarness.RocketConfig.plusArgs" \
  -LDFLAGS " -L/nscratch/jerryz/chipyard-proj/chipyard-master/.conda-env/riscv-tools/lib -Wl,-rpath,/nscratch/jerryz/chipyard-proj/chipyard-master/.conda-env/riscv-tools/lib -L/nscratch/jerryz/chipyard-proj/chipyard-master/sims/verilator -L/nscratch/jerryz/chipyard-proj/chipyard-master/tools/DRAMSim2 -lriscv -lfesvr -ldramsim "  \
  --threads 1 \
  --threads-dpi all \
  -O3 \
  --x-assign fast \
  --x-initial fast \
  --output-split 10000 \
  --output-split-cfuncs 100 \
  --assert -Wno-fatal \
  --timescale 1ns/1ps \
  --max-num-width 1048576 \
  +define+CLOCK_PERIOD=1.0 \
  +define+RESET_DELAY=777.7 \
  +define+PRINTF_COND=TestDriver.printf_cond \
  +define+STOP_COND=!TestDriver.reset \
  +define+MODEL=TestHarness \
  +define+RANDOMIZE_MEM_INIT \
  +define+RANDOMIZE_REG_INIT \
  +define+RANDOMIZE_GARBAGE_ASSIGN \
  +define+RANDOMIZE_INVALID_ASSIGN \
  +define+VERILATOR \
  --top-module TestDriver \
  --vpi \
  -f /nscratch/jerryz/chipyard-proj/chipyard-master/sims/verilator/generated-src/chipyard.harness.TestHarness.RocketConfig/sim_files.common.f  \
  -o /nscratch/jerryz/chipyard-proj/chipyard-master/sims/verilator/simulator-chipyard.harness-RocketConfig \
  -Mdir /nscratch/jerryz/chipyard-proj/chipyard-master/sims/verilator/generated-src/chipyard.harness.TestHarness.RocketConfig/chipyard.harness.TestHarness.RocketConfig \
  -CFLAGS "-include /nscratch/jerryz/chipyard-proj/chipyard-master/sims/verilator/generated-src/chipyard.harness.TestHarness.RocketConfig/chipyard.harness.TestHarness.RocketConfig/VTestDriver.h"                                                                                                                                                                         
```

This is using Ubuntu 20.04.

I'd be happy to investigate further on my end, if I knew where to look further.

wsnyder, 2023-12-28:
See #4730, if you can reproduce it without trouble, the change to make is described there.

Note we regress every commit with Ubuntu 20.04 successfully, so there must be some local configuration you have that is nonstandard.

jerryz123, 2023-12-28:
It looks like the CI for 20.04 uses gcc-9.4.0 and 10.5.0, while the CI for 22.04 uses 11.4.0.
I am using 12.3.0, which is the same as what the user in #4730 reported.

I'm not sure how the other user resolved his issue, since after upgrading to Ubuntu 23, he may have also reverted back from 12.3.0.

I am having trouble getting this to work with the PCHs.
I expected passing `-include VTestDriver__pch.h.fast.gch`  to work, based on the GCC documentation, but that results in a lot of  bizarre warnings and errors, like the file format or encoding isn't correct.

```
./VTestDriver__pch.h.fast.gch:37:72: warning: missing terminating " character                                                                                                                                                                 
./VTestDriver__pch.h.fast.gch:38:1: warning: null character(s) ignored                                                                                                                                                                        
   38 | <U+0001><U+0010><U+0000><U+0000><U+0000><U+0000><f7>                                                                                                                                                                                  
      | ^~~~~~~~                                                                                                                                                                                                                              
./VTestDriver__pch.h.fast.gch:38:5: warning: null character(s) ignored                                                                                                                                                                        
   38 | <U+0001><U+0010><U+0000><U+0000><U+0000><U+0000><f7>                                                                                                                                                                                  
      |                                 ^~~~~~~~                                                                                                                                                                                              
./VTestDriver__pch.h.fast.gch:38:19: warning: null character(s) ignored                                                                                                                                                                       
   38 | <U+0001><U+0010><U+0000><U+0000><U+0000><U+0000><f7>                                                                                                                                                                                  
      |                                                                ^                                                                                                                                                                      
./VTestDriver__pch.h.fast.gch:38:23: warning: null character(s) ignored                                                                                                                                                                       
   38 | <U+0001><U+0010><U+0000><U+0000><U+0000><U+0000><f7>                                                                                                                                                                                  
      |                                                                    ^                                                                                                                                                                  
./VTestDriver__pch.h.fast.gch:38:53: warning: null character(s) ignored                                                                                                                                                                       
   38 | <U+0001><U+0010><U+0000><U+0000><U+0000><U+0000><f7>                                                                                                                                                                                  
      |                                                                                                  ^                                                                                                                                    
./VTestDriver__pch.h.fast.gch:38:57: warning: null character(s) ignored                                                                                                                                                                       
   38 | <U+0001><U+0010><U+0000><U+0000><U+0000><U+0000><f7>                                                                                                                                                                                  
      |                                                            
      ```

wsnyder, 2023-12-28:
The file seems corrupted, try first a make clean and ccache clean (make sure to use ccache), then if still not working making a gcc bash script that does a "sleep 5" after running gcc with the command line - See #4708.

Anyhow I suspect you'll find the suffix is needed, and so then consider making a pull to do the feature test in ./configure as suggested in #4730

jerryz123, 2023-12-30:
Actually I was wrong, it seems gcc will implicitly append a `.gch` to the filename passed in with `-include`, so the current flags are correct. Passing `-include xxx.h.gch` with the `.gch` suffix causes the mess of errors above. This seemed unintuitive to me, based on my reading of the GCC documentation:

> This also works with -include. So yet another way to use precompiled headers, good for projects not designed with precompiled header files in mind, is to simply take most of the header files used by a project, include them from another header file, precompile that header file, and -include the precompiled header.

The actual source of the error is from passing additional unrelated `-include` flags BEFORE the PCH `-include`. This seems to only show up in gcc 12.3.0. In 12.3.0, passing unrelated `-include` seems to cause the file not found error, even though the correctly named `VTestDriver__pch.h.fast.gch` is there. 

For my use case, the extraneous `-include` were inserted from how I set `-CFLAGS` argument. These could be removed without issue. 

No further action is required here.

closed 2023-12-30 by jerryz123

reoLantern, 2024-01-29:
> Actually I was wrong, it seems gcc will implicitly append a `.gch` to the filename passed in with `-include`, so the current flags are correct. Passing `-include xxx.h.gch` with the `.gch` suffix [...]

Hi, I encountered the same issue, but I didn't understand the solution you mentioned. What file should be modified?
p.s. I use gcc 11.4.0

jerryz123, 2024-01-29:
Can you check the invocation of gcc that caused the error? Any instances of `-include` that are not the pch include caused the error for me.

reoLantern, 2024-01-29:
> Can you check the invocation of gcc that caused the error? Any instances of `-include` that are not the pch include caused the error for me.

The error I got is:
```
<command-line>: fatal error: VGPGPU_SimWrapper__pch.h.slow: No such file or directory
```

The compile command is:

```
ccache g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=1 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -O1 -DVL_USER_STOP -DVL_USER_FATAL -DVL_USER_FINISH -DTOP_TYPE=VGPGPU_SimWrapper -include VGPGPU_SimWrapper.h -fPIC -shared -I/usr/lib/jvm/java-11-openjdk-amd64/include -I/usr/lib/jvm/java-11-openjdk-amd64/include/linux -fvisibility=hidden   -Os -c -o GPGPU_SimWrapper-harness.o ../GPGPU_SimWrapper-harness.cpp
```

so perhaps the `-include VGPGPU_SimWrapper.h` caused the problem. But this command is generated by Chisel when I use the VerilatorBackendAnnotation, so currently I haven't make it clear ╯︿╰

reoLantern, 2024-01-29:
I change my system to Ubuntu 22.04 and the same error occured. Compile command is like:

> ccache g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=1 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -O1 -DVL_USER_STOP -DVL_USER_FATAL -DVL_USER_FINISH -DTOP_TYPE=VGPGPU_SimWrapper -include VGPGPU_SimWrapper.h -fPIC -shared -I/usr/lib/jvm/java-11-openjdk-amd64/include -I/usr/lib/jvm/java-11-openjdk-amd64/include/linux -fvisibility=hidden   -Os -c -o GPGPU_SimWrapper-harness.o ../GPGPU_SimWrapper-harness.cpp
> ccache g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=1 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -O1 -DVL_USER_STOP -DVL_USER_FATAL -DVL_USER_FINISH -DTOP_TYPE=VGPGPU_SimWrapper -include VGPGPU_SimWrapper.h -fPIC -shared -I/usr/lib/jvm/java-11-openjdk-amd64/include -I/usr/lib/jvm/java-11-openjdk-amd64/include/linux -fvisibility=hidden   -include VGPGPU_SimWrapper__pch.h.fast -c -o VGPGPU_SimWrapper_ArrayMultiplier__DepSet_h7eefd941__6.o VGPGPU_SimWrapper_ArrayMultiplier__DepSet_h7eefd941__6.cpp

Phantom1003, 2024-04-24:
> For my use case, the extraneous `-include` were inserted from how I set `-CFLAGS` argument. These could be removed without issue.

Jerry's solution is correct. I triggered this bug on Ubuntu 22.04 with gcc 11.4.0 once again. It is recommended that hardware maintainers carefully review their `include` lists.